arendst/Tasmota #4848: BlastBot

thucar, 2019-01-07:
Got my hands on a new IR blaster: Blastbot. It's running an ESP8266 12E and it has 6 IR blasters and one receiver. 

 I don't have necessary dupont wires at hand right now in my office, so I'll be trying to flash it with Tasmota tomorrow to see if this puppy is what I've been looking for to handle my IR needs.

Jason2866, 2019-01-07:
Backup the orig. firmware with esptool ;-)

thucar, 2019-01-08:
Ok, so - progress. Flashing Tasmota was a breeze with the pins broken out and the button being wired directly to GPIO0.

The pinout seems to be:
gpio2 = Onboard LED (red)
gpio4 = Onboard LED (green)
gpio12 = IR SEND
gpio13 = DS18B20
gpio16 = IR RCV

All the pins seem inverted though, so if I define GPIO12 as IRsend(08) then the LEDS instantly come on and stay on. Gotta look into inverting the logics in xdrv_05_irremote.ino

thucar, 2019-01-08:
Well that was easier than expected. IRremoteESP8266 library comes with a built in option for inverting the output. After inverting, sending out IR codes works fine.

CallMeAreks, 2019-04-06:
Hi, did you compile a custom bin for this device? If yes, could you share it?