arendst/Tasmota #4849: Ntp with raspberry

leranp, 2019-01-07:
Hi,
I have closed the internet access to al of my sonoffs, but no the clock isn't sync, i have opened NTP server with raspberry pi and i am tried to change the ntp server address in the tasmota console, but it doesn't take the clock and date from the ntp server.
I have checked with my computer and it works.
Any help?

ascillato, 2019-01-07:
I used the standard ntp from raspberry and worked fine. No modifications, just default.

Now I moved to Home Assistant with the NTP server add on. That works fine also.

ascillato, 2019-01-07:
See https://github.com/arendst/Sonoff-Tasmota/issues/2871

leranp, 2019-01-07:
I have removed the ntp and deleted the ntp config file from raspberry, but the tasmota still doesn't pull the data.
What can be the problem?

ascillato, 2019-01-07:
Please, in order to properly help you, we need more information as requested before. Please, complete the troubleshooting template and post the output of your console at startup. Please use weblog 4 for more debug information

leranp, 2019-01-07:
i have check the raspberry ntp and it look like the sonoff ask for the time
`     0      0    0 . 3 4 2049986  4097 192.168.31.30
     1     24    0 . 3 4    281 49153 192.168.31.229
     2      4    0 . 3 4   1881  4097 192.168.31.20
   172   2174    0 . 3 3      3 59563 192.168.31.224
`
`00:00:00 CFG: Loaded from flash at F7, Count 173
00:00:00 APP: Boot Count 75
00:00:00 SRC: Restart
00:00:00 Project Wemos Wemos (Topic sonoff_wemos, Fallback Sonoff_Wemos, GroupTopic sonoffs) Version 6.2.1-2_3_0
00:00:00 CFG: Saved to flash at F6, Count 174, Bytes 3584
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 LH in mode 11N as sonoff_wemos-3720...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Connected
00:00:03 DNS: Initialized
00:00:03 HTP: Web server active on sonoff_wemos-3720.local with IP address 192.168.31.30
00:00:04 MQT: Attempting connection...
00:00:04 MQT: Connected
00:00:04 MQT: tele/sonoff_wemos/LWT = Online (retained)
00:00:04 MQT: cmnd/sonoff_wemos/POWER = 
00:00:04 MQT: Subscribe to cmnd/sonoff_wemos/#
00:00:04 MQT: Subscribe to cmnd/sonoffs/#
00:00:04 MQT: Subscribe to cmnd/Sonoff_Wemos/#
00:00:04 SRC: MQTT
00:00:04 RSL: Received Topic cmnd/sonoff_wemos/POWER, Data Size 2, Data ON
00:00:04 RSL: Group 0, Index 1, Command POWER, Data ON
00:00:04 MQT: stat/sonoff_wemos/RESULT = {"POWER":"ON"}
00:00:04 MQT: stat/sonoff_wemos/POWER = ON (retained)
00:00:04 MQT: tele/sonoff_wemos/INFO1 = {"Module":"Generic","Version":"6.2.1","FallbackTopic":"Sonoff_Wemos","GroupTopic":"sonoffs"}
00:00:04 MQT: tele/sonoff_wemos/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff_wemos-3720","IPAddress":"192.168.31.30"}
00:00:04 MQT: tele/sonoff_wemos/INFO3 = {"RestartReason":"Software/System restart"}
00:00:04 MQT: stat/sonoff_wemos/RESULT = {"POWER":"ON"}
00:00:04 MQT: stat/sonoff_wemos/POWER = ON (retained)
00:00:05 SRC: PulseTimer
00:00:05 MQT: stat/sonoff_wemos/RESULT = {"POWER":"OFF"}
00:00:05 MQT: stat/sonoff_wemos/POWER = OFF (retained)
00:00:13 MQT: tele/sonoff_wemos/STATE = {"Time":"1970-01-01T00:00:13","Uptime":"0T00:00:13","Vcc":2.732,"POWER":"OFF","Wifi":{"AP":1,"SSId":"LH","RSSI":100,"APMac":"24:05:88:38:E1:47"}}
00:00:13 MQT: tele/sonoff_wemos/SENSOR = {"Time":"1970-01-01T00:00:13","Switch1":"OFF"} (retained)`

leranp, 2019-01-07:
after long investigation this was solve the problem by:

sudo /etc/init.d/ntp stop
sudo ntpd -q -g
sudo /etc/init.d/ntp start