arendst/Tasmota #4854: SHELLY1 - TASMOTA 6.4.1 - NO GPIO configuration

joseluisgodinho, 2019-01-08:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
_A clear and concise description of what the bug is._

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ ] _Searched the problem in issues and in the wiki_
- [ ] _Hardware used_ : 
- [ ] _Development/Compiler/Upload tools used_ :
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [ ] _You have tried latest release or development binaries?_ :
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```

**To Reproduce**
_Steps to reproduce the behavior:_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

joseluisgodinho, 2019-01-08:
Hello All

After flashing the shelly 1, the GPIO configuration is not available

Jason2866, 2019-01-08:
Please try latest dev version from here. [sonoff.bin](http://thehackbox.org/tasmota/sonoff.bin)
There where changes for Shelly

arendst, 2019-01-08:
There is no GPIO configuration for Shelly. Why should there be?

meingraham, 2019-01-08:
@arendst Theo,

[This](https://www.youtube.com/watch?v=mSENAY9_AlI&feature=youtu.be&t=695&ab_channel=digiblurDIY)?

arendst, 2019-01-08:
Ah. Makes sense. I will update Shelly1 config and hope no-one ever tries this when connected to mains...

arendst, 2019-01-08:
Good luck and only to be used when Shelly 1 is connected to 12V as advertised.

closed 2019-01-08 by arendst

digiblur, 2019-01-09:
I do mention several times to treat it all as mains voltage and use approved mains switches.