arendst/Tasmota #4855: Added configuration for TYWE3S module on YTF Universal IR device (send & recv)

pastudan, 2019-01-08:
I was looking for a smart IR device to control some IR curtains, and tried this YTF Wifi Smart IR remote from Amazon (https://www.amazon.com/gp/product/B07H46LJZK). 

Unfortunately the app didn't allow custom programmed IR codes to be hooked up to Automation / Google Home / IFTTT (only predefined ones), so I cracked it open to see what was inside and if it was hackable. I became hopeful after seeing what looked like a familiar ESP8266, and discovering that the TYWE3S is exactly that. 

I stumbled across this repo in my googling, and decided to give it a try. Got it running with no problem, and I wanted to share my pin configuration after a few hours of trial and error:

```
{ "YTF-Universal",
    0,                  // GPIO0
    0,                  // GPIO1
    0,                  // GPIO2
    0,                  // GPIO3
    GPIO_LED1_INV,      // GPIO4 Blue Led
    GPIO_IRRECV,        // GPIO5 IR Receiver
    0,                  // GPIO6
    0,                  // GPIO7
    0,                  // GPIO8
    0,                  // GPIO9
    0,                  // GPIO10
    0,                  // GPIO11
    0,                  // GPIO12
    GPIO_KEY1,          // GPIO13 Button
    GPIO_IRSEND,        // GPIO14 IR Transmitter
    0,                  // GPIO15
    0,
    0                   // ADC0 A0 Analog input
  },
```

This device is a sleek little smart IR module for ~$30, and this software makes it a ton more useful. Thanks!

Jason2866, 2019-01-08:
Great finding, could you add to wiki please?
Thx!

ceaswaran, 2019-01-08:
@pastudan Great work. i happen to buy a Tuya app based universal IT remote, I have flashed it with TASMOTA.
Product Link: https://www.aliexpress.com/item/Zemismart-IR-Bridge-Control-Air-condition-Fan-TV-For-Smart-Life-App-Google-Home-Alexa-Echo/32922000072.html?spm=a2g0s.9042311.0.0.35214c4dTdP77D&fbclid=IwAR2vFVs9KPsP4d6DKNMCrmDJtk8iwrhP1RHjEB4U6odEAzy-W_0_ovPj7-0

I was able to figure out IR receive & Send GPIO (ref below image)

[image]
[image]
[image]
[image]

**Query:** 
1. My TV Remote has a single POWER button . Which Pressed 1st time TURN ON. 2nd Time TURN OFF. In tasmota console using IR receive I see the same code when I pressed the power button once or twice. If I send the same code (Which I received using IRreceive) I see TV only TUN ON. TURN OFF NOT WORKING. please help

OFF: IRsend {"Protocol":"NEC","Bits":32,"Data":50153655} WORKING

ON: IRsend {"Protocol":"NEC","Bits":32,"Data":50153655} NOT WORKING

2.  None of my AC (Air conditioner's) Remote is unable to recive or send the IR signal.Since all my AC's protocal is UNKNOWN i am try to receive RAW codes. but AC RAW dosent work in Tasmota
teminates half way as below dot dot dot
`19:07:25 MQT: tele/sonoffir/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":114,"Data":2896510163,"RawData":[3788,1448,622,1136,616,1138,592,442,590,442,614,418,590,1160,618,418,612,418,622,1134,592,1054,726,416,614,1090,664,452,580,414,616,1138,616,1136,620,416,616,1136,618,1136,618,416,616,454,580,1134,618,414,618,414,620,1136,616,454,580,414,616,416,618,414,618,414,618,414,616,414,620,452,580,416,614,416,616,418,614,416,616,452,580,452,578,418,620,416,616,414,616,416,616,416,616,418,616,1134,618,416,616,418,618, ...`

@arendst advised to change the tasmoto source with following values
`my_user_config.h
IR_RCV_BUFFER_SIZE 1000
IR_RCV_TIMEOUT 50
PubSubClient.h and sonoff_post.h
MQTT_MAX_PACKET_SIZE 2000` 

, but still with those changes i dont get the full RAW code.

Have you come across these issues? Do u have a solution?

andrethomas2, 2019-01-08:
Closing this issue as it the PR made for this request has been merged into development.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-08 by andrethomas2

ceaswaran, 2019-01-09:
@pastudan prefixed 0 & added code till dot dot dot. using IRsend
none of the AC's work :(

jandae, 2019-02-11:
I got the same device, how do I apply your configuration? Sorry very new to this

viertaxa, 2019-02-13:
This seems to be the same layout for the Monoprice Stitch device. (Same hardware)

viertaxa, 2019-02-13:
Interestingly, I had to cut the trace going from the RX and TX serial pads towards the USB port. It appears they were being bridged somehow, causing all sorts of serial errors.

arendst, 2019-02-13:
@viertaxa nice solution. In the latest development versions I disabled serial comms for this device which also solves the serial loopback issue. So no need to cut wires.

pastudan, 2019-02-13:
@viertaxa thanks for mentioning that. I thought I just had a faulty device! I didn't have the loopback issue when I first loaded the firmware, but it popped up later. Can you show an image with exactly which trace you cut? My PCB is black so it's difficult to follow the traces.

I wonder why they would bridge those 🤔

Jason2866, 2019-02-13:
@pastudan  load latest firmware and you dont have this issue without cutting traces

pastudan, 2019-02-13:
@Jason2866 @arendst wouldn't that create a problem where you can't configure your device over serial? I needed the serial interface to configure my wifi username / password after loading the sonoff firmware, since I didn't want to take the time to compile a version with them built in. Maybe there's a better way I'm missing?

Jason2866, 2019-02-13:
Via wifi. Read wiki...

jamietang68, 2019-02-13:
@pastudan 
Do you mind to share how did you set the device into flashing mode? I have the exact same unit, but couldnt set it into flashing mode.

I tried shorting GPIO 0 to GND and RST to GND, but failed.

Thanks!

pastudan, 2019-02-13:
@jamietang68 if I recall correctly, you need to connect GPIO 0 to GND during boot, and then release it during flashing

jamietang68, 2019-02-13:
@pastudan Thanks for reply. Yeah i tried that but it doesnt seems to work.. do you remember if the blue led is blinking or stays on during flashing mode?

meingraham, 2019-02-13:
If any LEDs blink while powering up then it didn't go into programming mode. When GPIO0 is grounded (and kept grounded for 3-4 seconds), no LEDs should light.

jamietang68, 2019-02-13:
@meingraham got it! thanks for the info!

rjcooper725, 2019-03-22:
Anyone still here?  I develop tech for special needs kids.  I am wanting to control this YTF IR blaster via a iOS app.  Is there an API or has anyone hacked with Python or something?
RJ :)

pastudan, 2019-03-22:
@rjcooper725 this firmware supports MQTT which is a well documented API that you could build an iOS app around. https://github.com/arendst/Sonoff-Tasmota/wiki/MQTT-Overview

ascillato, 2019-03-23:
@rjcooper725, besides MQTT, Tasmota also supports control over HTTP and over KNX IP MULTICAST

See wiki at: https://github.com/arendst/Sonoff-Tasmota/wiki/Commands for control over HTTP

rjcooper725, 2019-03-23:
Hi!

Please check this one out for me? (IR blaster)

Tx!

2016for, 2019-05-06:
Thanks for providing the pinout for this device @pastudan.

Slightly off topic, but I want to use the codes from [IRRemoteESP8266](https://github.com/markszabo/IRremoteESP8266) (as that's the only library that has my Fujitsu AC codes), can I create my own firmware using that library and flash it onto this device using these pins?

Thanks again for the information!

ceaswaran, 2019-05-06:
@2016for  use Tasmota it can send receive raw codes

2016for, 2019-05-19:
Hey I am finally ready to flash this. Any suggestions on how to flash it without having to solder wires onto the flat PCB solder pads?
(Want to backup firmware before flashing, so I don't think Tuya-Convert will be helpful)

arduino43, 2019-05-20:
Tuya Convert does backup firmware

damocles-git, 2019-05-25:
Hello, I flashed this IR blaster:

> @pastudan Great work. i happen to buy a Tuya app based universal IT remote, I have flashed it with TASMOTA.
> Product Link: [...]

and it works only when I power the esp directly from the usb/serial programmer, but if I power it from the microusb connector it does not show any working activity, green leds are powered and the voltage in the gnd and vcc pins are correct.

I am out of ideas :-( ... any idea why it might be happening?

arduino43, 2019-05-25:
Can you upload a picture of the internal board/PCB?

ceaswaran, 2019-05-26:
@damocles-git i use it with tasmota . are u able to see sonoff-XXXX wifi?

damocles-git, 2019-05-27:
> Can you upload a picture of the internal board/PCB?

I'm sorry, I forgot to upload them:

[image]
[image]

damocles-git, 2019-05-27:
> @damocles-git i use it with tasmota . are u able to see sonoff-XXXX wifi?

Hi ceaswaran, yes .. I can use tasmota with the esp8266 chip (this is mqtt and http) only when it is powered directly from the serial programmer even the green leds are on, wifi settings are configured through the user_config_override_sample.h file, but when I disconnect the programmer and use the microusb connector to power it the green leds are on again and not wifi activity is showed in my wifi network, no smartconfig, nothing!

arduino43, 2019-05-27:
Did you flash the chip using DIO? This happened to me when using DOUT before.

damocles-git, 2019-05-27:
> Did you flash the chip using DIO? This happened to me when using DOUT before.

I flashed it with DOUT, as stated in the wiki:

`Only Flash Mode DOUT is supported. Do not use Flash Mode DIO / QIO / QOUT as it might seem to brick your device.`  

this is my command line: 

'esptool.py -c esp8266  -p /dev/ttyUSB0 -b 921600 write_flash --flash_mode dout --flash_size 1MB 0x0 sonoff-6.5.0.2.bin'

damocles-git, 2019-05-31:
Finally it worked :-)

Normally I use an usb hub to connect the devices, I test it with a different power adapter and it worked, very big thanks for this fantastic software!!

Plexcon, 2019-08-16:
> @pastudan prefixed 0 & added code till dot dot dot. using IRsend
> none of the AC's work :(

Were you able to fix the dot dot dot issue?

thomas-lentz, 2019-08-23:
Hi Pastudan,
how did you open the IR Blaster? There aren't any screws visible on mine.

Thanks for your help.
Greetings
 Tom

arduino43, 2019-08-23:
If I remember correctly you have to use a thin screwdriver or knife to get the case open (separating the top and bottom of the case).

There is a way to do it without snapping plastic or otherwise damaging the unit, just move around the edges until you feel a spot with less resistance and start applying a little pressure there.

arduino43, 2019-08-23:
I just looked at my device and it seems like the place to start is the green highlighted area in the attached photo

https://imgur.com/a/XtxcgVC

thomas-lentz, 2019-08-25:
Hi Arduino43,
thank you for your quick help - that worked fine.
Greetings
 Tom

Freakadude, 2019-08-31:
Hi,
I have 3 tuya type Plugs which i successfully flashed with the tuyaconvert OTA method. I also have this same YTF tuya type IR bridge and wanted to flash it as well via the same method since based on the wiki it should work the same without having to crack the unit open.

I did the exact same steps as i did with the plugs. I put the YTF bridge into pairing mode (blue led blinking fast) and then started the flashing process. Initial results show that the convert program is connecting to the device as the flashing stops after a few seconds and i get a solud blue led which then turns off. 

Unfortunately the flashing process does not go further from this point onwards. The original FW is not downloaded and i cannot flash tasmota.

Is there something i should do differently for this to work? Don't really want to crack the unit open and solder. I just want to get the same tuyaconvert OTA method working if thats a doable way.

Cheers

ffleandro, 2020-12-31:
@Freakadude Did you manage to flash tasmota using tuya convert?
Also interested in this.
Thanks

koivunen, 2021-01-08:
A listing in Aliexpress promising TYWE3S (with pictures) shipped me a WR3E so be careful with new devices. 
The seller shipped me the correct version afterwards but did hint me that Tuya has stopped selling TYWE3S chips.

ffleandro, 2021-01-08:
I managed to flash an YTF IR Bridge (TYWE3S) using tuya-convert without needing to open it's case or solder any wires.

ffleandro, 2021-01-08:
@koivunen Regarding these new non-ESP based devices, here is a full thread discussing them: 
 - https://community.home-assistant.io/t/unkown-tuya-chip/153591/85

s-hadinger, 2021-05-12:
Please refrain from posting in an old and closed issue. Reach out on Discord.

The encoding is explained here: https://tasmota.github.io/docs/IRSend-RAW-Encoding/ and there's an online tool to convert to uncompressed format

adrianslabu, 2021-05-18:
> How to handle some raw data like what I'm getting from my ceiling fan,
> {"IrReceived":{"Protocol":"UNKNOWN","Bits":312,"Hash":"0xDA16662A","Repeat":0,"RawData":"+615-635+595bC-1635+565d+590-1610E-1660+540-1690+535-670FgF-1640E-695+570-690I-4745C-685F-665A-1580F-1605+620-1590+560gVgF-720KgE-1615FpOpF-4720V-715I-715I-1630E-1585FmCgEg+430-940+510mE-1585+645-725+485-740I-4840+610-710IrCuVtFwA-1585N-1560EnF-1560VzXyEnE-4750FpKpEhEgEdFhI-1665IlFt+645-1585F-775+485pE-4805XsKpO-1630F-1585VgFgFgVaV-1680+495gAsVl+640-4775VbVrI-1630Ou+625t+600-1600V-1585VnIhEmCp+510-715E-4720V-660FpA-1530V-1585AdF-1585V-1585E-700FdEgC-640CpFqO-715FbV-1550CdCgA-1560FmCsFtVmE-640VsC-4880Fa+640-745+510tVgF-1585VgFwA-700XtVgFnEsF-4725VrIsV-1630EtAgF-1585VgFlFmX-1585A-750+485pC-4775EpErFtCgFgOmF-1585Vy+510dFzEpFyI-4855V-640A-660CuEgFgVmEt+645-640CuFgE-750I-640+640-4700AsFpOh+520-1630FgE-1665EdInE-1720+510zA-645FlE-4770C-630C-750+505-1630EdFtVdI-1670IsVgFmEsEsA-4860Ln+480yK-1630FgEdFzFtVnXmFzX-640V-770+485-4800+515yXlXh+510-1665XgF-1665F-1560+640-640CmIdKnVpX-4775OpEsFgXdAwFgEmCy+51

Go to [https://tasmota.hadinger.fr/util](https://tasmota.hadinger.fr/util) and paste the code from your RawData and press Go to decode it.
Send the resulted code from output using IRSend like below:

`IRSend 0,your_decoded_raw_data`

Exemple:
If your decoded rawdata is `1200,475,1220,455,390,1290,405,1280,390,1290,390,1290` you have to use `IRSend 0,1200,475,1220,455,390,1290,405,1280,390,1290,390,1290`

floydcohen, 2021-07-09:
> I managed to flash an YTF IR Bridge (TYWE3S) using tuya-convert without needing to open it's case or solder any wires.

Can you give some info about how to do this?

adrianslabu, 2021-07-09:
> > I managed to flash an YTF IR Bridge (TYWE3S) using tuya-convert without needing to open it's case or solder any wires.
> 
> Can you give some info about how to do this?

Find here all the information you need:
https://github.com/ct-Open-Source/tuya-convert