arendst/Tasmota #4858: ArduinoIDE compiled vs precompiled differences? mqtt issue related

sindudas, 2019-01-08:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Flashing a Wemos D1 Mini like Generic ESP8266 Module with 1M(no spiffs), DIO, v2 Higher Bandwith (tested other options too), and SDK 2.5.0b2.
Issue is tele is reporting as "RSL" not sending to MQTT "MQT", but other things where sending to MQTT like POWER cmnd and state changes
```
16:04:26 RSL: sonoff/tele/STATE = {"Time":"2019-01-08T16:04:26","Uptime":"0T00:08:14","Vcc":3.010,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"wifi","BSSId":"C0:4D:F0:02:B0:C6","Channel":9,"RSSI":100}}
16:04:47 MQT: sonoff/stat/RESULT = {"POWER":"ON"}
16:04:47 MQT: sonoff/stat/POWER = ON
16:04:56 RSL: sonoff/tele/STATE = {"Time":"2019-01-08T16:04:56","Uptime":"0T00:08:44","Vcc":3.012,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"wifi","BSSId":"C0:4D:F0:02:B0:C6","Channel":9,"RSSI":100}}
```
But if I upgrade with prebuilt version http://thehackbox.org/tasmota/020500/sonoff.bin it works will all messages to mqtt.

How to reproduce the prebuild compilation with ArduinoIDE? My .bin size is much bigger to than the prebuild. I make my own firmware to change some default configs.

I appreciate any help!

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : Arduino IDE 1.8.5 / Esp8266 SDK 2.5.0 Beta2
- [x] Hardware used : Wemos D1 Mini, Sonoff T1 1ch/2ch, Sonoff Pow
- [x] If a pre-compiled release or development binary was used, which one? : 
http://thehackbox.org/tasmota/020500/sonoff.bin
- [x] You have tried latest release or development binaries? : 6.4.1.7
- [x] Provide the output of command ``status 0`` :
```
16:00:32 RSL: sonoff/stat/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:00:32 RSL: sonoff/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/020500/sonoff.bin","RestartReason":"External System","Uptime":"0T00:04:20","StartupUTC":"2019-01-08T14:56:12","Sleep":50,"BootCount":5,"SaveCount":13,"SaveAddress":"FA000"}}
16:00:32 RSL: sonoff/stat/STATUS2 = {"StatusFWR":{"Version":"6.4.1.7(sonoff)","BuildDateTime":"2019-01-08T15:53:40","Boot":31,"Core":"2_5_0_BETA2","SDK":"3.0.0-dev(c0f7b44)"}}
16:00:32 RSL: sonoff/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["wifi",""],"TelePeriod":30,"SetOption":["00008009","558180C0","00000000"]}}
16:00:32 RSL: sonoff/stat/STATUS4 = {"StatusMEM":{"ProgramSize":647,"Free":356,"Heap":21,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640E0","FlashMode":2,"Features":["00000809","0FD8E594","000383A0","23B617CE","00003BC0"]}}
16:00:32 RSL: sonoff/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoff-5646","IPAddress":"192.168.1.2","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"5C:CF:7F:14:76:0E","Webserver":2,"WifiConfig":4}}
16:00:32 RSL: sonoff/stat/STATUS6 = {"StatusMQT":{"MqttHost":"domus1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_14760E0E","MqttUser":"mqttuser","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
16:00:32 RSL: sonoff/stat/STATUS7 = {"StatusTIM":{"UTC":"Tue Jan 08 15:00:32 2019","Local":"Tue Jan 08 16:00:32 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:15","Sunset":"17:34"}}
16:00:32 MQT: sonoff/stat/STATUS10 = {"StatusSNS":{"Time":"2019-01-08T16:00:32"}}
16:00:32 RSL: sonoff/stat/STATUS11 = {"StatusSTS":{"Time":"2019-01-08T16:00:32","Uptime":"0T00:04:20","Vcc":3.005,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"wifi","BSSId":"C0:4D:F0:02:B0:C6","Channel":9,"RSSI":100}}}
```

**(Please, remember to close the issue when the problem has been addressed)**

sindudas, 2019-01-08:
Already copied files (boards.txt and platform.txt).
In the Wiki shows Arduino IDE for SDK 2.3.0/2.4.0/2.4.1/2.4.2 but not for 2.5.0
Sorry, Flash Mode usually set to DOUT, but because too many tests, changing SDK version/Target, last change leave it as DIO by mistake.
Now, tested with correct Flash Mode (DOUT) and again with lwIP 2 - Higher Bandwidth (no features). Same result with MQTT not sending Tele/STATE and firmware size is 647Kb.

Jason2866, 2019-01-08:
Have you disabled Exceptions?
2.5.0. Is in wiki
[image]

sindudas, 2019-01-08:
Thanks for the fast replies!

Now tested with Exceptions = Disabled. Program Size now 521kB.

Seen section "Copy files" for 2.5.0, but understand to only overwrite with platform.txt and boards.txt from tasmota github. I didn't understand to have to do something with:

> Remove default compile options for floating point printf and scanf which are not being used by Tasmota and save 15k code space. Remove default exceptions code from Tasmota save 60k code space

[image]

Some clue to solve the mqtt issue?

sindudas, 2019-01-08:
I have no idea how to do that: 

> Remove default compile options for floating point printf and scanf which are not being used by Tasmota and save 15k code space

Tried to search, but with no luck. Maybe a little help to see the light. Last think that was differ from the prebuild image, and possible make mqtt issues?

Jason2866, 2019-01-08:
With copying the provided platforms.txt you have disabled printf and scanf.
If mqtt isnt working start from scratch with a new Arduino installation.
Do a portable Arduino install. There is no need to uninstall your current one.
Just follow exactly! this steps https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE

andrethomas2, 2019-01-08:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-08 by andrethomas2

sindudas, 2019-01-09:
Jason2866 finally found my issue compiling Tasmota.

Doing a portable Arduino fresh install, with success. Hard way to me, because I'm on Mac. Doing it on Parallels with a Windows XP installed version, with troubles with Arduino Install (not portable) and issues with WinXP. But finally success.

Then once I got it to work, I have reviewed my Arduino IDE installation on Mac. Found another library PubSubClient other than the PubSubClient-EespEasy attached with Tasmota. Erased the other and replaced the attached, and it worked.

Thank you very much for the help. Good work done with Tasmota.

PS: It will be more clear if move "Remove default exceptions code from Tasmota save 60k code space" to the unordered list below. As it's an action required to save code space, not information what replacing boards.txt and platform.txt do.

Jason2866, 2019-01-09:
Glad it works finally for you. You could help other users if you edit the wiki to be more helpful for users which are in the same situation like you