arendst/Tasmota #4859: V6.4.1 in and out of "Unavailable" On Home Assistant / Switch turning red TasmoAdmin

c3reeman, 2019-01-08:
After Months of flawless Tasmota operation I've started to run into problems that my devices are in and out or availability and constantly showing "unavailable" in Home Assistant. This is often only brief maybe 2-3 seconds but its consistently happening.

This could be due to the latest upgrade to V6.4.1 but not guaranteed it was a few days after the upgrade i noticed it. Ive got 28 devices running and up until a few days ago they were solid as a rock with no problems so wifi signal strength shouldn't be the problem in TasmoAdmin I've checked my lowest wifi signal is 65% i run a google mesh network with 5 nodes ensuring good coverage everywhere. 
The runtime of the sonoff's is not affected so this isn't a device restarting problem, its not a device specific problem either as its affecting all tasmota switches.

I wondered if interference could be an issue because i recently started using zigbee2mqtt, I've unplugged that and turned off any other device that i could think might cause interfearnce but it didn't make any difference... still the "unavailable"

below is a copy of my Home Assistant entry for an example switch, but like i say nothing has changed and this config has worked for the least 6 months.

`  - platform: mqtt
    name: "Dressing Room Lights"
    state_topic: "stat/sonoff003/POWER"
    command_topic: "cmnd/sonoff003/POWER"
    availability_topic: "tele/sonoff003/LWT"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    payload_available: "Online"
    payload_not_available: "Offline"
    retain: false`

This could be a home assistant fault but i have also noted that on TasmoAdmin the switches occasionally turn red, and this is at the same time HomeAssisistant shows "Unavailable" additionally i haven't upgraded home assistant in a while.

Jason2866, 2019-01-08:
Please fill out complete issue template with Status 0
Without this template we cant help

c3reeman, 2019-01-08:
Sorry about that please check now, Apologies

c3reeman, 2019-01-08:
it appears it is a connection issue, Cant understand why this has only just started happening though, here is the log straight after the fault occurs

```
18:11:49 WIF: Connecting to AP1 Mill House in mode 11N as Sonoff005...
18:11:55 WIF: Connected
18:11:55 DNS: Initialized
18:11:56 MQT: Attempting connection...
18:11:56 MQT: Connected
18:11:56 MQT: tele/sonoff005/LWT = Online (retained)
18:11:56 MQT: cmnd/sonoff005/POWER = 
```

Jason2866, 2019-01-08:
Try if with core 2.3.0 your issue is solved http://thehackbox.org/tasmota/release/sonoff.bin
Do an OTA with this firmware

andrethomas2, 2019-01-08:
Please follow https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#wifi-issues-arduino-core-versions-and-expressif-sdk

Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-08 by andrethomas2

c3reeman, 2019-01-08:
Thanks all, @Jason2866 your suggestion was the one that worked and as you said flashing OTA to 2.3.0 works fine, so pleased i didn't have to hard wire into them and re flash would have taken me days pulling them all out of the ceiling

john5788, 2019-01-14:
I just ran into this problem too. Are the future releases going to be built with 2.3.0 core or a fix will be in place for 2.4.2? Using @Jason2866 provided build allows device to show up in Home Assistant.

Jason2866, 2019-01-14:
@john5788 core 2.3.0 will not be supported in future. As long as it is possible without restrictions to upcoming features Tasmota will support it. For core 2.3.0 and 2.4.2 there will be no more fixes and updates from Arduino ESP team. The core(s) are just used from Tasmota and such issues cant be fixed in Tasmota.
The new upcoming core 2.5.0 seems to work fine

sir106, 2019-01-19:
just flashed the nightly build, and still have the same issues. Seems that there is improvement, but still happens (round about all 3 min). Using SonOFF S20 and T1 2ch, Router is FritzBox 7490.

Program Version	6.4.1.9(b5824fc-sonoff)
Build Date & Time	2019-01-18T12:13:06
Core/SDK Version	2_5_0_BETA2/3.0.0-dev(c0f7b44)

Logfile:
4:44:07 MQT: iot-so-10-t1/stat/STATUS6 = {"StatusMQT":{"MqttHost":"geminipi.fritz.box","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_CA951313","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
14:44:07 MQT: iot-so-10-t1/stat/STATUS7 = {"StatusTIM":{"UTC":"Sat Jan 19 13:44:07 2019","Local":"Sat Jan 19 14:44:07 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"08:35","Sunset":"17:26"}}
14:44:07 MQT: iot-so-10-t1/stat/STATUS10 = {"StatusSNS":{"Time":"2019-01-19T14:44:07"}}
14:44:07 MQT: iot-so-10-t1/stat/STATUS11 = {"StatusSTS":{"Time":"2019-01-19T14:44:07","Uptime":"0T18:07:43","Vcc":3.486,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"BitSchleuder24","BSSId":"xx:xx:xx:xx:xx:xx","Channel":11,"RSSI":66}}}
14:44:18 MQT: Attempting connection...
14:44:22 MQT: Connected
14:44:22 MQT: iot-so-10-t1/tele/LWT = Online (retained)
14:44:22 MQT: iot-so-10-t1/cmnd/POWER = 
14:44:33 MQT: iot-so-10-t1/stat/STATUS = {"Status":{"Module":29,"FriendlyName":["Gartenbrunnen","Zisterne"],"Topic":"iot-so-10-t1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:44:33 MQT: iot-so-10-t1/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff-minimal.bin","RestartReason":"Software/System restart","Uptime":"0T18:08:09","StartupUTC":"2019-01-18T19:36:24","Sleep":50,"BootCount":1817,"SaveCount":2010,"SaveAddress":"FA000"}}
14:44:33 MQT: iot-so-10-t1/stat/STATUS2 = {"StatusFWR":{"Version":"6.4.1.9(b5824fc-sonoff)","BuildDateTime":"2019-01-18T12:13:06","Boot":31,"Core":"2_5_0_BETA2","SDK":"3.0.0-dev(c0f7b44)"}}