arendst/Tasmota #4861: Socket error on client DVES

theflu, 2019-01-08:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : FlashESP8266.exe ESPEasy_mega-20181231
- [x] Hardware used : https://www.amazon.com/gp/product/B075N82CDL w/ bench psu
- [x] If a pre-compiled release or development binary was used, which one? : v6.4.1
- [ ] You have tried latest release or development binaries? :
- [x] Provide the output of command ``status 0`` :
```
19:51:12 MQT: stat/ifan02_1/STATUS = {"Status":{"Module":44,"FriendlyName":["Livingroom-Fan"],"Topic":"ifan02_1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:51:12 MQT: stat/ifan02_1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"External System","Uptime":"0T20:55:38","StartupUTC":"2019-01-07T21:55:34","Sleep":50,"BootCount":30,"SaveCount":107,"SaveAddress":"F9000"}}
19:51:12 MQT: stat/ifan02_1/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-24T14:41:00","Boot":6,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
19:51:12 MQT: stat/ifan02_1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FluFi",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
19:51:12 MQT: stat/ifan02_1/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
19:51:12 MQT: stat/ifan02_1/STATUS5 = {"StatusNET":{"Hostname":"Livingroom-Fan","IPAddress":"192.168.1.51","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"84:0D:8E:42:C9:32","Webserver":2,"WifiConfig":2}}
19:51:12 MQT: stat/ifan02_1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.50","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_42C932","MqttUser":"mqtt_user","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:51:12 MQT: stat/ifan02_1/STATUS7 = {"StatusTIM":{"UTC":"Tue Jan 08 18:51:12 2019","Local":"Tue Jan 08 19:51:12 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:42","Sunset":"17:11"}}
19:51:12 MQT: stat/ifan02_1/STATUS10 = {"StatusSNS":{"Time":"2019-01-08T19:51:12"}}
19:51:12 MQT: stat/ifan02_1/STATUS11 = {"StatusSTS":{"Time":"2019-01-08T19:51:12","Uptime":"0T20:55:38","Vcc":3.436,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","FanSpeed":0,"Wifi":{"AP":1,"SSId":"FluFi","BSSId":"80:2A:A8:54:B0:92","Channel":6,"RSSI":86}}}
```

Other things have noticed
```
19:50:54 MQT: tele/ifan02_1/STATE = {"Time":"2019-01-08T19:50:54","Uptime":"0T20:55:20","Vcc":3.434,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","FanSpeed":0,"Wifi":{"AP":1,"SSId":"FluFi","BSSId":"80:2A:A8:54:B0:92","Channel":6,"RSSI":88}}
19:51:00 WIF: Connecting to AP1 FluFi in mode 11N as Livingroom-Fan...
19:51:04 WIF: Connected
19:51:04 DNS: Initialized
19:51:05 MQT: Attempting connection...
19:51:05 MQT: Connected
19:51:05 MQT: tele/ifan02_1/LWT = Online (retained)
```
```
1546946552: New client connected from 192.168.1.51 as DVES_42C932 (c1, k10, u'mqtt_user').
1546946976: Socket error on client DVES_42C932, disconnecting.
1546946986: New connection from 192.168.1.51 on port 1883.
[INFO] found mqtt_user on local database
1546946986: New client connected from 192.168.1.51 as DVES_42C932 (c1, k10, u'mqtt_user').
```

Device: iFan02
Distance to AP: ~10ft
Broker: Mosquitto HASS addon
Hardware: HASS on rpi3 (hardwired), Edgerouter X, Unifi AC Lite
# Tasmota Devices: 1

ascillato2, 2019-01-08:
Hi,

You are having wifi issues.

Please, in order to solve your issue, check the wiki at: https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#wifi-issues-arduino-core-versions-and-expressif-sdk

ascillato2, 2019-01-08:
If you need further assistant please find us in the Tasmota Chat. Thanks 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-08 by ascillato2