arendst/Tasmota #4862: "lock" mode as opposite to "interlock"

tobby88, 2019-01-08:
With "SetOption14" it is possible to ensure that only one relais at a time is switched on. In my case I have several Shelly 2 devices with Tasmota firmware, where I need something like the opposite: Both relais have to be switched on and off at the same time, so they behave like just one switch for both relais.

Is it possible to add a "lock" mode, where both relais *always* have the same state?

At the moment I try to synchronize them in my OpenHAB installation, but that sometimes gets out of sync when using the switch-inputs of the Shelly.

Thank you for having a look at it :)

Jason2866, 2019-01-08:
Can be achieved via rules. See wiki rules and examples
https://github.com/arendst/Sonoff-Tasmota/wiki/Rules

ascillato2, 2019-01-09:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-09 by ascillato2

tobby88, 2019-01-09:
Ah, overlooked that :) Thank you!

Just a short question: Is this correct?
`backlog rule1 on Power1#State do POWER2 %value% endon on Power2#State do POWER1 %value% endon; rule1 1`

ascillato, 2019-01-09:
That is a loop. You should avoid loops. You can put relay1 like a master and relay2 as a follower and controlling by buttons and mqtt just the relay1:

```
Rule1 1 
Rule1 on Power1#State do POWER2 %value% endon on
```

tobby88, 2019-01-09:
Well... Sometimes "ghost switching" occurs because of induction in my switch input cables. So it could happen that just Relay 2 is switched on, while Relais 1 is switched off and OpenHAB shows "off".