arendst/Tasmota #4864: ADD Rule not working?

At-M, 2019-01-08:
**Describe the bug**
Can't add to a variable, neither a mem.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : Gosund SP111 (Same as the Blitzwolf SHP6)
- [x] _Development/Compiler/Upload tools used_ : Tasmota Console
- [x] _You have tried latest release or development binaries?_ : Latest Release
- [x] _Provide the output of command_``status 0`` :
```
22:07:26 RSL: stat/sonoff/STATUS = {"Status":{"Module":45,"FriendlyName":["BW1"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:07:26 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"External System","Uptime":"0T01:25:06","StartupUTC":"2019-01-08T19:42:20","Sleep":50,"BootCount":15,"SaveCount":137,"SaveAddress":"FB000"}}
22:07:26 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2019-01-08T16:30:36","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
22:07:26 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["GIMMEPRIVACYNET",""],"TelePeriod":300,"SetOption":["00008009","55A180C0","00000000"]}}
22:07:26 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":535,"Free":468,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
22:07:26 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-7202","IPAddress":"192.168.178.51","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"CC:50:E3:26:DC:22","Webserver":2,"WifiConfig":4}}
22:07:26 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_26DC22","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:07:26 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Tue Jan 08 21:07:26 2019","Local":"Tue Jan 08 22:07:26 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:42","Sunset":"17:11"}}
22:07:26 RSL: stat/sonoff/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
22:07:26 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-01-08T22:07:26","ENERGY":{"TotalStartTime":"2019-01-08T17:08:47","Total":0.006,"Yesterday":0.000,"Today":0.006,"Power":1,"ApparentPower":63,"ReactivePower":63,"Factor":0.02,"Voltage":230,"Current":0.274}}}
22:07:26 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-01-08T22:07:26","Uptime":"0T01:25:06","Vcc":3.442,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":56,"POWER":"ON","Wifi":{"AP":1,"SSId":"GIMMEPRIVACYNET","BSSId":"9C:C7:A6:D8:30:0D","Channel":1,"RSSI":100}}}
22:07:28 RSL: tele/sonoff/SENSOR = {"Time":"2019-01-08T22:07:28","ENERGY":{"TotalStartTime":"2019-01-08T17:08:47","Total":0.006,"Yesterday":0.000,"Today":0.006,"Power":5,"ApparentPower":63,"ReactivePower":62,"Factor":0.07,"Voltage":230,"Current":0.273}}
```

**To Reproduce**
Make three Rules like this:

```
Rule1 on ENERGY#Power<7 do backlog var1 = %mem1%; var1;add 1 %1%; mem1 = %var1% endon
Rule2 on mem1=%5% do Rule3 1 endon
Rule3 on Time#Minute|5 do backlog Power 0 ; mem1=%1%; Rule3 0 endon
```

Plug something into the Socket which uses more than 7 watts and turn it on,
On first run, manually set mem1 = 1 and var1 = 0
Enable Rule1 then Rule2 manually

**Expected behavior**
Short:
If the Powerdraw drops below 7 Watts five times, the Socket shuts off.

Long:
(Rule1)
If Power is below 7 Watts:
Move Value from Mem1 to Var1,
Add 1 to the Value of Var1
Move Value from Var1 to Mem1
(Rule2)
If Mem1 has Value of 5, enable Rule 3
(Rule 3)
Each 5 Minutes, Turn off the Power, reset Mem1 to 1 and Disable Rule 3

**Additional context**
Not too sure if this is a bug or my own incompetence though.
Sitting on this problem for several hours now.

I've tried a lot of syntaxes now, can't remember all of them sorry.
But this is what I remember (regarding the "add-part"):

add 1 %1%;
add1 %1%;
add1 1
add1 %1
add1 var1;
add var1
var1 add1

var1++ (well atleast i tried huh..)

ascillato, 2019-01-09:
Please, review your syntax with the information and examples of the wiki.

If you need help with that, please address this to the Tasmota Support Chat. Github issues are for bug reporting. Thanks

ascillato2, 2019-01-09:
### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experiencias. La

closed 2019-01-09 by ascillato2