arendst/Tasmota #4869: RF Bridge not sending received data

ttweb, 2019-01-09:
I have a strange behaviour with two RF Bridges here. First I thought the first bridge module was broken and bought a new one, but it also shows the same thing. With old Firmware, I think it was a 5.14, anything was fine, but now with the new 6.4.1 the bridge module does this:
-> Wifi connection is stable
-> Status updates, last will etc. are send to MQTT broker
-> When triggering a RF-Module (pir-sensors) the red LED on Module turns on, but no message is beeing sent to the MQTT broker. Both modules I own show this behaviour. 

What I did so far:
-> Downgrade to 5.14 via FTDI flashing does not change anyting
-> Changing the environment (other location in the cellar with no other signals than wifi accesspoint and one PIR
-> Flashing different versions of binaries with all different versions of ESP8266/Arduino core 

My last version on the new device is 6.4.1 with Core 2.4.2/2.2.1
What am I doing wrong here? Is it a firmware issue or a problem in front of the desk?
Thanks Thomas

wongnam, 2019-01-09:
FYI. I Sonoff Bridge running properly so far, no issue as you comment. I am running 6.4.1.4 core 2.4.2
Could you provide more info.? or recheck your device console info.

[image]

ascillato2, 2019-01-09:
Hi

I can't reproduce your issue. Sonoff Bridge in latest Tasmota is working fine.

Please, recheck you mqtt config and topics. If after that you still have problems, please erase all flash memory with esptool.py as explained in the and flash again with the precompiled bins from development branch

If you still have issues, please find us in the Tasmota Support Chat. Thanks

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-09 by ascillato2