arendst/Tasmota #4871: Tuya Dimmer: LoadAvg goes slowly up > 400 while powered on

Ysbrand, 2019-01-09:
Hi,

Not sure if this is a bug.
I'm running a Tuya Dimmer with 6.4.1. While the dimmer is on the LoadAvg increases all the time. Once the switch is off The LoadAvg returns back to normal (19).

The dimmer functions normal, so maybe its only a reporting error

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [X ] _Searched the problem in issues and in the wiki_
- [X ] _Hardware used_ : 
- [X ] _Development/Compiler/Upload tools used_ :
- [X ] _If a pre-compiled release or development binary was used, which one?_ : 6.4.1
- [X ] _You have tried latest release or development binaries?_ : 6.4.1
- [ ] _Provide the output of command_``status 0`` :
```
stat/Dimmer1/STATUS {"Status":{"Module":54,"FriendlyName":["Dimmer1"],"Topic":"Dimmer1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
Client mosqsub|13968-domoticz received PUBLISH (d0, q0, r0, m0, 'stat/Dimmer1/STATUS1', ... (288 bytes))
stat/Dimmer1/STATUS1 {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.0.88:81/api/arduino/sonoff-minimal-6.4.1.bin","RestartReason":"Software/System restart","Uptime":"0T15:00:18","StartupUTC":"2019-01-08T18:46:41","Sleep":50,"BootCount":30,"SaveCount":685,"SaveAddress":"F6000"}}
Client mosqsub|13968-domoticz received PUBLISH (d0, q0, r0, m0, 'stat/Dimmer1/STATUS2', ... (126 bytes))
stat/Dimmer1/STATUS2 {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-30T14:06:16","Boot":4,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
Client mosqsub|13968-domoticz received PUBLISH (d0, q0, r0, m0, 'stat/Dimmer1/STATUS3', ... (191 bytes))
stat/Dimmer1/STATUS3 {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":2,"LogHost":"192.168.0.22","LogPort":514,"SSId":["ROUTER","indebuurt2"],"TelePeriod":300,"SetOption":["00120009","558180C0","00000201"]}}
Client mosqsub|13968-domoticz received PUBLISH (d0, q0, r0, m0, 'stat/Dimmer1/STATUS4', ... (200 bytes))
stat/Dimmer1/STATUS4 {"StatusMEM":{"ProgramSize":520,"Free":480,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000809","0FDAE794","240383A1","23B617CE","00003BC0"]}}
Client mosqsub|13968-domoticz received PUBLISH (d0, q0, r0, m0, 'stat/Dimmer1/STATUS5', ... (200 bytes))
stat/Dimmer1/STATUS5 {"StatusNET":{"Hostname":"Dimmer1-0545","IPAddress":"192.168.0.1","Gateway":"192.168.0.253","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"B4:E6:EE:EE:EE:21","Webserver":2,"WifiConfig":3}}
Client mosqsub|13968-domoticz received PUBLISH (d0, q0, r0, m0, 'stat/Dimmer1/STATUS6', ... (186 bytes))
stat/Dimmer1/STATUS6 {"StatusMQT":{"MqttHost":"192.168.0.88","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_50C221","MqttUser":"iot_user","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
Client mosqsub|13968-domoticz received PUBLISH (d0, q0, r0, m0, 'stat/Dimmer1/STATUS7', ... (212 bytes))
stat/Dimmer1/STATUS7 {"StatusTIM":{"UTC":"Wed Jan 09 09:46:59 2019","Local":"Wed Jan 09 10:46:59 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:47","Sunset":"16:43"}}
Client mosqsub|13968-domoticz received PUBLISH (d0, q0, r0, m0, 'stat/Dimmer1/STATUS10', ... (44 bytes))
stat/Dimmer1/STATUS10 {"StatusSNS":{"Time":"2019-01-09T10:46:59"}}
stat/Dimmer1/STATUS11 {"StatusSTS":{"Time":"2019-01-09T10:46:59","Uptime":"0T15:00:18","Vcc":3.459,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Dimmer":21,"Fade":"OFF","Speed":5,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"ROUTER","BSSId":"16:DA:FF:FF:FF:EA","Channel":1,"RSSI":100}}}
```

**To Reproduce**
Switch the dimmer on: LoadAvg goes constantly up
Jan  8 20:21:25 Dimmer1-0545 ESP-MQT: tele/Dimmer1/STATE = {"Time":"2019-01-08T20:21:25","Uptime":"0T00:34:44","Vcc":3.454,"SleepMode":"Dynamic","Sleep":0,"**LoadAvg":188**,"POWER":"ON","Dimmer":34,"Fade":"OFF","Speed":5,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"ROUTER","BSSId":"16:DA:FF:FF:FF:EA","Channel":1,"RSSI":100}}
.....
Jan  8 22:01:25 Dimmer1-0545 ESP-MQT: tele/Dimmer1/STATE = {"Time":"2019-01-08T22:01:25","Uptime":"0T02:14:44","Vcc":3.463,"SleepMode":"Dynamic","Sleep":0,"**LoadAvg":399**,"POWER":"ON","Dimmer":34,"Fade":"OFF","Speed":5,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"ROUTER","BSSId":"16:DA:FF:FF:FF:EA","Channel":1,"RSSI":100}}
.......
Jan  8 22:16:25 Dimmer1-0545 ESP-MQT: tele/Dimmer1/STATE = {"Time":"2019-01-08T22:16:25","Uptime":"0T02:29:44","Vcc":3.460,"SleepMode":"Dynamic","Sleep":50,"**LoadAvg":19**,"POWER":"OFF","Dimmer":23,"Fade":"OFF","Speed":5,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"ROUTER","BSSId":"16:DA:FF:FF:FF:EA","Channel":1,"RSSI":100}}

Switch the dimmer off: LoadAvg will go back to 19

Also, Dynamic sleep is on and will go to "0" when the LoadAvg increases. Once the Dimmer is off, sleep will go back to 50.

**Expected behavior**
I would expect the LoadAvg to be more or less stable

[Dimmer1-out.txt](https://github.com/arendst/Sonoff-Tasmota/files/2740275/Dimmer1-out.txt)

ascillato, 2019-01-09:
Hi, it is expected when the cpu is in use. No issue. Thanks for reporting.

closed 2019-01-09 by ascillato2

Ysbrand, 2019-01-09:
Hi,

Thanks for looking into this?

What I don't understand is:
- The load continues to increase, shouldn't it stabilize at some point? Ans what is the CPU then actually doing, I'm not making any changes to the dimming or whatsoever?
- WHy is this only happnign on the dimmer and not on my other devices?