arendst/Tasmota #4873: MCP230xx consistency with other Tasmota areas.

benzino77, 2019-01-09:
**Have you look for this feature in other issues and in the wiki?**
Yes.

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._
Not really, there is a workaround/another approach. Not so elegant and simple, but it is doable other way.

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
There is some "inconsistency" in Tasmota and the MCP230xx driver.
We have:
```
power1 on
power1 off
switch1
button1
ruletimer1
var1
mem1
etc.
```
numbers starts from "1". In the MCP230xx driver, numbers for pins, starts from "0" (it is completely natural in IT - we start counting from "0", and this is the first element). There is nothing wrong with that, but if there would be consistency one could be able to do some tricky things (less bytes consumed on ruleset):
```
on event#E do 
backlog RuleTimer%value% 120;sensor29 %value%,ON
endon
on rules#timer do 
sensor29 %value%,OFF
endon
```
or using vars holding times for timers (I was surprised, that the embedded construction `%var%value%%` works) 
```
on event#E do
backlog RuleTimer%value% %var%value%%;sensor29 %value%,ON
endon
on rules#timer do
sensor29 %value%, OFF
endon
```
In those examples I start corresponding, to the MCP230xx pin number, timer - this is clearer, simpler, more intuitive (at least for me).

_Of course there is a need to compile Tasmota with proper `MAX_RULE_TIMERS` and/or `MAX_RULE_VARS` to handle 16 relays that way._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
As I wrote above, I have a workaround for that. It is rather things to consider than feature request, bug or issue.

**Additional context**
MCP230xx driver pushed Tasmota to completely new level, I can compare this to rules. Absolutely great job done by Andre. We have a possibility to handle 16 relays or 16 switches or... whatever. Why not to use it  in more consistent manner? ;)

**(Please, remember to close the issue when the problem has been addressed)**
I will or it will be closed right after it was open ;)

ascillato2, 2019-01-09:
I just want to add a side comment: To make work the trigger var1#state there is a byte to control the changes of those vars and another for mems. I put a compiler check that will prevent Tasmota to compile if vars or mems are more than 8. In that case, besides your proposed change, also it is needed to change the config memory map on settings and also all the routines about vars and mem.

benzino77, 2019-01-09:
Correct me is I'm wrong: 8bits for vars and 8bits for mems? 
In which way `var1#state` will work? It will be triggered every time `var1` is changed or it can be also used in that way: `var1#state<100` ?
I'm working on the latest stable v6.4.1 and for now there is no such limitation (max 8 vars and max 8 mems), I can compile Tasmota with defines I need.

jziolkowski, 2019-01-09:
It's because people count relays/switches from 1, and the MCP counts the pins from 0. It's in the datasheets, so it will be much more confusing to explain to users why the datasheet says of A0-A7/B0-B7, and not A1-A8 etc. When you start working with tasmota, you quickly see, learn (and hopefully) understand the devices numbering scheme. But the MCP is a device on its own, so it has full right to have its own numbering scheme.

andrethomas2, 2019-01-09:
[image]

Guess or destiny? :)

closed 2019-01-09 by andrethomas2

benzino77, 2019-01-10:
@andrethomas2  Guess... but it took longer than I expected :) I couldn't find you on Discord, and that's why I opened an issue. I choose the title of the issue to be "clickbait", I hope you don't mind that :)

@jziolkowski I'm not complaining about how Andre implement the driver (as I've wrote above **I do think that this driver is like "Falcon Heavy" for Tasmota**), I'm also not blaming the manufacturer that decided to name PINs A0-A8 and B0-B8 (it's completely natural that one could start counting from "0") I just wanted to point that in some situation It can be convenient to start counting from "1". Probably I've described the use case unclear and that's why you have a problem to understand my point ;)

I was rather thinking on something like SetOption26 for MCP230xx PINs ... default behavior is start numbering PINs from "0" (natural way) and when SetOptionXX is ON, than PINs are numbered from "1" (unnatural way, for special use cases). 

@ascillato Is there a chance for `vars_event` and `mems_event` to be 16bit/32bit long?