arendst/Tasmota #4875: Separating Sonoff T1 button from relay

Hurricane208, 2019-01-09:
I am very new to Tasmota and MQTT so please be kind.

I have a Sonoff T1 (3Ch) successfully flashed to Tasmota. I want to set the device so that "pressing" a button does NOT automatically toggle the relay but creates a MQTT message that I can use elsewhere.

I read some thing similar at #3957 which ascillato was able to help with. I am also reading about SwitchMode but I don't know how this works with the T1.

Help!

ascillato, 2019-01-09:
You can use the solution provided in the referenced issue. If you have questions about that issue, please ask there (in order to not having duplicated issues and make it easy to find for other users) and we can continue the thread if it is needed.

closed 2019-01-09 by ascillato2

Hurricane208, 2019-01-09:
Adrian - thanks. Making progress. 

Alex - See #3957 to see how I am getting on