arendst/Tasmota #4876: Switchdebounce

Lawrencezarb, 2019-01-09:
Is it possible to modify the Arduino sketch to increase the maximun switch debounce option from 1 second to 2 seconds?

ascillato2, 2019-01-09:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-09 by ascillato2

benzino77, 2019-01-10:
@ascillato2 the maximum debounce possible is 1s (that is based on the wiki, I don't know how and if it is checked in the code) but the @Lawrencezarb needs 2s.
@Lawrencezarb try this rule (switchtopic has to be 0):
```java
on switch1#state do
RuleTimer1 2
endon
on rules#timer=1 do
//do your stuff
endon
```
It will work that way:
when switch state will be changing over 2 seconds, the timer will be restarted every change. When the state of the switch will be "stable" for 2s, the timer will fire and do what you want.

ps.
Haven't tested that ;)

Jason2866, 2019-01-10:
Switch debounce >1sec why?  What switch is this?
For providing a propper solution it would be better to describe your complete problem / use case.
For doing this visit us at our Chat.

benzino77, 2019-01-11:
@Jason2866 it has not to be a switch that you can think of. Just give up the idea of using Tasmota with those two:
[image]
[image]
and you will see how deep is a rabbit burrow (and what Tasmota is capable of) ;)
For example, in one place I've seen using Tasmota with this little puppies:
[image]
When the water surface is moving up and down... well you can imagine what happens.

... and please don't take this too serious ;)

Jason2866, 2019-01-11:
Not thinking just in switches and bulbs with Tasmota was the reason asking for the use case...
If switchdebounce with 1 second isnt enough there is probably a other solution needed.

benzino77, 2019-01-12:
Tasmota gives you ability to get to the destination (the behavior/state you expect/need) in many different ways. Start thinking of switch as a **_sensor_**. One second debounce for the switch from the picture above is a nightmare (even if it is possible and doable by Tasmota). If you start thinking of switch as a actuator/sensor one or more seconds is not a big deal.