arendst/Tasmota #4877: Toggle a second Domiticz idx using "Hold"

sammyke007, 2019-01-09:
Hi

I'm a very happy user of your Tasmota firmware, running on all of my Sonoffs.

In my bedroom I toggle the main light with an external wall switch, connected to GPIO14 and set as "Switchmode 5". Idx1 is set to the idx of the bedroom main light in my Domiticz. This works perfectly.

Is it possible to toggle another light (eg. bedrest) when holding the wall switch? 

I suspect I need to use rules?

When holding the wall switch for 4 seconds, I get the HOLD mqtt message, but I want it to switch another Domoticz' idx number. Any suggestions please?

Tnx in advance

ascillato, 2019-01-11:
Hi,

Yes, it is possible and you are right that you will need rules for that. 

Sending messages to domoticz using the command publish was already addressed in issues. Please, search for that.

And for catching the hold trigger for buttons is explained in the wiki at rules examples. 

If you need help putting all that together, please do no hesitate on asking. ;)

andrethomas2, 2019-01-11:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-11 by andrethomas2

sammyke007, 2020-01-14:
> Hi,
>
> Yes, it is possible and you are right that you will need rules for that.
>
> Sending messages to domoticz using the command publish was already addressed in issues. Please, search for that.
>
> And [...]

I'm still trying to make this work... I know... #shame 

I tought of this:

on switch1#state==3 do publish domoticz/in  {"command":"switchlight", "idx":29,"switchcmd":"on"}

But that doesn't seem to work. Can you please help me out?

sammyke007, 2020-01-16:
To answer myself:

First (!!!) check that your Tasmota firmware is at least 6.1.0 (for rules support) 
configure switch1 in configuration (GPIO14 for me on an old Sonoff Basic)
switchtopic has to be 0
switchmode 5 (to get HOLD)
setoption32 40 (press switch 4 seconds for HOLD command)

rule1 on switch1#state==3 do publish domoticz/in {"command":"switchlight","idx":29,"switchcmd":"Toggle"} endon

rule1 1

This enables me to toggle an IDX (eg Light 1) when I press the external switch and toggle another IDX (eg Light 2) when holding the switch for more than 4 seconds.

Working great!