arendst/Tasmota #4878: If i choose PulseTime for relay and sent mqqt "cmnd/POWER ON" to toggle, the mqqt broker store in DataBase the RelayState with disable "PowerRetain 0"

peiperakos, 2019-01-09:
**Describe the bug**
_A clear and concise description of what the bug is._

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x ] _Searched the problem in issues and in the wiki_
- [x ] _Hardware used_ : NodeMCU
- [ x] _Development/Compiler/Upload tools used_ : Arduino IDE
- [x ] _If a pre-compiled release or development binary was used, which one?_ :
- [x ] _You have tried latest release or development binaries?_ : Developer 6.4.1.8
- [ x] _Provide the output of command_``status 0`` :
```
23:42:01 MQT: Home/r00t/stat/STATUS = {"Status":{"Module":18,"FriendlyName":["r00t Home","r00t Home2","r00t Home3","r00t Home4"],"Topic":"r00t","ButtonTopic":"r00t","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"r00t","SwitchMode":[0,0,0,0,2,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:42:01 MQT: Home/r00t/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"External System","Uptime":"0T00:15:47","StartupUTC":"2019-01-09T21:26:14","Sleep":50,"BootCount":74,"SaveCount":710,"SaveAddress":"3F9000"}}
23:42:01 MQT: Home/r00t/stat/STATUS2 = {"StatusFWR":{"Version":"6.4.1.8(sonoff)","BuildDateTime":"2019-01-08T19:37:02","Boot":31,"Core":"2_5_0_BETA2","SDK":"3.0.0-dev(c0f7b44)"}}
23:42:01 MQT: Home/r00t/stat/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["r00t",""],"TelePeriod":10,"SetOption":["00028009","558180C0","00000200"]}}
23:42:01 MQT: Home/r00t/stat/STATUS4 = {"StatusMEM":{"ProgramSize":779,"Free":2292,"Heap":16,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640C8","FlashMode":2,"Features":["00000809","4FFAE79C","000387A0","23B617CC","00003BC0"]}}
23:42:01 MQT: Home/r00t/stat/STATUS5 = {"StatusNET":{"Hostname":"r00t-0000","IPAddress":"192.168.1.102","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:01:94:66:E6:85","Webserver":2,"WifiConfig":2}}
23:42:02 MQT: Home/r00t/stat/STATUS6 = {"StatusMQT":{"MqttHost":"m15.cloudmqtt.com","MqttPort":12271,"MqttClientMask":"r00t-0000","MqttClient":"r00t-0000","MqttUser":"uftnwmep","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
23:42:02 MQT: Home/r00t/stat/STATUS7 = {"StatusTIM":{"UTC":"Wed Jan 09 21:42:02 2019","Local":"Wed Jan 09 23:42:02 2019","StartDST":"Sun Mar 31 03:00:00 2019","EndDST":"Sun Oct 27 04:00:00 2019","Timezone":99,"Sunrise":"07:46","Sunset":"17:11"}}
23:42:02 MQT: Home/r00t/stat/STATUS10 = {"StatusSNS":{"Time":"2019-01-09T23:42:02","Switch5":"OFF","ANALOG":{"A0":0},"AM2301":{"Temperature":20.0,"Humidity":50.6},"BMP180":{"Temperature":18.6,"Pressure":999.5},"PressureUnit":"hPa","TempUnit":"C"}}
23:42:02 MQT: Home/r00t/stat/STATUS11 = {"StatusSTS":{"Time":"2019-01-09T23:42:02","Uptime":"0T00:15:48","SleepMode":"Dynamic","Sleep":50,"LoadAvg":40,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","Wifi":{"AP":1,"SSId":"r00t","BSSId":"78:96:82:32:BE:D6","Channel":11,"RSSI":60}}}
23:42:06 MQT: Home/r00t/tele/STATE = {"Time":"2019-01-09T23:42:06","Uptime":"0T00:15:52","SleepMode":"Dynamic","Sleep":50,"LoadAvg":34,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","Wifi":{"AP":1,"SSId":"r00t","BSSId":"78:96:82:32:BE:D6","Channel":11,"RSSI":48}}
23:42:06 MQT: Home/r00t/tele/SENSOR = {"Time":"2019-01-09T23:42:06","Switch5":"OFF","ANALOG":{"A0":0},"AM2301":{"Temperature":20.0,"Humidity":50.6},"BMP180":{"Temperature":18.6,"Pressure":999.5},"PressureUnit":"hPa","TempUnit":"C"}
```

**To Reproduce**
i use cloudmqqt.com as mqqt broker!
I have choise the relay4 as PulseTime with command "PulseTime4 30" for 3 sec.
i have disable "PowerRetain 0"

if i hit to toggle the POWER4 in WEBSERVER the toggle for 3 sec works perfect and send to mqqt broker the correct massage and works! and the mqqt broker DONT store the PowerState of POWER4

if i hit to toggle the POWER4 with my mobile app mqqt dashbord and sent the commant "Home/r00t/cmnd/POWER4 ON" works perfect and send to mqqt broker the correct massage and works!
BUT the mqqt broker for unexpected reason store the PowerState of POWER4 with this command "Home/r00t/cmnd/POWER4 | ON" and if i restart my NodeMCU Tasmota ofcorse the POWER4 toggle itself because the mqqtcloud send this command "Home/r00t/cmnd/POWER4 | ON"

**Expected behavior**
why in WebServer works GREAT without STORE the PowerRetain in DataBase
and in mqqt command stored in DataBase?

*** i have onother 3 POWER output without PulseTime and for this POWER output work Great from my APP and from Webserver without store the the PowerRetain in DataBase of mqqtcloud!

andrethomas2, 2019-01-10:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-10 by andrethomas2

peiperakos, 2019-01-10:
Ohhh you have right!! I forgote the settings in my app so many sorry! Thenx for your help