arendst/Tasmota #4880: How to connect multiple DS18B20 to Sonoff TH16

rikray02, 2019-01-10:
Hello,

I read in closed issue that it's possible to connect multiple DS18B20 to Sonoff TH10/16.  So, i don't know how.  Does it's possible to explain how can i do that?  Can i connect many DS18B20 to the same GPIO14 (external connector) or i need to use other one (GPIO4)?

Thank a lot for your help.
Richard

edelstahlratte, 2019-01-10:
They share one single-wire bus... See [https://github.com/arendst/Sonoff-Tasmota/wiki/Sensor-Configuration#single-wire-sensor](url) and [https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs](url) to get an idea... It's simple...

Nico.

Jason2866, 2019-01-10:
You have to connect like this
https://www.microcontroller-project.com/uploads/2/2/1/5/22159166/nodemcu-esp8266-with-multiple-ds18b20-sensors_orig.png

closed 2019-01-10 by andrethomas2