arendst/Tasmota #4883: The standard configuration for MagicLed is not working

MarScheff, 2019-01-10:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
The standard configuration for MagicHome Led Controller Is not working. Command Color 1 will result in no light, Color 2 will be red, color 3 will be green, color 4 will show no color

- [X] _Searched the problem in issues and in the wiki_
- [X] _Hardware used_ : MagicHome LED controllers Type RGB
- [X] _Development/Compiler/Upload tools used_ : Atom
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [ ] _You have tried latest release or development binaries?_ : 
- [ ] _Provide the output of command_``status 0`` :
```
20:44:32 MQT: stat/MagicLed/STATUS = {"Status":{"Module":18,"FriendlyName":["MagicLed"],"Topic":"MagicLed","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:44:32 MQT: stat/MagicLed/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"1T20:44:56","StartupUTC":"2019-01-08T22:59:36","Sleep":0,"BootCount":19,"SaveCount":286,"SaveAddress":"F6000"}}
20:44:32 MQT: stat/MagicLed/STATUS2 = {"StatusFWR":{"Version":"6.3.0.6(sonoff)","BuildDateTime":"2019.01.08 22:31:39","Boot":7,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
20:44:32 MQT: stat/MagicLed/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SmartHome",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
20:44:32 MQT: stat/MagicLed/STATUS4 = {"StatusMEM":{"ProgramSize":493,"Free":508,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000407","0FDAE794","240183A0","23B617CE","00003BC0"]}}
20:44:32 MQT: stat/MagicLed/STATUS5 = {"StatusNET":{"Hostname":"MagicLed-6852","IPAddress":"X.X.X.X","Gateway":"X.X.X.X","Subnetmask":"255.255.255.0","DNSServer":"X.X.X.X","Mac":"XX:XX:XX:XX:XX","Webserver":2,"WifiConfig":5}}
20:44:32 MQT: stat/MagicLed/STATUS6 = {"StatusMQT":{"MqttHost":"X.X.X.X","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_DDFAC4","MqttUser":"XXXXX","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:44:32 MQT: stat/MagicLed/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 10 19:44:32 2019","Local":"Thu Jan 10 20:44:32 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:41","Sunset":"17:13"}}
20:44:32 MQT: stat/MagicLed/STATUS10 = {"StatusSNS":{"Time":"2019-01-10T20:44:32"}}
20:44:32 MQT: stat/MagicLed/STATUS11 = {"StatusSTS":{"Time":"2019-01-10T20:44:32","Uptime":"1T20:44:56","Vcc":3.201,"POWER":"OFF","Dimmer":100,"Color":"FFA04200","HSBColor":"30,74,100","Channel":[100,63,26,0],"Scheme":0,"Fade":"OFF","Speed":20,"LedTable":"ON","Wifi":{"AP":1,"SSId":"XXXX","BSSId":"XXXXX","Channel":6,"RSSI":52}}}
```
**To Reproduce**
_Steps to reproduce the behavior:_
* Connect LED
* Connect Power
* Change color via console

**Expected behavior**
_A clear and concise description of what you expected to happen._
Show all the colors correct.

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
I did find a workaround, so that all the colors are working
[image]

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-01-10:
Glad you find a working config for your MagicHome module
Since there are many different MagicHome Modules from different manufactors there isnt the "one" correct config 😕 So Tasmota cant handle every possible config because device resources are very limited

andrethomas2, 2019-01-10:
Closing this issue as it has been answered - please add it to the Wiki if you have the time to do so.

Thanks!

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-10 by andrethomas2