arendst/Tasmota #4888: disable `Pressing the button for over forty seconds`

devanl, 2019-01-11:
**Have you look for this feature in other issues and in the wiki?**
yes

**Is your feature request related to a problem? Please describe.**
Light-switch in pantry often finds a box pushed against it for more than 40 secs causing config to clear.

**Describe the solution you'd like**
Option to disable `Pressing the button for over forty seconds` 

**Describe alternatives you've considered**
maybe some more animated gesture?
Physically applying border to switch.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-01-11:
Hi, it is already implemented, please read the wiki at commands. Thanks

closed 2019-01-11 by ascillato2

devanl, 2019-01-11:
I read through it again, it doesn’t look to me like there is an obvious way to disable... I’ll try to find more on Discord.

ascillato, 2019-01-11:
Search for disable multipress. It is a setoption command

devanl, 2019-01-11:
Sorry to keep driving on this, just want to make sure I'm getting it correct and see if I can help people who might have the same question later.

This is what I found in wiki:

SetOption1 |   | Show current button multi press mode
-- | -- | --
SetOption1 | 0 / off | (default) Allow all button actions
SetOption1 | 1 / on | Allow only single, double and **hold press** button actions

Are you saying that this means that (long) hold press is disabled?

ascillato, 2019-01-11:
Please, go to the Tasmota console and type:

```
Setoption1 1
```

And press enter. 

So the 40 seconds to reset will be disabled.