arendst/Tasmota #4890: Lot of Outages

MaxBec, 2019-01-11:
Hey guys,

since i updated to 6.4.1 i have a lot of unreachable Sonoff devices.
I compiled with 2.3.0 and TasmotaMQTT.

[image]

Do you have any idea on how to reduce those outages? I already tried 2.4.2 and 2.5.0... they were even worse.

Thanks.

Jason2866, 2019-01-11:
First fill out issue template. No template - No help (possible)
Read:
https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting
https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#wifi-issues-arduino-core-versions-and-expressif-sdk

MaxBec, 2019-01-11:
I will try the 2.5.0 again.
But are there any recommendations for which MQTT version to use?
Tasmota, PubSub, etc.?

Jason2866, 2019-01-11:
`#define MQTT_LIBRARY_TYPE      MQTT_PUBSUBCLIENT   // Use PubSubClient library`

ascillato, 2019-01-11:
Please, use the precompiled bins to solve your issue as explained in the wiki at 

https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#wifi-issues-arduino-core-versions-and-expressif-sdk

closed 2019-01-11 by ascillato2