arendst/Tasmota #4894: Led not blinking if not connected to wifi/mqtt if using LedPower

loku02, 2019-01-11:
I'm sorry if this is a known topic. I was looking for an answer quite a while and no luck.
I'm using Tasmota with Sonoff Basic.
I want Sonoff LED to be on, every time the power is on. However I'd like to preserve LED blinking when not connected to wifi/mqtt.
If I set LedState 1, LED is blinking until connectef to wifi and mqtt, but the LED goes off, when the Sonoff is switched off.
If I set LedPower 1, LED stays on all the time Sonoff has the power, but I'm loosing LED blinking is not connected to wifi/mqtt.
Is there a way to combine both: blink the LED if wifi/mqtt not connected, but keep the LED on all the time if both connected, no matter on the switch state?

ascillato, 2019-01-11:
> Is there a way to combine both: blink the LED if wifi/mqtt not connected, but keep the LED on all the time if both connected, no matter on the switch state?

Yes, it is possible with rules. Please, check the wiki at rules and also at commands.

If you need further assistance with this, please, find us in the Tasmota support chat. Thanks.

closed 2019-01-11 by ascillato2

ramanraja, 2021-03-03:
I have a much simpler requirement, but faced similar issue with Link LED.

Problem:
The Link LED is not blinking when WiFi or MQTT connectivity is lost. The relevant settings seem to be correct.

Experiment:

Take the MQTT broker offline; Tasmota is unable to connect to MQTT. Now GET the status of ledpower and setoption31 :

$ Backlog LedPower; SetOption31 

20:25:07 RSL: stat/mytopic/RESULT = {"LedPower1":"OFF"}
20:25:07 RSL: stat/mytopic/RESULT = {"SetOption31":"OFF"}

Both the settings seem to be correct, but the Link LED is not blinking !

Solution:
EXPLICITLY SET both ledpower and setoption31 to 0 again, manully:

19:53:56 CMD: Backlog LedPower 0; SetOption31 0
19:53:56 MQT: stat/mytopic/RESULT = {"LedPower1":"OFF"}
19:53:56 MQT: stat/mytopic/RESULT = {"SetOption31":"OFF"}

Now the Link LED starts blinking whenever wifi or MQTT connectivity is lost.

ascillato, 2021-03-03:
https://github.com/arendst/Tasmota/issues/6244