arendst/Tasmota #4896: Intermittent Wifi connection

tomlut, 2019-01-12:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
_A clear and concise description of what the bug is._
Just upgraded a bunch of POW 2, Basic and S23 Sonoffs from 6.2.1. to 6.4.1. Since upgrading all devices are intermittently becoming unavailable on the network.

All have better than 80% signal strength all have fixed IP addresses. 

Downgrading back to 6.2.1 seems to have fixed the issue.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : Sonoff S23, Basic, POW 2
- [x] _Development/Compiler/Upload tools used_ : OTA update
- [x] _If a pre-compiled release or development binary was used, which one?_ : sonoff.bin
- [x] _You have tried latest release or development binaries?_ : yes 6.4.1
- [x] _Provide the output of command_``status 0`` : 
```
16:59:04 MQT: stat/sonoff_lounge_dehumidifier/STATUS = {"Status":{"Module":43,"FriendlyName":["Lounge Dehumidifier"],"Topic":"sonoff_lounge_dehumidifier","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":1}}
16:59:04 MQT: stat/sonoff_lounge_dehumidifier/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://10.1.1.100:9541/data/firmwares/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:10:29","StartupUTC":"2019-01-12T05:48:35","Sleep":50,"BootCount":79,"SaveCount":389,"SaveAddress":"FA000"}}
16:59:04 MQT: stat/sonoff_lounge_dehumidifier/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-09T16:50:26","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
16:59:04 MQT: stat/sonoff_lounge_dehumidifier/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WAPLO",""],"TelePeriod":300,"SetOption":["00028029","558180C0","00000000"]}}
16:59:04 MQT: stat/sonoff_lounge_dehumidifier/STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":15,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","00000000"]}}
16:59:04 MQT: stat/sonoff_lounge_dehumidifier/STATUS5 = {"StatusNET":{"Hostname":"sonoff_lounge_dehumidifier","IPAddress":"10.1.1.185","Gateway":"10.1.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.1.1.1","Mac":"80:7D:3A:17:E3:02","Webserver":2,"WifiConfig":1}}
16:59:04 MQT: stat/sonoff_lounge_dehumidifier/STATUS6 = {"StatusMQT":{"MqttHost":"10.1.1.100","MqttPort":1883,"MqttClientMask":"sonoff_lounge_dehumidifier","MqttClient":"sonoff_lounge_dehumidifier","MqttUser":"mqtt_user","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
16:59:04 MQT: stat/sonoff_lounge_dehumidifier/STATUS7 = {"StatusTIM":{"UTC":"Sat Jan 12 05:59:04 2019","Local":"Sat Jan 12 16:59:04 2019","StartDST":"Sun Oct 06 02:00:00 2019","EndDST":"Sun Apr 07 03:00:00 2019","Timezone":99,"Sunrise":"18:40","Sunset":"03:16"}}
16:59:04 MQT: stat/sonoff_lounge_dehumidifier/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
16:59:04 MQT: stat/sonoff_lounge_dehumidifier/STATUS10 = {"StatusSNS":{"Time":"2019-01-12T16:59:04","ENERGY":{"Total":28.514,"Yesterday":0.004,"Today":0.000,"Power":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
16:59:04 MQT: stat/sonoff_lounge_dehumidifier/STATUS11 = {"StatusSTS":{"Time":"2019-01-12T16:59:04","Uptime":"0T00:10:29","Vcc":3.238,"POWER":"OFF","Wifi":{"AP":1,"SSId":"WAPLO","RSSI":84,"APMac":"FC:EC:DA:38:94:62"}}}
```

**To Reproduce**
_Steps to reproduce the behavior:_
 Install v6.4.1

**Expected behavior**
_A clear and concise description of what you expected to happen._
As reliable network connection as I have using v6.2.1

**Screenshots**
_If applicable, add screenshots to help explain your problem._
N/A

**Additional context**
_Add any other context about the problem here._
N/A

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-01-12:
Take a look here
https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#wifi-issues-arduino-core-versions-and-expressif-sdk

ascillato2, 2019-01-12:
Closing this issue as it is duplicated. 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-12 by ascillato2

tomlut, 2019-01-12:
Huh? 

So you are going to just sweep this legitimate issue under the carpet rather than fixing the firmware?

This does not explain why reverting to 6.2.1 solves the issue.

Way to bury your head in the sand.