arendst/Tasmota #4898: Device switches itself off after some time

stefaanv, 2019-01-12:
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Development/Compiler/Upload tools used : **uploaded the stock image using ESPtool.py**
- [X] Hardware used : **iTead Sonoff RF**
- [X] If a pre-compiled release or development binary was used, which one? : **sonoff.bin**
- [X] You have tried latest release or development binaries? : **Yes**
- [X] Provide the output of command ``status 0`` :
```
14:57:27 CMD: status 0
14:57:27 MQT: home/kerstboom/STATUS = {"Status":{"Module":2,"FriendlyName":["Sonoff"],"Topic":"home/kerstboom/sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:57:27 MQT: home/kerstboom/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T23:54:52","StartupUTC":"2019-01-11T14:02:35","Sleep":50,"BootCount":24,"SaveCount":180,"SaveAddress":"F8000"}}
14:57:27 MQT: home/kerstboom/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-24T14:41:00","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
14:57:27 MQT: home/kerstboom/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SVDV",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
14:57:27 MQT: home/kerstboom/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
14:57:27 MQT: home/kerstboom/STATUS5 = {"StatusNET":{"Hostname":"sonoff_kerstboom","IPAddress":"192.168.0.29","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"84:0D:8E:46:8E:E9","Webserver":2,"WifiConfig":2}}
14:57:27 MQT: home/kerstboom/STATUS6 = {"StatusMQT":{"MqttHost":"home","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_468EE9","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
14:57:27 MQT: home/kerstboom/STATUS7 = {"StatusTIM":{"UTC":"Sat Jan 12 13:57:27 2019","Local":"Sat Jan 12 14:57:27 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:40","Sunset":"17:16"}}
14:57:27 MQT: home/kerstboom/STATUS10 = {"StatusSNS":{"Time":"2019-01-12T14:57:27"}}
14:57:27 MQT: home/kerstboom/STATUS11 = {"StatusSTS":{"Time":"2019-01-12T14:57:27","Uptime":"0T23:54:52","Vcc":3.438,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"SVDV","BSSId":"9C:C7:A6:D1:6E:69","Channel":13,"RSSI":64}}}
```

Hi, I have a setup of **one** iTead Sonoff RF device on which I uploaded the stock sonoff.bin firmware downloaded from this website.  I managed to set up bi-directional control to openhab2 over MQTT meaning I can control the device from the openHAB app **and** when the device is actuated with the local pushbutton, the state is reflected in the openHAB app.

When the device is turned on from openhab, it stays on as expected, 
The problem is that when it is controlled locally, it will always switch off after some random time.
When I remove the configuration for this device from openHAB, the problem remains.

From studying the mosquitto broker logs, I found that the unsollicited turnoff always happens after the sonoff-tasmota device sends out **LWT** messages followed by a **POWER=&lt;null&gt;** message.  Here's an extract from the associated sonoff device's log, the associated (less clear) mosquitto log is further down.
```
15:04:56 MQT: home/kerstboom/POWER = ON (retained)
15:09:01 MQT: home/kerstboom/STATE = {"Time":"2019-01-12T15:09:01","Uptime":"1T00:06:26","Vcc":3.457,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"SVDV","BSSId":"9C:C7:A6:D1:6E:69","Channel":13,"RSSI":70}}
15:12:39 DNS: Initialized
15:12:39 MQT: Attempting connection...
15:12:39 MQT: Connected
15:12:39 MQT: home/kerstboom/LWT = Online (retained)
15:12:39 MQT: home/kerstboom/cmnd/POWER = 
15:12:39 MQT: home/kerstboom/RESULT = {"POWER":"OFF"}
15:12:39 MQT: home/kerstboom/POWER = OFF (retained)
```
The **POWER=&lt;null&gt;** message is bounced by the MQTT broker and makes the device turn off.  
The problem seems to be initiated by a (DNS?) re-initialisation of the sonoff-tasmota device.  I understand from [this issue](https://github.com/arendst/Sonoff-Tasmota/issues/890) that the POWER=&lt;null&gt; is sent to satisfy ioBroker, but it seems to cause this problems in my simple setup.  Is there a way to prevent the sonoff-tasmota device from sending out this POWER=&lt;null&gt;  message ?
```
1547301897: Received PUBLISH from DVES_468EE9 (d0, q0, r1, m0, 'home/kerstboom/POWER', ... (2 bytes))
1547302141: Received PUBLISH from DVES_468EE9 (d0, q0, r0, m0, 'home/kerstboom/STATE', ... (204 bytes))
1547302360: Received PUBLISH from DVES_468EE9 (d0, q0, r1, m0, 'home/kerstboom/LWT', ... (6 bytes))
1547302360: Received PUBLISH from DVES_468EE9 (d0, q0, r0, m0, 'home/kerstboom/cmnd/POWER', ... (0 bytes))
1547302360:     home/kerstboom/cmnd/# (QoS 0)
1547302360: DVES_468EE9 0 home/kerstboom/cmnd/#
1547302360: Sending PUBLISH to DVES_468EE9 (d0, q0, r1, m0, 'home/kerstboom/cmnd/POWER', ... (3 bytes))
1547302360: Received PUBLISH from DVES_468EE9 (d0, q0, r0, m0, 'home/kerstboom/RESULT', ... (15 bytes))
1547302360: Received PUBLISH from DVES_468EE9 (d0, q0, r1, m0, 'home/kerstboom/POWER', ... (3 bytes))
1547302441: Received PUBLISH from DVES_468EE9 (d0, q0, r0, m0, 'home/kerstboom/STATE', ... (205 bytes))
```
My problem is similar but not equal to these problems.

https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#disconnects-and-reconnects

https://github.com/arendst/Sonoff-Tasmota/issues/890

Just for clarity, the openHAB2 config line below 
```
Switch vlg_kerstboom {mqtt="<[mosquitto:home/kerstboom/POWER:state:default], >[mosquitto:home/kerstboom/cmnd/POWER:command:*:default]"}

```

Jason2866, 2019-01-12:
You have retaining messages in your broker. Search for deleting messages in mosquitto 
Tasmota works perfect with Open Hab (i use it too...)
```
Switch LivingRoom_Light "Living Room Light" <light> (LR,gLight)
    { mqtt=">[broker:cmnd/sonoff-A00EEA/POWER:command:*:default],
            <[broker:stat/sonoff-A00EEA/POWER:state:default]" }
```

andrethomas2, 2019-01-12:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-12 by andrethomas2

stefaanv, 2019-01-14:
OK, clearing the retained messages from Mosquitto seems to have done the trick.
Thanks for the help !!