arendst/Tasmota #4900: Restart if failed ping.

OfferLa, 2019-01-12:
A nice feature I've seen in dedicated PDUs, is the ability to ping a specific address (IP or URI), and if ping fails - to "restart" the connected device (means - turn off the power for some seconds, and then turn in back on).

In case of router or network device that is connected to the power line - such a simple action can save lot of troubles.

What do you think?

ascillato, 2019-01-12:
Hi,

Sorry, duplicated issue: https://github.com/arendst/Sonoff-Tasmota/issues/3075

Please, check that. Thanks.

ascillato2, 2019-01-12:
Closing this issue as it is duplicated.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-12 by ascillato2

OfferLa, 2019-01-12:
I'm sorry if I'm wrong - but it seems to me to be little funny - to reference a 'closed' issue (that wasn't solved) as a reason to close this also...

I've commented there also. Would you PLEASE re-consider ignoring this feature request?

Thanks!

ascillato2, 2019-01-12:
@OfferLa 

Hi,

Sorry. Don't take it personal. The request was dismissed before, by the Owner of Tasmota (Theo) and also was provided another solution for this specific use-case.

This is a open-software, so you can fork it and develop on top of it the exact solution you want, if the provided workaround don't satisfy you. Then, if you want, you can make a pull request with the modifications of the code if you are willing to share it.

Please, again do not take it personal. These are just ideas and code been shared. If the proposed idea by you is not taken into account is not something against you. The owner has the final decision over his own software. We are not related or sponsored by any company. We (the community) are just helping here as a hobby and for free.

OfferLa, 2019-01-12:
@ascillato2 
Thank you very much for clarification. Totally understandable.

Would you please refer me to the 'workaround' to get this done? I didn't see it there.

Thanks again!

ascillato2, 2019-01-12:
The proposed workaround in that issue was:

Theo said (https://github.com/arendst/Sonoff-Tasmota/issues/3075#issuecomment-400701306):

> You might want to run a small node-red service on a your HA server pinging the internet every minute. When no reply is received let it toggle a tasmota driven device controlling your DSL router.

Frogmore42 said (https://github.com/arendst/Sonoff-Tasmota/issues/3075#issuecomment-400703870):

> If you want something configurable, you can use NodeRED or perhaps some other HA software to get exactly the behavior you want. This would be something you could implement today. It would also help you understand exactly what behavior you need.

Andre said (https://github.com/arendst/Sonoff-Tasmota/issues/3075#issuecomment-401282314):

> I would run node-red on a pi right next to the router connected with physical cable... when node red cannot ping external ip address like google you can use one of the GPIO's to remove power from the router using a relay breakout board for a few seconds and turn it back on.
> 
> So basically as soon as node-red does not get ping returns from a remote host it will turn power (the 9v or 12v cable through a relay) off for a few seconds and then on again.

And finally adamrhoades (the user that open that issue, conclude - https://github.com/arendst/Sonoff-Tasmota/issues/3075#issuecomment-400709216):

> Useful feedback, thank you. I had considered automating this with my HA system, but I've been using Samsung Smartthings which is pretty much useless if you loose connectivity.
> 
> I'll have a look at NodeRed for this task and see what I can come up with.
>