arendst/Tasmota #4901: PWM for 3 channels via MQTT at same time

egs2003, 2019-01-12:
I need to submit pwm values via mqtt at the same time. If I use the backlog topic, there is a delay between each command.

egs2003, 2019-01-12:
But I can see it. I use PWM1 to 3 for the colors RGB. If I use the backlog I can see each setting of the color. For example the LED stripe is off. So I send via backlog all three PWM RGB values (1023). The LED stripe will show first red then yellow and then white. I dont'T want to use the color comand cause of the rounding failure.

ascillato2, 2019-01-12:
Ok, sorry. That is not supported.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-12 by ascillato2

egs2003, 2019-01-12:
this was an feature request!