arendst/Tasmota #4902: Time cicling

Valerio1975, 2019-01-12:
**Have you look for this feature in other issues and in the wiki?**

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

Valerio1975, 2019-01-12:
Sorry, I do not know English well and if the request has already been made I'm sorry (please tell me the link of the discussion).
I have a SonOff Basic RF connected to an attic fan that I would have to switch on at 4 pm and switch off at 10 pm until here everything can be done.
I should do a cycle that turns on at 22:00 and finally turns off at 07:00 (the day after) doing a cycle on off (turns on for 30 minutes and goes off for 15 minutes) choosing the days of the week and several hours per day, can you implemetate or is there already but I do not see it?

ascillato2, 2019-01-12:
Hi,

It is already implemented. In the Tasmota web menu, go to TIMERS CONFIGURATION and set there everything you need.

If you need further assistance, please find us in the Tasmota Support Chat. Thanks

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-12 by ascillato2

Valerio1975, 2019-01-12:
Grazie ma non funziona la chat. Ma non funziona neppure un semplice on off.
00:00:00 Project sonoff Ventilatore Version 6.4.1.8(sonoff)-2_5_0_BETA2
00:00:00 WIF: Connecting to AP1 Telecom-66782150 in mode 11N as sonoff-0288...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on sonoff-0288 with IP address 192.168.1.147
19:15:46 RSL: tele/sonoff/STATE = {"Time":"2019-01-12T19:15:46","Uptime":"0T00:04:59","Vcc":3.466,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"Telecom-66782150","BSSId":"A4:B1:E9:F0:E4:C6","Channel":11,"RSSI":88}}
19:20:46 RSL: tele/sonoff/STATE = {"Time":"2019-01-12T19:20:46","Uptime":"0T00:09:59","Vcc":3.466,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"Telecom-66782150","BSSId":"A4:B1:E9:F0:E4:C6","Channel":11,"RSSI":86}}
19:20:47 APP: Serial logging disabled
19:25:46 RSL: tele/sonoff/STATE = {"Time":"2019-01-12T19:25:46","Uptime":"0T00:14:59","Vcc":3.466,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"Telecom-66782150","BSSId":"A4:B1:E9:F0:E4:C6","Channel":11,"RSSI":86}}
19:30:46 RSL: tele/sonoff/STATE = {"Time":"2019-01-12T19:30:46","Uptime":"0T00:19:59","Vcc":3.465,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"Telecom-66782150","BSSId":"A4:B1:E9:F0:E4:C6","Channel":11,"RSSI":86}}
19:35:46 RSL: tele/sonoff/STATE = {"Time":"2019-01-12T19:35:46","Uptime":"0T00:24:59","Vcc":3.464,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"Telecom-66782150","BSSId":"A4:B1:E9:F0:E4:C6","Channel":11,"RSSI":92}}
19:39:03 RSL: stat/sonoff/RESULT = {"POWER":"OFF"}
19:39:03 RSL: stat/sonoff/POWER = OFF
19:40:46 RSL: tele/sonoff/STATE = {"Time":"2019-01-12T19:40:46","Uptime":"0T00:29:59","Vcc":3.466,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Telecom-66782150","BSSId":"A4:B1:E9:F0:E4:C6","Channel":11,"RSSI":84}}
19:45:46 RSL: tele/sonoff/STATE = {"Time":"2019-01-12T19:45:46","Uptime":"0T00:34:59","Vcc":3.466,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Telecom-66782150","BSSId":"A4:B1:E9:F0:E4:C6","Channel":11,"RSSI":88}}
19:50:46 RSL: tele/sonoff/STATE = {"Time":"2019-01-12T19:50:46","Uptime":"0T00:39:59","Vcc":3.466,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Telecom-66782150","BSSId":"A4:B1:E9:F0:E4:C6","Channel":11,"RSSI":84}}
19:55:46 RSL: tele/sonoff/STATE = {"Time":"2019-01-12T19:55:46","Uptime":"0T00:44:59","Vcc":3.464,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Telecom-66782150","BSSId":"A4:B1:E9:F0:E4:C6","Channel":11,"RSSI":84}}
20:00:46 RSL: tele/sonoff/STATE = {"Time":"2019-01-12T20:00:46","Uptime":"0T00:49:59","Vcc":3.466,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Telecom-66782150","BSSId":"A4:B1:E9:F0:E4:C6","Channel":11,"RSSI":86}}
20:02:00 RSL: tele/sonoff/UPTIME = {"Time":"2019-01-12T20:02:00","Uptime":"0T00:51:13"}
20:05:46 RSL: tele/sonoff/STATE = {"Time":"2019-01-12T20:05:46","Uptime":"0T00:54:59","Vcc":3.458,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Telecom-66782150","BSSId":"A4:B1:E9:F0:E4:C6","Channel":11,"RSSI":86}}

Come ho detto non conosco l'inglese e quindi se mi dite quale sia la la procedura da fare mi fate un piacere, va bene anche il link dove viene spiegato quello che voglio fare.
Ma ripeto ho impostato semplicemente on time on off e non funziona.

ascillato, 2019-01-12:
é vero. Discord non funciona in questo momento. 

Per favore, un screenshoot di la tua configurazione di TIMERS. Chè un quadro per fare click per attivare tutti i TIMERS. Le ha fatto click?

ascillato, 2019-01-12:
anche convidi lo che Tasmota dice in la consola quando fa il commando `STATUS 0`

Valerio1975, 2019-01-12:
Grazie mille ora l'on off normale funziona.
Ma se volessi fare un ciclo continuo dalle 22 alle 7 che stia su on per 30 minuti e off per 15 minuti e spengersi alle 7
La differenza tra ARM e REPEAT?

Valerio1975, 2019-01-12:
Grazie per il BIN ma da errore
Upload Failed
Upload buffer miscompare

ascillato, 2019-01-12:
> La differenza tra ARM e REPEAT?
ARM: attivare il timer
REPEAT: per repettire il timer. Si non si fa un click supra REPEAT il TIMER funziona solo una volta
ENABLE TIMERS: disattivazione globale di tutti i TIMERS

> Grazie per il BIN ma da errore
> Upload Failed
> Upload buffer miscompare

Devi ussare primo il BIN che si chiama MINIMAL per liberare memoria e dopo il BIN sonoff-IT.bin

> Ma se volessi fare un ciclo continuo dalle 22 alle 7 che stia su on per 30 minuti e off per 15 minuti e spengersi alle 7

ah, quello è altra cosa. Devi usare RULES. Ti lo scrivo in cinque minuti.

Valerio1975, 2019-01-12:
Ok grazie mille, molto gentile.
Aspetto questo RULES, che non saprei neppure dove sia e come si fa.

ascillato, 2019-01-12:
> Ma se volessi fare un ciclo continuo dalle 22 alle 7 che stia su on per 30 minuti e off per 15 minuti e spengersi alle 7

In TIMERS CONFIGURATION configura:

* TIMER1 alle 22 per accendere RULE
* TIMER2 alle 7 per accendere RULE

Dopo, nella Consola di Tasmota scrivi:

```
Rule1 1
Rule1 on Clock#Timer=1 do backlog power1 1; ruletimer1 1800 endon on Rules#Timer=1 do backlog power1 0; ruletimer2 900 endon on Rules#Timer=2 do backlog backlog power1 1; ruletimer1 1800 endon on Clock#Timer=2 do backlog ruletimer1 0; ruletimer2 0; endon
```

Spiegazione:

1- `Clock#Timer=1` si attiva quando il tuo TIMER1 è attivato alle 22. In quello momento il RELAY 1 se attiva e anche un RULETIMER1 che è un contatore.
2- Dopo 30 minuti il contatore RULETIMER1 è attivato (Rules#Timer=1), il RELAY1 si spegnere e anche un altro contatore si attiva (RULETIMER2)
3- Dopo 15 minuti il contatore RULETIMER2 è attivato (Rules#Timer=2), il RELAY1 si attiva e anche di nuevo il contatore RULETIMER1. Dopo il paso 2 è ripetuto.
4- Questo ciclo si ferma quando il tuo TIMER2 é attivato alle 7 (Clock#Timer=2)

Valerio1975, 2019-01-12:
Grazie mille davvero molto gentile

Valerio1975, 2019-01-12:
Grazie davvero ma non mi funziona. ho messo i contatori tutti a 60 (1 minuto). ma si accende ae si spenge seguendo on off dei timer impostati.

Rule1 1
Rule1 on Clock#Timer=1 do backlog power1 1;
ruletimer1 60 endon on Rules#Timer=1 do backlog power1 0;
ruletimer2 60 endon on Rules#Timer=2 do backlog backlog power1 1;
ruletimer1 60 endon on Clock#Timer=2 do backlog ruletimer1 0;
ruletimer2 0;
endon

Scusami di nuovo, sono proprio ignorante

Valerio1975, 2019-01-12:
Ho provato a fare linea per linea

`21:36:27 CMD: Rule1 1
21:36:27 RSL: stat/sonoff/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":265,"Rules":"on Clock#Timer=1 do backlog power1 1; ruletimer1 120 endon on Rules#Timer=1 do backlog power1 0; ruletimer2 60 endon on Rules#Timer=2 do backlog backlog power1 1; ruletimer1 1800 endon on Clock#Timer=2 do backlog ruletimer1 0; ruletimer2 0; endon"}
21:36:35 CMD: Rule1 on Clock#Timer=1 do backlog power1 1;
21:36:35 RSL: stat/sonoff/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":474,"Rules":"on Clock#Timer=1 do backlog power1 1;"}
21:37:06 CMD: ruletimer1 60 endon on Rules#Timer=1 do backlog power1 0;
21:37:06 RSL: stat/sonoff/RESULT = {"T1":60,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
21:37:12 CMD: ruletimer2 60 endon on Rules#Timer=2 do backlog backlog power1 1;
21:37:12 RSL: stat/sonoff/RESULT = {"T1":53,"T2":60,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
21:37:19 CMD: ruletimer1 60 endon on Clock#Timer=2 do backlog ruletimer1 0;
21:37:19 RSL: stat/sonoff/RESULT = {"T1":60,"T2":53,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
21:37:25 CMD: ruletimer2 0;
21:37:25 RSL: stat/sonoff/RESULT = {"T1":54,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
21:37:32 CMD: endon
21:37:32 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}`
da errore alla fine

ascillato, 2019-01-12:
Rules funciona solo si e scritto tutto in una sola linea:

```
Rule1 on Clock#Timer=1 do backlog power1 1; ruletimer1 1800 endon on Rules#Timer=1 do backlog power1 0; ruletimer2 900 endon on Rules#Timer=2 do backlog backlog power1 1; ruletimer1 1800 endon on Clock#Timer=2 do backlog ruletimer1 0; ruletimer2 0; endon
```

e dopo attivare la RULE:

`rule1 1`

Valerio1975, 2019-01-12:
Grazie mille ma a linee separate ho provato solo per vedere se funzionava.
Ho provato anche ad attivare dopo, ma non funziona. forse bisogna fare delle impostazioni diverse nella scheda configure time, forse bisogna giocare con ARM e Repeat.
Grazie mille, ho provato a leggere wifi RULE, ma non ho trovato nessuna differenza da quello che hai scritto tu.

Valerio1975, 2019-01-14:
Ok ho fatto tutto.

`00:00:08 WIF: Connected
00:00:08 HTP: Web server active on sonoff-0288 with IP address 192.168.1.147
00:00:08 APP: (UTC) Mon Jan 14 17:16:28 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
18:16:30 APP: Boot Count 32
18:16:30 CFG: Saved to flash at F7, Count 165, Bytes 3584
18:16:45 HTP: Main Menu
18:16:48 WIF: Checking connection...
18:16:48 WIF: Connected
18:16:49 HTP: Console
18:17:01 HTP: Main Menu
18:17:05 HTP: Configuration
18:17:07 HTP: Configure Timer
18:17:08 WIF: Checking connection...
18:17:08 WIF: Connected
18:17:25 HTP: Configure Timer
18:17:25 MQT: Timers 1,0x887F844B,0x807F844D,0x887F83C0,0x807F8528,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000
18:17:25 HTP: Configuration
18:17:25 CFG: Saved to flash at F6, Count 166, Bytes 3584
18:17:28 HTP: Main Menu
18:17:28 WIF: Checking connection...
18:17:28 WIF: Connected
18:17:30 HTP: Console
18:17:36 CMD: rule1 1
18:17:36 SRC: WebConsole from 192.168.1.149
18:17:36 RSL: Received Topic /rule1, Data Size 1, Data 1
18:17:36 RSL: Group 0, Index 1, Command RULE, Data 1
18:17:36 RSL: stat/sonoff/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":270,"Rules":"on event#T=1 do backlog power1 1; ruletimer1 5 endon on Rules#Timer=1 do backlog power1 0; ruletimer2 5 endon on Rules#Timer=2 do backlog backlog power1 1; ruletimer1 5 endon on event#T=2 do backlog ruletimer1 0; ruletimer2 0; power1 0 endon"}
18:17:45 CMD: rule1 on event#T=1 do backlog power1 1; ruletimer1 5 endon on Rules#Timer=1 do backlog power1 0; ruletimer2 5 endon on Rules#Timer=2 do backlog backlog power1 1; ruletimer1 5 endon on event#T=2 do backlog ruletimer1 0; ruletimer2 0; power1 0 endon
18:17:45 SRC: WebConsole from 192.168.1.149
18:17:45 RSL: Received Topic /rule1, Data Size 241, Data on event#T=1 do backlog power1 1; ruletimer1 5 endon on Rules#Timer=1 do backlog power1 0; ruletimer2 5 endon on Rules#Timer=2 do backlog backlog power1 1; ruletimer1 5 endon on event#T=2 do backlog ruletimer1 0; ruletimer2 0; power1 0 endon
18:17:45 RSL: Group 0, Index 1, Command RULE, Data on event#T=1 do backlog power1 1; ruletimer1 5 endon on Rules#Timer=1 do backlog power1 0; ruletimer2 5 endon on Rules#Timer=2 do backlog backlog power1 1; ruletimer1 5 endon on event#T=2 do backlog ruletimer1 0; ruletimer2 0; power1 0 endon
18:17:45 RSL: stat/sonoff/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":270,"Rules":"on event#T=1 do backlog power1 1; ruletimer1 5 endon on Rules#Timer=1 do backlog power1 0; ruletimer2 5 endon on Rules#Timer=2 do backlog backlog power1 1; ruletimer1 5 endon on event#T=2 do backlog ruletimer1 0; ruletimer2 0; power1 0 endon"}
18:17:48 WIF: Checking connection...
18:17:48 WIF: Connected
18:18:08 WIF: Checking connection...
18:18:08 WIF: Connected
18:18:28 WIF: Checking connection...
18:18:28 WIF: Connected
18:18:48 WIF: Checking connection...
18:18:48 WIF: Connected
18:19:00 SRC: Timer
18:19:00 RSL: stat/sonoff/RESULT = {"POWER":"ON"}
18:19:00 RSL: stat/sonoff/POWER = ON
18:19:00 CFG: Saved to flash at F5, Count 167, Bytes 3584
18:19:08 WIF: Checking connection...
18:19:08 WIF: Connected
18:19:28 WIF: Checking connection...
18:19:28 WIF: Connected
18:19:48 WIF: Checking connection...
18:19:48 WIF: Connected
18:20:08 WIF: Checking connection...
18:20:08 WIF: Connected
18:20:28 WIF: Checking connection...
18:20:28 WIF: Connected
18:20:48 WIF: Checking connection...
18:20:48 WIF: Connected
18:21:00 SRC: Timer
18:21:00 RSL: stat/sonoff/RESULT = {"POWER":"OFF"}
18:21:00 RSL: stat/sonoff/POWER = OFF
18:21:00 CFG: Saved to flash at F4, Count 168, Bytes 3584
18:21:08 WIF: Checking connection...
18:21:08 WIF: Connected
18:21:19 RSL: tele/sonoff/STATE = {"Time":"2019-01-14T18:21:19","Uptime":"0T00:04:59","Vcc":3.466,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Telecom-66782150","BSSId":"A4:B1:E9:F0:E4:C6","Channel":1,"RSSI":70}}
18:21:28 WIF: Checking connection...
18:21:28 WIF: Connected`

Valerio1975, 2019-01-14:
ah ok grazie mille, provo, ma solo sul 1 timer o anche sul secondo?

ascillato2, 2019-01-14:
si, anche il secondo. Le regole usano il timer 1 e il timer 2:

```
on Clock#Timer=1

on Clock#Timer=2
```

Valerio1975, 2019-01-14:
grazie mille, ora parte e fa il suo ciclo ma non si spegge piu a nebi che non eseguo `rule1 0`

Valerio1975, 2019-01-15:
Grazie mille, per ora funziona perfettamente.
Grazie ancora e scusa la mia ignoranza

ascillato, 2019-01-15:
Complimenti :)

Valerio1975, 2019-01-15:
Ora ho problemi ad aggiornare la lingua in italino, ho provato di tutto ma da sempre 
Upload started ...

e poi

Upload Failed

Upload buffer miscompare.

Apro un'altro isuue per avere una mano?

Ho scaricato "esptool-0.4.13-win32" ho cancellato la flash (almeno credo) e poi ho riprogrammato con Arduino IDE, ma puttroppo non funziona mai. Io sto usando SonOff DUal R2. Comunque ho provato anche su un sonoff rf, ma non mi funziona neppure li.

ascillato, 2019-01-15:
Leggi la wiki in troubleshooting. Si non funciona, per favore usa il chat di Tasmota per aiuto.

Valerio1975, 2019-01-15:
ok ho letto tutto ma a me non funziona mai.
Provero la chat se funziona
grazie ancora