arendst/Tasmota #4903: Change binary sensor name in mqtt discovery

kurniawan77, 2019-01-12:
Hi,
I have set up a sonoff with a PIR on gpio14 with "SwitchMode2 1" so the relais won't be switched with the PIR. Enabled SetOption19 en rebooted. In the console i get to see:

`21:11:27 MQT: homeassistant/switch/583785_RL_1/config = {"name":"Sonoff-01",.......`
and
`21:11:27 MQT: homeassistant/binary_sensor/583785_SW_2/config = {"name":"Sonoff2 SW",......`

I would like to change the switch2 name from "Sonoff2 SW" to "Sonoff-01 SW2" to be more in sync with the given friendly name (Sonoff-01).

How can i do this?

ascillato, 2019-01-12:
When using Home Assistant Discovery, you can change the displayed name in Home Assistant right from the Home Assistance interfase

ascillato2, 2019-01-12:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-12 by ascillato2

ascillato, 2019-01-14:
Yes, you are right, but that will depends on your friendly name. Try to put different names for different devices.

kurniawan77, 2019-01-14:
Done that. But like i said earlier, it is not in sync with the given friendly name (Sonoff-01). 
Please, i dont want to be rude but you closed my issue without answering properly. I would like to know if it is possible to change the switch2 name in sonoff and how to do.

ryaske, 2019-11-19:
I was wondering the same. On the main Tasmota web ui, I have DS18B20-1, DS18B20-2, DS18B20-3.

How can these be renamed to friendly names? I am assuming they cant?

ElvisJaime, 2020-01-04:
> Done that. But like i said earlier, it is not in sync with the given friendly name (Sonoff-01).
> Please, i dont want to be rude but you closed my issue without answering properly. I would like to know if it is possible to change the switch2 name in sonoff and how to do.

Have You already found the answer to what you were saying at the beginning of this post ????
I am with the same dude/Problem.

Thanks

austwhite, 2020-05-02:
I'd like to know the answer to.
I do not believe this question was ever answerd.  I am unsure why it was closed.
The question was how to rename the switch in the Tasmota/Sonoff firmware, not how to rename it in Home Assistant.
If the answer is no they can't be renamed in Tasmota, that would have been appreciated rather than closing the issue and saying it was answered.....
This still issue still comes up in Google searches.

ascillato, 2020-05-03:
> I do not believe this question was ever answerd. I am unsure why it was closed.

This is a very old issue (more than a year old) and duplicated. It was already answered in other issues. Sorry that this was your first searching result.

In Home Assistant it can be changed form the Home Assistant interface.

In Tasmota can't be modified as a common config, because it is meant to be used in your home automation software. But if you really need a specific TOPIC and PAYLOAD for your switch, you can use a rule as explained in the docs.

Example:

Asumming a `Switch2` connected to a PIR sensor and that `switchmode2` is set to 1, and you want that the switch2 is being sent as PIR_GARAGE with payloads MOV and NO_MOV. So, your desided mqtt message would:

for switch2 with 1 sent:
`stat/GARAGE/PIR_GARAGE MOV`

and for switch2 with 0 sent:
`stat/GARAGE/PIR_GARAGE NO_MOV`

You can solve that using the following rules that must be typed in the Tasmota Console:

```lua
switchmode2 1
rule1 1
rule1 on switch2#state=0 do publish stat/GARAGE/PIR_GARAGE NO_MOV endon on switch2#state=1 do publish stat/GARAGE/PIR_GARAGE MOV endon
```

Hope this solves your issue/question.

If you need further assistance on Tasmota usage, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)