arendst/Tasmota #4904: Add support for Luminea ZX2820

knop-ster, 2019-01-12:
Please add support for Luminea ZX2820.

Please create a new device with the following settings:

{ "Luminea ZX2820",
    GPIO_KEY1,        // GPIO00 Button
     0, 0, 0,
     GPIO_HLW_CF,      // GPIO04 HLW8012 CF power
     GPIO_NRG_CF1,     // GPIO05 HLW8012 CF1 voltage / current
                       // GPIO06 (SD_CLK   Flash)
                       // GPIO07 (SD_DATA0 Flash QIO/DIO/DOUT)
                       // GPIO08 (SD_DATA1 Flash QIO/DIO/DOUT)
     0,                // GPIO09 (SD_DATA2 Flash QIO or ESP8285)
     0,                // GPIO10 (SD_DATA3 Flash QIO or ESP8285)
                       // GPIO11 (SD_CMD   Flash)
     GPIO_NRG_SEL_INV, // GPIO12 HLW8012 SEL (0 = Voltage)
     GPIO_LED1_INV,    // GPIO13 Green Led - Link and Power status
     GPIO_REL1,        // GPIO14 Relay
     0, 0, 0
  }

The basis for this are the tickets #1190 and #2389.

ascillato2, 2019-01-12:
Have you test it?

knop-ster, 2019-01-12:
Sure. Tested and it‘s working.

ascillato2, 2019-01-13:
There is no other device with the exact same pinout in the template files.

Please, can you make a PR to add this?

knop-ster, 2019-01-13:
That is the point. Luminea ZX2820 is not 100% the same socket to KMC 70011. Compared to KMC 70011 they are the same connections, only the power consumption has to be inverted. For this it seems sufficient to set the variable GPIO_NRG_SEL_INV instead of GPIO_NRG_SEL.

My code snippet above was for release version 6.4.1. For the current development branch, the snippet must look like this:

{ "Luminea ZX2820",
    GPIO_KEY1,        // GPIO00 Button
     0, 0, 0,
     GPIO_HLW_CF,      // GPIO04 HLW8012 CF power
     GPIO_NRG_CF1,     // GPIO05 HLW8012 CF1 voltage / current
                       // GPIO06 (SD_CLK   Flash)
                       // GPIO07 (SD_DATA0 Flash QIO/DIO/DOUT)
                       // GPIO08 (SD_DATA1 Flash QIO/DIO/DOUT)
     0,                // GPIO09 (SD_DATA2 Flash QIO or ESP8285)
     0,                // GPIO10 (SD_DATA3 Flash QIO or ESP8285)
                       // GPIO11 (SD_CMD   Flash)
     GPIO_NRG_SEL_INV, // GPIO12 HLW8012 SEL (0 = Voltage)
     GPIO_LED1_INV,    // GPIO13 Green Led - Link and Power status
     GPIO_REL1,        // GPIO14 Relay
     0, 0, 0
  }

Comparing the snippets for KMC 70011:

{ "KMC 70011",       // KMC 70011
                       // https://www.amazon.com/KMC-Timing-Monitoring-Network-125V-240V/dp/B06XRX2GTQ
     GPIO_KEY1,        // GPIO00 Button
     0, 0, 0,
     GPIO_HLW_CF,      // GPIO04 HLW8012 CF power
     GPIO_NRG_CF1,     // GPIO05 HLW8012 CF1 voltage / current
                       // GPIO06 (SD_CLK   Flash)
                       // GPIO07 (SD_DATA0 Flash QIO/DIO/DOUT)
                       // GPIO08 (SD_DATA1 Flash QIO/DIO/DOUT)
     0,                // GPIO09 (SD_DATA2 Flash QIO or ESP8285)
     0,                // GPIO10 (SD_DATA3 Flash QIO or ESP8285)
                       // GPIO11 (SD_CMD   Flash)
     GPIO_NRG_SEL,     // GPIO12 HLW8012 SEL (1 = Voltage)
     GPIO_LED1_INV,    // GPIO13 Green Led - Link and Power status
     GPIO_REL1,        // GPIO14 Relay
     0, 0, 0
  }

ascillato, 2019-01-13:
PR https://github.com/arendst/Sonoff-Tasmota/pull/4921

closed 2019-01-13 by ascillato2

Jason2866, 2019-01-13:
@knop-ster  Is this device screwed or glued?
Could you add the device to wiki and / or post some pictures from inside?

knop-ster, 2019-01-14:
This device is screwed.
I will gladly document these days in the wiki. I'll let you know when I've done that.

EnerGeh, 2019-10-06:
I'd prefer a new issue for that, but don't want to spam your issue tracker, just gimme a hint if you also prefer a new issue.
The referenced template doesn't work for me concerning the LedState with the "Luminea ZX2820-675". In the [fhem forum](https://forum.fhem.de/index.php/topic,93324.msg926886.html#msg926886) I found the following setting that perfectly fits the device and correctly sets the LED as configured with LED state:
`{"NAME":"ZX2820","GPIO":[0,0,56,0,0,134,0,0,131,17,132,21,0],"FLAG":0,"BASE":45}`
Maybe you can update the wiki and the template?

rrr-iob, 2020-01-26:
Does anyone get the info of the power consumption?
I configured MQTT, selected the Luminea ZX2820 Module and the plug is showing up in ioBroker. But the value.power.consumption value is set once and than never change dynamically. On the tastmota site of the plug is power consumption is shown.
What am I doing wrong?

jogge777, 2020-08-20:
> I'd prefer a new issue for that, but don't want to spam your issue tracker, just gimme a hint if you also prefer a new issue.
> The referenced template doesn't work for me concerning the LedState with [...]

I have the same plug and this settings fits for my plug also. Thanks for your post. The template was not sufficient.