arendst/Tasmota #4905: Using both Powerlow AND Powerdelta simultaniously

rasti70, 2019-01-12:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ x] Development/Compiler/Upload tools used : ATOM with own 6.3.0.5 and original 6.4. binary
- [x ] Hardware used : Blitzwolf SHP 2
- [ x] If a pre-compiled release or development binary was used, which one? : 6.4 DE
- [x ] You have tried latest release or development binaries? : latest dev.
- [ x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
`22:30:07 CMD: status 0
22:30:07 MQT: tele/Mess_Steckdose_99/STATUS = {"Status":{"Module":45,"FriendlyName":["Mess_Steckdose_99"],"Topic":"Mess_Steckdose_99","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:30:07 MQT: tele/Mess_Steckdose_99/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"Blitzwoelfe","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T06:48:08","StartupUTC":"2019-01-12T14:41:59","Sleep":0,"BootCount":8,"SaveCount":110,"SaveAddress":"F6000"}}
22:30:07 MQT: tele/Mess_Steckdose_99/STATUS2 = {"StatusFWR":{"Version":"6.3.0.5(sonoff)","BuildDateTime":"2018.12.15 14:18:49","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
22:30:07 MQT: tele/Mess_Steckdose_99/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Offline3",""],"TelePeriod":300,"SetOption":["00208009","55898000","00000000"]}}
22:30:07 MQT: tele/Mess_Steckdose_99/STATUS4 = {"StatusMEM":{"ProgramSize":494,"Free":508,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashMode":3,"Features":["00000407","0FDAE794","240183A0","23B617CE","00003BC0"]}}
22:30:07 MQT: tele/Mess_Steckdose_99/STATUS5 = {"StatusNET":{"Hostname":"Mess_Steckdose_99-4155","IPAddress":"192.168.178.99","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"CC:50:E3:D4:30:3B","Webserver":2,"WifiConfig":5}}
22:30:07 MQT: tele/Mess_Steckdose_99/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.6","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D4303B","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:30:07 MQT: tele/Mess_Steckdose_99/STATUS7 = {"StatusTIM":{"UTC":"Sat Jan 12 21:30:07 2019","Local":"Sat Jan 12 22:30:07 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:40","Sunset":"17:16"}}
22:30:07 MQT: tele/Mess_Steckdose_99/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
22:30:07 MQT: tele/Mess_Steckdose_99/STATUS10 = {"StatusSNS":{"Time":"2019-01-12T22:30:07","ENERGY":{"TotalStartTime":"2019-01-09T21:37:37","Total":0.010,"Yesterday":0.001,"Today":0.000,"Power":0.5,"ApparentPower":6.9,"ReactivePower":6.9,"Factor":0.07,"Voltage":231,"Current":0.030}}}
22:30:07 MQT: tele/Mess_Steckdose_99/STATUS11 = {"StatusSTS":{"Time":"2019-01-12T22:30:07","Uptime":"0T06:48:08","Vcc":3.116,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Offline3","BSSId":"FC:EC:DA:C3:B3:85","Channel":6,"RSSI":100}}}

`

**(Please, remember to close the issue when the problem has been addressed)**

It seems the combination of powerlow and powerdelta does not work. 

I want to setup the following configuration with a Blitzwolf SHP2 / SHP6 wifi power measing  plug
teleperiod 300 => send regularly a MQTT message  => that works
powerdelta 10 => send a MQTT anytime measured power has increased/decreased by 10% => that works too (tested with variable load - soldering iron)
powerlow 10 => no MQTT messages if measured power is less than 10W => NOT WORKING 

The problem I have is that if the device is ON and power is zero, i.e. 0W or next to that value (<1W) the server is sending out MQTT 
messages every few seconds and I do not really understand why.

 I turned (rules off) AND (powerdelta 10) AND (powerlow 10). 
Powerdelta works, but powerlow doesn't. It should work in combination. 

Below you see now that I turned rules (rule1) off and set powerlow 10 and powerdelta 10.

A workaround could be to set up a tasmota rule. Maybe. I also tried that and couldn't get it working.
But mainly, I think that powerdelta and powerlow should work together the same time.

Or am I doing something wrong here ??

 In the webinterface I see a power of 0 ... to .... 0.3W. And in the console you can see all the unwanted messages :    ``20:14:18 CMD: rule1
20:14:18 MQT: tele/Mess_Steckdose_99/RESULT = {"Rule1":"OFF","Once":"ON","StopOnError":"OFF","Free":366,"Rules":"on ENERGY#Current>0.100 do publish tele/Mess_Steckdose_99/SENSOR 1 endon on ENERGY#Current<0.100 do publish tele/Mess_Steckdose_99/SENSOR 0 endon"}
20:14:22 CMD: powerdelta
20:14:22 MQT: tele/Mess_Steckdose_99/RESULT = {"PowerDelta":10}
20:14:26 CMD: powerlow
20:14:26 MQT: tele/Mess_Steckdose_99/RESULT = {"PowerLow":10}
20:14:31 MQT: tele/Mess_Steckdose_99/STATE = {"Time":"2019-01-11T20:14:31","Uptime":"0T03:50:13","Vcc":3.116,"POWER":"ON","Wifi":{"AP":1,"SSId":"Offline3","BSSId":"FC:EC:DA:C3:B3:85","Channel":6,"RSSI":98}}
20:14:31 MQT: tele/Mess_Steckdose_99/SENSOR = {"Time":"2019-01-11T20:14:31","ENERGY":{"TotalStartTime":"2019-01-09T21:37:37","Total":0.008,"Yesterday":0.001,"Today":0.000,"Period":0.0,"Power":0.0,"ApparentPower":0.0,"ReactivePower":0.0,"Factor":0.00,"Voltage":235,"Current":0.000}}
20:14:38 MQT: tele/Mess_Steckdose_99/SENSOR = {"Time":"2019-01-11T20:14:38","ENERGY":{"TotalStartTime":"2019-01-09T21:37:37","Total":0.008,"Yesterday":0.001,"Today":0.000,"Power":0.1,"ApparentPower":5.6,"ReactivePower":5.6,"Factor":0.02,"Voltage":235,"Current":0.024}}
`
and so on and on, MQTT messages are fired every few seconds.

ascillato2, 2019-01-13:
> The problem I have is that if the device is ON and power is zero, i.e. 0W or next to that value (<1W) the server is sending out MQTT
> messages every few seconds and I do not really understand why.

From your logs it is working. If your power consumption is less than 10w, a mqtt is being sent (powerlow 10).

Powerlow is powerlow to a value. 

May be you wanted power higher to 10w?

ascillato2, 2019-01-13:
May be what you want is something like in https://github.com/arendst/Sonoff-Tasmota/issues/2880 ? 

```
rule 1
rule 4
rule on Energy#Power>5 do publish stat/sonoff/energy %value% endon
```

arendst, 2019-01-13:
> The problem I have is that if the device is ON and power is zero, i.e. 0W or next to that value (<1W) the server is sending out MQTT
> messages every few seconds and I do not really understand why.

It's because of PowerDelta. Any change of 10% will send a MQTT message. With Power toggling between 0 and some a 10% change is very likely.

rasti70, 2019-01-13:
Hello Adrian,

yes I would like the rule in #2880 BUT in combination with powerdelta.

So

if power>5w => powerdelta function (and teleperiod) to trigger mqtt message
if power<5W only teleperiod mqtt message

It doesn't work with the combination of powerlow & Powerdelta
and also doesn't work with the combination of powerhigh & Powerdelta (I just tried that)

In the forum you said that a rule in combination with powerdelta doesn't work
So most probably I will have to setup a rule for what I need ?

ascillato, 2019-01-13:
Hi, I didn't say it won't work. I told you that if you duplicate behaviours with powerlow, powerhigh, powerdelta AND also the same with rules you will have a situation very difficult to debug.

From your use-case, seems to be better to use just rules to accomplish what you want.

As Theo explained, powerdelta is a % of variation, so for low values it is very likely to sent mqtt messages very often.

You have an example in rules to send mqtt messages when the value reaches a delta. Have you tried that?

rasti70, 2019-01-13:
> You have an example in rules to send mqtt messages when the value reaches a delta. Have you tried that?

I suppose you mean https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#8a-sending-the-value-of-a-sensor-to-mqtt-only-when-a-delta-is-reached 
No I haven't tried. I still don't understand the code so I also don't know how to adapt it to my needs.

ascillato, 2019-01-13:
Ok. No problem. Let me rewrite them for your use-case. I will post the modified example later.

rasti70, 2019-01-13:
> Ok. No problem. Let me rewrite them for your use-case. I will post the modified example later.

that really would be  great. I'll be waiting anxiously....

ascillato, 2019-01-13:
From your use-case:

> if power>5w => powerdelta function (and teleperiod) to trigger mqtt message
> if power<5W only teleperiod mqtt message

So, if power <5w, normal tasmota teleperiod. Nothing to add
But, if power >5w you want a mqtt message if the power reaches a delta.

So, go to the Tasmota console and type:

```
powerdelta 0
powerlow 0
powerhigh 0
rule1 0
rule1 4
rule1 on Energy#Power>5 do powerdelta 10 endon on Energy#Power<5 do powerdelta 0 endon
rule1 1
```

rasti70, 2019-01-13:
thank you. 

This results in endless
```
19:01:28 MQT: tele/Mess_Steckdose_99/RESULT = {"PowerDelta":0}
19:01:30 RUL: ENERGY#POWER<5 performs "powerdelta 0"
```
(or endless powerdelta 1 for Power>5)

So my guess was to set the ONCE-Parameter to ON 
=> 
rule1 5

Seems to work ! I will observe some time but I think this is solved now

Thanks so much !

closed 2019-01-13 by ascillato2

rasti70, 2019-01-13:
One last question in this context:

I am compiling binaries for each of my devices containing target device, wifi credentials ect. I would to include that rule as well. Can I write it in sonoff\my_user_config.h or elsewhere ? 

Thanks

ascillato, 2019-01-13:
Sorry, rules must be written in the console, or by mqtt command or by http command.

rasti70, 2019-01-13:
OK no problem Thanks !

rasti70, 2019-01-13:
There is one small issue remaing. 
In the code below I changed the limit from 5W to 3W. Rest is as above.

The problem: When there is change in the condition (e.g. power goes up 0 from e.g. 15),
then the threshold of 3 is surpassed. Then I get the message

```
23:37:23 RUL: ENERGY#POWER>3 performs "powerdelta 10"
23:37:23 MQT: tele/Mess_Steckdose_2/RESULT = {"PowerDelta":10}
```
So the power increased to 15W .........BUT : No mqtt message with that power 
value is sent out at this point. 

Only if I increase power a little bit higher to 21W, powerdelta 10 applies and I get
`23:38:17 MQT: tele/Mess_Steckdose_2/SENSOR = {"Time":"2019-01-13T23:38:17","ENERGY":{"TotalStartTime":"2019-01-13T23:30:50","Total":5.561,"Yesterday":0.000,"Today":0.090,"Power":21.3,"ApparentPower":25.9,"ReactivePower":14.7,"Factor":0.82,"Voltage":235,"Current":0.110}}`

If I had not increased the power further after the threshold crossing, the next MQTT that would have been sent out  is the teleperiod mqtt message.

Same thing happens when crossing the threshold from high to low power.

Right now the rule only defines that the powerdelta command changes.
Can the rule be changed such that additionally one MQTT message is sent out
when the threshold is crossed ?

Hope you understand what I mean....

meingraham, 2019-01-13:
I saw this thread and it is applicable to one of my automation scenarios. I'm seeing something similar. I posted this over on Discord but thought I'd share it here as it may shed light and document for posterity once this is understood/resolved.

I'm not clear on what `Rule1 5` ("Enable one-shot detection (Perform commands once until trigger is not met)") does exactly.

I'm looking to implement a variation of your scenario:

My rule:
```
Rule1 ON Energy#Power<3 DO PowerDelta 0 ENDON ON Energy#Power>75 DO PowerDelta 2 BREAK ON Energy#Power>2 DO PowerDelta 5 ENDON
```

Note that I am using `BREAK` so that if the Power>75 is triggered, the Power>2 won't also trigger.

I want the rule to trigger only once for when Power<3 until Power>75 (and then only once) or  until Power>2 (and then only once). So, I want it to trigger each time one of the triggers is met, but only once until one of the other conditions occur to trigger the rule.

Is this what `Rule1 5` will does?

The rule seems to trigger (each time) when Power<3 threshold is _**crossed**_ (`PowerDelta 0`).
```
ENERGY#POWER<3 performs "PowerDelta 0"
stat/sonoff-washer_monitor/RESULT = {"PowerDelta":0}
```
However, when  power was greater than 2 (e.g., `Teleperiod` message with power=4), the rule (`PowerDelta 5`) did not trigger. However, when power went back under 3, the rule (`PowerDelta 0`) triggered once again.

Regards,

Mike

rasti70, 2019-01-13:
> Is this what Rule1 5 will does?

when you type rule in the console you see something like this => 

`00:32:38 MQT: tele/Mess_Steckdose_4/RESULT = {"Rule1":"ON","Once":"ON","StopOnError":"OFF","Free":431,"Rules":"on Energy#Power>3 do powerdelta 10 endon on Energy#Power<3 do powerdelta 0 endon"}`

You see Once = ON. This means if the condition is fullfilled, the corresponding powerdelta command applies you get a MQTT message about that like

```
00:33:56 RUL: ENERGY#POWER>3 performs "powerdelta 10"
00:33:56 MQT: tele/Mess_Steckdose_4/RESULT = {"PowerDelta":10}
```

Now, if ONCE is set to ON using the command rule 5, said last message will be sent out only once.
If not, you will get an endless again and again of the same mqtt messages

meingraham, 2019-01-14:
> If ONCE is set to ON using the command rule 5, said last message will be sent out only once.
If not, you will see again and again the same mqtt messages

That's how I understand it too.

But what seems to be happening is that the only thing that's happening once is the first trigger in the rule. The other commands never get triggered even though the ON condition is met. So, `Rule1 5` seems to be doing its thing properly (only letting the trigger happen once until a different trigger "resets" it. It's just that the other triggers never happen.

Hopefully Adrian can shed some light on my problem and the one you posted.

Mike

rasti70, 2019-01-14:
> Note that I am using BREAK so that if the Power>75 is triggered, the Power>2 won't also trigger.

I don't know if once=ON  applies to all conditions or only the 1st condition in your rule.

I googled a little bit and found https://github.com/arendst/Sonoff-Tasmota/issues/4477 and https://github.com/arendst/Sonoff-Tasmota/pull/4500

Some ideas :
- Maybe simply your syntax is wrong ??  
- Or use backlog ?
- Or you can use a rule which switches on and off another rule ? So you can be sure that ONCE=ON for each of the rules, see example https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#10a-advanced-pir-rules-for-staircase

These are just some thoughts but to be honest, I am not very familiar with rules and have problems to understand the syntax in particular with complicated conditions and nested loops. Maybe it would be helpful to have some comparison table with 1st column syntax in C and 2nd column tasmota rule syntax ...

meingraham, 2019-01-14:
@rasti70

> I googled a little bit and found #4477 and #4500

Yes, these were the topics I'd followed when Adrian was in the process of adding the new feature several weeks ago. That is the only place where it is currently documented. When I get some time I'll grab Adrian's comments from the topics and move them to the wiki. I just wanted to make sure that BREAK is here to stay before I do.

> I don't know if once=ON applies to all conditions or only the 1st condition in your rule.

Right, I'm not completely clear on `Rule1 5`'s behavior. My issue is that I never see a `ENERGY#POWER>75 performs "PowerDelta 2"` or a `ENERGY#POWER>2 performs "PowerDelta 5"`. I may need to put my ON Energy#Power<3 DO PowerDelta 0 ENDON as the final trigger instead of the first.

As to using `Backlog` - I don't have multiple commands that I want to execute as the result of one of the conditions triggering. However, in answer to your question about getting TASMOTA to send the power reading MQTT message when your rule triggers, you can have a `Backlog` in your `ON`/`ENDON` that not only sets `PowerDelta` but then also `PUBLISH`es an MQTT message (either just send the power sensor value or send out a `STATUS 8` to which your device will respond to which will publish the Energy metrics.

@ascillato - thoughts?

Mike

ascillato2, 2019-01-14:
@rasti70

> In the code below I changed the limit from 5W to 3W. Rest is as above.
>
> The problem: When there is change in the condition (e.g. power goes up 0 from e.g. 15),
> then the threshold of 3 is surpassed. [...]

What is happening is that Tasmota triggers the rule when you reach ENERGY#POWER>3 and then powerdelta is executed 50ms later. So, the powerdelta will start from the actual ENERGY#POWER value when it was executed. 

> Right now the rule only defines that the powerdelta command changes.
> Can the rule be changed such that additionally one MQTT message is sent out
> when the threshold is crossed ?

Yes, with a `backlog` to add more commands when the rule is triggered and a `publish` to send the mqtt message you want.

Another option is to call the `status 11`:

```
powerdelta 0
powerlow 0
powerhigh 0
rule1 0
rule1 5
rule1 on Energy#Power>5 do backlog powerdelta 10; status 11 endon on Energy#Power<5 do backlog powerdelta 0; status 11 endon
rule1 1
```

rasti70, 2019-01-15:
@ascillato2 : I tried that. 

When I cross the threshold from low power to high power it works :
```
18:09:34 RUL: ENERGY#POWER>3 performs "backlog powerdelta 10; status 11"
18:09:34 MQT: tele/Mess_Steckdose_5/RESULT = {"PowerDelta":10}
18:09:34 MQT: tele/Mess_Steckdose_5/STATUS11 = {"StatusSTS":{"Time":"2019-01-15T18:09:34","Uptime":"0T00:16:14","Vcc":3.459,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"Offline3","BSSId":"FC:EC:DA:C3:B3:85","Channel":6,"RSSI":100}}}
18:09:35 MQT: tele/Mess_Steckdose_5/SENSOR = {"Time":"2019-01-15T18:09:35","ENERGY":{"TotalStartTime":"2019-01-15T17:39:41","Total":0.014,"Yesterday":0.000,"Today":0.014,"Power":16.3,"ApparentPower":23.8,"ReactivePower":17.4,"Factor":0.68,"Voltage":234,"Current":0.102}}
```

Now, when crossing the threshold from high power to a low power, I also get MQTT message but that message does not contain the power value. See here what happens when switch is ON and I turn off the load (15W=>0W)
```
18:06:20 RUL: ENERGY#POWER<3 performs "backlog powerdelta 0; status 11"
18:06:20 MQT: tele/Mess_Steckdose_5/RESULT = {"PowerDelta":0}
18:06:21 MQT: tele/Mess_Steckdose_5/STATUS11 = {"StatusSTS":{"Time":"2019-01-15T18:06:21","Uptime":"0T00:13:01","Vcc":3.458,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"Offline3","BSSId":"FC:EC:DA:C3:B3:85","Channel":6,"RSSI":100}}}
```
That message does not contain the power value.
Isn't it possible to produce instead of `tele/Mess_Steckdose_5/STATUS11` I would like to have  `tele/Mess_Steckdose_5/SENSOR` ? Or output both messages STATUS11 and SENSOR like in the case when crossing the threshold from high power to a low power,

meingraham, 2019-01-15:
SENSOR is generated by TASMOTA when TelePeriod triggers. To have TASMOTA send the sensor metrics on demand, put a `STATUS 8` in your DO (e.g., with `Backlog`).

rasti70, 2019-01-15:
> `SENSOR` is generated by TASMOTA when `TelePeriod` triggers. To have TASMOTA send the sensor metrics on demand, put a `STATUS 8` in your DO (e.g., with `Backlog`).

thanks you. 

Now I get the sensor metrics as expected when going from low Power to high Power.
But still no  power / sensor metric mqtt message when crossing the threshold from high to low

ascillato2, 2019-01-15:
Even another option is to call the teleperiod :

```
powerdelta 0
powerlow 0
powerhigh 0
rule1 0
rule1 5
rule1 on Energy#Power>5 do backlog powerdelta 10; teleperiod 300 endon on Energy#Power<5 do backlog powerdelta 0; teleperiod 300 endon
rule1 1
```

rasti70, 2019-01-15:
> Even another option is to call the teleperiod :
> 
> ```
> powerdelta 0
> powerlow 0
> powerhigh 0
> rule1 0
> rule1 5
> rule1 on Energy#Power>5 do backlog powerdelta 10; teleperiod 300 endon on Energy#Power<5 do backlog powerdelta 0; teleperiod 300 endon
> rule1 1
> ```

goooooooooooooooooooooooooooool !!!

Yes. You made it. Works perfectly !

Thanks so much

ascillato, 2019-01-15:
Great :) :+1:

meingraham, 2019-01-15:
> Even another option is to call the teleperiod

There's my new TASMOTA thing learned for today!