arendst/Tasmota #4906: Sonoff Basic TH 6.4.1.8 (Core 2_5_0) / DS18b20

raymajor, 2019-01-12:
On the Sonoff Basic TH 6.4.1.8 (Core 2_5_0) is the DS18b20 not working (Recordnice) any more.
It's working from 6.4.1.5 until 6.4.17 (Core 2_5_0) I have tested that.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x ] _Searched the problem in issues and in the wiki_
- [x ] _Hardware used_ : Sonoff Basic TH
- [x ] _Development/Compiler/Upload tools used_ : OTA and file upgrade
- [ x] _If a pre-compiled release or development binary was used, which one?_ : 6.4.1.8 (Core 2_5_0)
- [ x] _You have tried latest release or development binaries?_ :
- [ x] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE

23:06:10 CMD: Status 0
23:06:10 MQT: stat/sonoff/STATUS = {"Status":{"Module":4,"FriendlyName":["Storage"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:06:10 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/020500/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:17","StartupUTC":"2019-01-12T22:05:53","Sleep":50,"BootCount":266,"SaveCount":1033,"SaveAddress":"FA000"}}
23:06:10 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1.8(0d7b7c8-sonoff)","BuildDateTime":"2019-01-10T19:16:16","Boot":6,"Core":"2_5_0_BETA2","SDK":"3.0.0-dev(c0f7b44)"}}
23:06:11 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["RayMan2.4",""],"TelePeriod":300,"SetOption":["00008009","55C180C0","00000001"]}}
23:06:11 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":547,"Free":456,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
23:06:11 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-7278","IPAddress":"192.168.10.187","Gateway":"192.168.10.254","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"2C:3A:E8:4E:7C:6E","Webserver":2,"WifiConfig":3}}
23:06:11 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.10.75","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4E7C6E6E","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
23:06:11 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sat Jan 12 22:06:11 2019","Local":"Sat Jan 12 23:06:11 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:40","Sunset":"17:16"}}
23:06:11 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-01-12T23:06:11"}}
23:06:11 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-01-12T23:06:11","Uptime":"0T00:00:18","Vcc":3.447,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"RayMan2.4","BSSId":"70:4F:57:43:E3:73","Channel":11,"RSSI":94}}}
```

ascillato2, 2019-01-13:
Hi,

From 6.4.1.7 to 6.1.4.1.8 there is no change related to DS18B20.
As you saw when searching issues for DS18B20, there were some people of just a wire disconnection. Please, could you test your wirings again and try again? This seems to be just hardware related.

raymajor, 2019-01-13:
Hi Adrian,
I have it on the Sonoff Basic TH and also with the Wemos D1 mini. (Told Mike also this morning)
From 6.4.1.5 until 6.4.1.7 it's working and with 6.4.18 not anymore. Is it still then wiring?

Thanks, Raymond-NL

raymajor, 2019-01-13:
Hi Adrian,
I did some test with Mike and the Sonoff-sensor.bin file works with the DS. But the standard Sonoff.bin is not working with the DS18b20.

Cheers, Raymond-NL

ascillato2, 2019-01-14:
Good to know that it is working for you using **sonoff-sensors.bin** :+1: So, it isn't a hardware issue.

As there is no change from 6.4.1.7 to 6.1.4.8 related to DS18B20, please, try again but with the latest version from today (commit 7a7cb40) from http://thehackbox.org/tasmota/020500/sonoff.bin

This sensor is included in sonoff.bin

Is not included in sonoff-basic.bin

raymajor, 2019-01-14:
Hi Adrian, I test it on thursday when i'm at home. I let you now...

Cheers, Raymond

Fleshi1981, 2019-01-15:
> Good to know that it is working for you using **sonoff-sensors.bin** 👍 So, it isn't a hardware issue.
> 
> As there is no change from 6.4.1.7 to 6.1.4.8 related to DS18B20, please, try again but with the latest version from today (commit [7a7cb40](https://github.com/arendst/Sonoff-Tasmota/commit/7a7cb40fdc7df879e968ef067c439ab124b0f606)) from http://thehackbox.org/tasmota/020500/sonoff.bin
> 
> This sensor is included in **sonoff.bin**
> 
> Is not included in **sonoff-basic.bin**

Thanks, with this new bin file my DS18B20 is working again on my Sonoff TH :)

ascillato, 2019-01-15:
Cool. So indeed was an issue with the automatic compilation. Now it is working fine. Thanks.

Please, close this issue. Thanks

raymajor, 2019-01-16:
Hi Adrian, It works also on V6.4.1.9 with the Sonoff.bin
Thanks, See you

You can close this one

ascillato, 2019-01-16:
You have opened the issue, so you can close it.

closed 2019-01-16 by ascillato2

unreal4u, 2019-01-17:
@ascillato2 thank you sooooo much for this comment: 
This sensor is included in sonoff.bin

Is not included in sonoff-basic.bin

This was the problem for me as well! Thank you so much for all your effort on this repo! It certainly helped me as well 👍

ascillato, 2019-01-17:
@unreal4u

Great that this solved your issue too. If you want to check the full list, see the releases page. Thanks