arendst/Tasmota #4908: Sonoff Basic new layout - GPIO2 confusion

speedyrazor, 2019-01-12:
Hi, I can see on the Sonoff Basic page ([https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Basic](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Basic)) it states:

> GPIO14 is not inuse anymore. Instead you can use GPIO2 labled as IO2 on the board.

But then on this page ([https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs](https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs)) is states:

> GPIO2 can't be low at boot, so you can't connect a switch to it.

I am desperately trying to use this with a standard light switch (as with GPIO14 on the original Basic). I tried GPIO2 and gnd, configured 'Switch1 (09)' and while this worked, the lights kept flickering (anyway to prevent this, please be descriptive as I am new to this). Currently I am using 'key' and 'gnd' which works, but only on a rocker switch, which is not perfect. Looking around, there are quite a few people . in the same boat, so would be good to get some workaround (GPIO14 on the chip is way too small for me to solder).
Kind regards.

digiblur, 2019-01-13:
I'd probably use GPIO3 if you are using a switch that could be low at boot.  GPIO2 can be difficult to connect to for some.  GPIO1 can also be used with ease but of course you can't have it low during boot.  One other idea is possible stealing the LED GPIO and switching to a generic template.

speedyrazor, 2019-01-13:
Thanks for the suggestion. I am now using GPIO3 (RX) and gnd, with this setting on the Configuration/Module parameters:

> GPIO3 Serial In - Switch1 (09)

Have been running it for a couple of hours, and all seems fine so far (fingers crossed).

closed 2019-01-13 by ascillato2