arendst/Tasmota #4909: Tuya Dimmer not fading properly

AalianKhan, 2019-01-12:
Hello, I just flashed a tuya dimmer and everything works fine. I was just wondering if the dimmer can fade to a value. is this possible?(edited)
I saw the fade function but it is not working for me. it seems like it sets the value dimmer to random numbers.
```
00:21:29 CMD: dimmer 75
00:21:29 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":75}
00:21:29 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":15}
```
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used :
- [x] Hardware used : Moes dimmer (Tuta dimmer) connected to a wal socket
- [x] If a pre-compiled release or development binary was used, which one? : 6.4.1.5
- [x] You have tried latest release or development binaries? :
- [x] Provide the output of command ``status 0`` :
```
 CMD: status 0
 stat/sonoff/STATUS = {"Status":{"Module":54,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
 stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:26:21","StartupUTC":"2019-01-12T23:20:43","Sleep":50,"BootCount":4,"SaveCount":232,"SaveAddress":"F4000"}}
 stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1.5(f8ab1a8-sonoff)","BuildDateTime":"2019-01-04T23:00:56","Boot":4,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
 stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["5920 Talisman",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
 stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":522,"Free":480,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
 stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-1880","IPAddress":"192.168.0.42","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"5C:CF:7F:7D:C7:58","Webserver":2,"WifiConfig":4}}
 stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7DC758","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
 stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sat Jan 12 23:47:04 2019","Local":"Sun Jan 13 00:47:04 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:39","Sunset":"17:17"}}
 stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-01-13T00:47:04"}}
 stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-01-13T00:47:04","Uptime":"0T00:26:21","Vcc":3.438,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Dimmer":99,"Fade":"ON","Speed":3,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"5920 Talisman","BSSId":"B4:75:0E:96:FD:44","Channel":6,"RSSI":100}}}
```

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2019-01-13:
Please, could you try latest version from today? Thanks

AalianKhan, 2019-01-13:
@ascillato2 I upgraded to 6.4.1.8 today via OTA form the URL (http://thehackbox.org/tasmota/sonoff.bin) but still no luck
I get this when I turn on the dimmer. FYI the dimmer value is 100 at this time.
```
CMD: power1 1
stat/sonoff/RESULT = {"POWER":"ON"}
stat/sonoff/POWER = ON
stat/sonoff/RESULT = {"POWER":"ON","Dimmer":34}
stat/sonoff/RESULT = {"POWER":"ON","Dimmer":51}
stat/sonoff/RESULT = {"POWER":"ON","Dimmer":49}
```

chaosmaster, 2019-01-13:
I remember having looked at this before and wasn't able to get fading to work properly.
My hardware (Touch EU version) has it's on fading implemented in the MCU which seemed to interfere with Tasmota.
This would cause Tasmota to interpret the feedback from the MCU as new commands which broke the current fading operation.
I may have to take another look at this.
Maybe @digiblur could also help debugging this, since he also has the same hardware you have and is familiar with the protocol.
@AalianKhan you can also set _Weblog 4_ to see what exact messages are being transmitted between the MCU and the ESP.

AalianKhan, 2019-01-13:
@chaosmaster Should I post the logs from the MCU here?

AalianKhan, 2019-01-13:
Is it possible to use the wake function instead?

digiblur, 2019-01-13:
Which dimmer is this? Maybe I missed it but didn't see a name at a quick glance.

chaosmaster, 2019-01-13:
@digiblur 

>  Hardware used : _Moes dimmer (Tuya dimmer) connected to a wall socket_

digiblur, 2019-01-13:
Will check it out on mine in a bit. Using a slightly modified fork to fix the dimming and power state reporting to home Assistant with various scenarios of turning on or off, but it should be the same with the base of things pertaining to fading.

digiblur, 2019-01-13:
looks like mine does some sort of random number at times, I tried this on a PWM based dimmer as well and while the number doesn't jump I'm not really seeing a difference with fade on or off in the light turning on or off.  I also tried the fade on another Tuya based dimmer with a segmented number display and it does the same thing.

AalianKhan, 2019-01-13:
Yeah.....at least I think so

michael-bartmann, 2019-01-17:
Got this log the other day, maybe it is related to this issue:

20:45:48 MQT: tele/tuya-1/INFO3 = {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x4000dce5 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}

It was when playing the first time with tasmota settings under 6.3.0, if I remember right I sent "Dimmer 0" and "WakeUp 100" commands before and Tasmota crashed/rebooted.

michael-bartmann, 2019-01-17:
Maybe I have to correct my above statement. The exception happend using

```
Program Version 6.4.1.8(fc1198c-sonoff)
Build Date & Time 2019-01-14T12:13:41
Core/SDK Version 2_5_0_BETA2/3.0.0-dev(c0f7b44)
```

ascillato2, 2019-01-17:
From #4961, @michael-bartmann

When changing dimmer value explicitly (e.g. command "Dimmer 42"), the result message is:

`stat/sonoff/RESULT = {"POWER":"ON","Dimmer":42}`

When changing the power only (e.,g. "Power ON", the same topic is used for the result, but there is no dimmer tag in the Json message:

`stat/sonoff/RESULT = {"POWER":"ON"}`

This is inconsistent and confuses binding the dimmer value to that topic in openHAB, because independent from the specific transformation the result message without the dimmer value triggers an update of the dimmer, but does not have a value, which results in a WARN-logging in the best case. (Worst case would be to overwrite the last dimmer value).

My expectation is that there was a Dimmer tag with the current dimmer value in every message on that topic.

Used hardware: Tuya-compatible EU clone (works more or less, fading issues are already mentioned in another thread).

Used firmware: 6.3.0/2.3.0, 6.4.1/2.3.0, 6.4.1/2.4.x, 6.4.1.8/2.5.0beta2

chaosmaster, 2019-01-17:
@ascillato2
This looks like a different issue not related to the fading.
I believe this is related to what has been discussed here:

https://github.com/arendst/Sonoff-Tasmota/issues/469#issuecomment-441358159

As I understood this was on purpose?

michael-bartmann, 2019-01-18:
I am a bit confused as well. I still believe we have two different issues, both related to tuya dimmer.
I had opened another issue on purpose. But never mind, it is OK for me to discuss it here.

michael-bartmann, 2019-01-18:
There was this question of @digiblur in that closed thread #4961 about setOption.

My opinion is that it would be desirable to always have a message formatted consistently, i.e. with the same Json content/tags.

But what I would understand as a reasonable option: Introducing a specific non-Json topic for dimmer status updates, like it is done for power, e.g.

stat/tuya-1/DIMMER = 42

This would make parsing easy, but not everybody would like to see two messages on each update. So it could be configurable.

As far as my request goes: My openHAB binding would profit from either approach, but consistence message formatting is less confusing, doesn't need configuration and may be a low hanging fruit when it comes to implementation efforts.

digiblur, 2019-01-18:
@michael-bartmann Theo answered in this link: https://github.com/arendst/Sonoff-Tasmota/issues/4961#issuecomment-455483609

While I don't have a dimmer state topic like you suggested but for my small fork I use for my dimmers I have it report the power relay state and dimmer state anytime anything changes with the light to work around the issue I have in Home Assistant.  It made things super simple on implementation for me as well with no crazy templates.

flavorplus, 2019-01-18:
I actually have the EU hardware version and experiencing the same issue with dimming. I think the MCU is messing things up with the reply's it's sending. Did some checking on weblog 4:
<pre><code>15:25:12 CMD: Dimmer 50
15:25:12 SRC: WebConsole from 192.168.50.52
15:25:12 RSL: Received Topic /Dimmer, Data Size 2, Data 50
15:25:12 RSL: Group 0, Index 1, Command DIMMER, Data 50
15:25:12 MQT: stat/casaSupreme/RESULT = {"POWER":"ON","Dimmer":50}
15:25:12 TYA: Send Serial Packet Dim Value=127 (id=2)
15:25:12 TYA: TX Packet: "55aa00060008020200040000007f94"
15:25:12 TYA: RX Packet: "55aa00070008020200040000007f95"
15:25:12 TYA: RX Dim State=127
15:25:12 TYA: RX Packet: "55aa0007000501010001010f"
15:25:12 TYA: RX - On State
15:25:12 TYA: RX Packet: "55aa00070008020200040000007e94"
15:25:12 TYA: RX Dim State=126
15:25:12 CFG: Saved to flash at F6, Count 94, Bytes 3584</code></pre>

So it looks like maybe the second and third really are canceling the fade operation?
Some output when fading is enabled:
<pre><code>15:35:52 CMD: Fade 1
15:35:52 SRC: WebConsole from 192.168.50.52
15:35:52 RSL: Received Topic /Fade, Data Size 1, Data 1
15:35:52 RSL: Group 0, Index 1, Command FADE, Data 1
15:35:52 MQT: stat/casaSupreme/RESULT = {"Fade":"ON"}
15:35:53 CFG: Saved to flash at F6, Count 102, Bytes 3584
15:35:59 WIF: Checking connection...
15:35:59 WIF: Connected
15:35:59 DNS: MDNS.update
15:36:01 CMD: Dimmer 50
15:36:01 SRC: WebConsole from 192.168.50.52
15:36:01 RSL: Received Topic /Dimmer, Data Size 2, Data 50
15:36:01 RSL: Group 0, Index 1, Command DIMMER, Data 50
15:36:01 SRC: Light
15:36:01 TYA: Send Serial Packet Dim Value=25 (id=2)
15:36:01 TYA: TX Packet: "55aa0006000802020004000000192e"
15:36:01 TYA: SetDevicePower.rpower=1
15:36:01 TYA: TX Packet: "55aa0006000501010001010e"
15:36:01 MQT: stat/casaSupreme/RESULT = {"POWER":"ON","Dimmer":50}
15:36:01 TYA: RX Packet: "55aa0007000802020004000000192f"
15:36:01 TYA: RX Dim State=25
15:36:01 TYA: Send CMND_DIMMER_STR=Dimmer 10
15:36:01 SRC: Switch
15:36:01 RSL: Received Topic /Dimmer, Data Size 2, Data 10
15:36:01 RSL: Group 0, Index 1, Command DIMMER, Data 10
15:36:01 MQT: stat/casaSupreme/RESULT = {"POWER":"ON","Dimmer":10}
15:36:01 TYA: RX Packet: "55aa0007000501010001010f"
15:36:01 TYA: RX - On State
15:36:01 TYA: RX Packet: "55aa0007000802020004000000162c"
15:36:01 TYA: RX Dim State=22
15:36:01 TYA: RX Packet: "55aa0007000501010001010f"
15:36:01 TYA: RX - On State
15:36:01 TYA: Send Dim Level skipped due to 0 or already set. Value=5
15:36:01 TYA: Send Serial Packet Dim Value=25 (id=2)
15:36:01 TYA: TX Packet: "55aa0006000802020004000000192e"
15:36:01 TYA: RX Packet: "55aa0007000802020004000000192f"
15:36:01 TYA: RX Dim State=25
15:36:01 TYA: Send Serial Packet Dim Value=25 (id=2)
15:36:01 TYA: TX Packet: "55aa0006000802020004000000192e"
15:36:02 TYA: RX Packet: "55aa0007000802020004000000192f"
15:36:02 TYA: RX Dim State=25
15:36:02 TYA: Send Serial Packet Dim Value=25 (id=2)
15:36:02 TYA: TX Packet: "55aa0006000802020004000000192e"
15:36:02 TYA: RX Packet: "55aa0007000802020004000000192f"
15:36:02 TYA: RX Dim State=25
15:36:02 TYA: Send Serial Packet Dim Value=25 (id=2)
15:36:02 TYA: TX Packet: "55aa0006000802020004000000192e"
15:36:02 TYA: RX Packet: "55aa0007000802020004000000192f"
15:36:02 TYA: RX Dim State=25
15:36:02 TYA: Send Serial Packet Dim Value=25 (id=2)
15:36:02 TYA: TX Packet: "55aa0006000802020004000000192e"
15:36:02 TYA: RX Packet: "55aa0007000802020004000000192f"
[... 41 lines cut ...]
15:36:02 TYA: RX Dim State=25
15:36:02 TYA: Send Serial Packet Dim Value=25 (id=2)
15:36:02 TYA: TX Packet: "55aa0006000802020004000000192e"
15:36:02 TYA: RX Packet: "55aa0007000802020004000000192f"
15:36:02 TYA: RX Dim State=25
15:36:02 TYA: Send Serial Packet Dim Value=25 (id=2)
15:36:02 TYA: TX Packet: "55aa0006000802020004000000192e"
15:36:02 TYA: RX Packet: "55aa0007000802020004000000192f"
15:36:02 TYA: RX Dim State=25
15:36:02 TYA: Send Serial Packet Dim Value=25 (id=2)
15:36:02 TYA: TX Packet: "55aa0006000802020004000000192e"
15:36:02 TYA: RX Packet: "55aa0007000802020004000000192f"
15:36:02 TYA: RX Dim State=25
15:36:02 TYA: Send Serial Packet Dim Value=25 (id=2)
15:36:02 TYA: TX Packet: "55aa0006000802020004000000192e"
15:36:02 TYA: RX Packet: "55aa0007000802020004000000192f"
15:36:02 TYA: RX Dim State=25
15:36:02 TYA: Send Serial Packet Dim Value=25 (id=2)
15:36:02 TYA: TX Packet: "55aa0006000802020004000000192e"
15:36:02 TYA: RX Packet: "55aa0007000802020004000000192f"
15:36:02 TYA: RX Dim State=25</code></pre>

some version info: `{"StatusFWR":{"Version":"6.4.1.9(sonoff)","BuildDateTime":"2019-01-17T19:26:36","Boot":4,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}`

I wonder if this is fixable with the MCU still running the stock firmware.

chaosmaster, 2019-01-18:
@flavorplus that's exactly the issue I was having trying to get fading to work on the Touch EU.
It might still be possible to implement it for other tuya-based dimmers (I don't have other hardware though)

closed 2019-03-11 by stale[bot]