arendst/Tasmota #4912: Wemos D1 mini pro GPIO not triggering when grounded with momentary button

crispy78, 2019-01-13:
I've created a doorbell using a Wemos D1 mini pro but forgot to write down how I've setup Tasmota.
For those interested: 230v AC will be converted to 5v DC using a Hi-Link HLK-PM01, this provides power to the Wemos D1 mini pro and Wemos D1 mini relay shield. The relays will power the 230v AC power supply for the 8v DC doorbell. 

The relays is triggered by D1 (GPIO 5), one end of the outdoor doorbell button will connect D2 (GPIO 4) and the other end to ground.

The way it worked was that the doorbell button was pressed and this would trigger the relay for some time. In my current situation when I power D2 (GPIO4) the relay will turn on, when I ground D2 (GPIO 4) it would turn off the relay. This is not how it worked. Any suggestions?

fluppie, 2019-01-13:
I would suggest you to look after rules: https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#10-simple-staircase-timer

10. Simple Staircase Timer

Rule:

rule1 on button1#state do backlog power1 %value%; ruletimer1 600 endon on rules#timer=1 do power1 off endon

Result:

on button1#state do backlog power1 %value%;

    On Button press the Light in the Staircase will switch on/off

ruletimer1 600 endon

    Additionaly ruletimer1 will begin to countdown 10 minutes

on rules#timer=1 do power1 off endon

    After the ruletimer1 expires the ligth will be switched off (if you forgot to switch it off)

ascillato2, 2019-01-13:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-13 by ascillato2

crispy78, 2019-01-13:
Thank for the feedback.
Any ideas why I need to put power on the switch to turn the relay on and connect it to ground to turn the relay off? I had/have the switch connected to ground and this worked, the hardware hasn’t changed, I only had to flash the device. What could I have forgotten?