arendst/Tasmota #4914: Root cause for Shelly 1 button config issues

tolboel, 2019-01-13:
It seems to me that the root cause of all the issues people (including myself) are having using Tasmota on a Shelly 1 with a push button rather than a switch is the fact that when you push the button it sees the GPIO 5 go HIGH where other devices like Sonoff Basic goes LOW. This effectively prevents me from setting up Tasmota on the Shelly 1 like I would on a nodemcu, Sonoff Basic, Wemos etc. 

Could a "simple" solution be that when you define the device as "Shelly 1" or even "Generic" you should be able to specify GPIO5 as a GPIO_KEY1_INV (similar to being able to specify a relay as GPIO_REL1_INV). 

I ran a simple arduino sketch on a Shelly 1 and it clearly indicated that when my push button is pressed GPIO5 goes HIGH and when the button is open GPIO5 is LOW. 

I - like many others - would like to set up Shelly 1 to toggle on single press and send MQTT on double press or hold. This is possible on all other devices I am using with Tasmota - but not my Shelly 1 devices.

Thank you for a great job!!!! I recommend Tasmota to all my friends and colleges

ascillato2, 2019-01-14:
Hi,

You have the command `switchmode1 6` to transform your input to an INVERTED PUSHBUTTON.

Have you tried that? 

_As a side note:_ `switchmode` _will only work if you have set your inputs as switches and for Tasmota, Shelly 1 and 2 have switches in the template, so this should work_

tolboel, 2019-01-14:
Yes I have tried switchmode1 6 and it gives me single click and hold functionality but not double click as this is not supported for switches - only buttons.

It seems to me that if we were able to invert the input the shelly would work just like any other device and you would not have to make special handling/exceptions for the shelly 1. You could define the input as a button without risking a reset after 40 seconds and all other functionality would be like any other tasmota device.

Throughout my house I have a mix of Sonoffs, nodemcu, wemos and shellys for controlling light etc and I would like to have the same functionality and user experience on all switches no matter what device is hidden in the wall/attic/lamp. I just can't get double-click to work with shelly 1. I have retrofitted my lightswitches with a small spring so they act as pushbuttons (detecting buttonpush with an optocoupler board or directly with the shelly 1).

tolboel, 2019-01-18:
I've noticed that ESPhomelib has an invert parameter for inputs and output. In fact you shall use invert on a Sonoff as a button press pulls GPIO0 low. I find this more logical.

fhloston, 2019-01-18:
I would second that feature. I would like to use click, double click, hold etc on my Gira momentary wall switches for my covers aswell.

ascillato, 2019-01-25:
> I ran a simple arduino sketch on a Shelly 1 and it clearly indicated that when my push button is pressed GPIO5 goes HIGH and when the button is open GPIO5 is LOW.

That is exactly also how Tasmota works. See https://github.com/arendst/Sonoff-Tasmota/blob/1fb7eb646fba2a95aecfc1cc65a9dc705d89731e/sonoff/support_switch.ino#L77

You can try the generic module and set the switch gpio as a button gpio to try double tap feature. As shelly has some noise in the input, you may have some ghost switching, so you can play with buttondebounce command.

tolboel, 2019-01-25:
The source code you are referencing is for a SWITCH, not a BUTTON. It is just not true that the input for a Sonoff Basic and the Shelly 1 is the same. In fact the picture on the Sonoff-Tasmota github page is wrong. See 

"https://github.com/arendst/Sonoff-Tasmota/wiki/Shelly-1-and-2"

This picture says that GPIO5 reads LOW when SW is shorted to AC/L. This is not true, it reads HIGH on all of my Shelly 1 devices (version 1). And this is the root cause of all the problems a LOT of people are having. The button on a Sonoff Basic pulls the GPIO0 pin to LOW but the Shelly 1 GPIO5 is pulled to HIGH. Unless Tasmota can invert the button GPIO pin it will never be possible to configure the Shelly as a button and enable single, double and hold. In fact if you configure the Shelly as Generic and do not Setoption1 1 then Tasmota will think the button is constantly pressed and after 40 seconds it will reset. 

I have done a lot of research and experimentation into this including reading parts of the source. I do not believe there is any way to implement a Shelly1/Tasmota device which is defined as a button and support single, double and hold unless you can configure a GPIO button as inverted. I would love for you to prove me wrong.

Configuring the Shelly 1 as a SWITCH is not a solution either as double click is not supported for switches. Not even using SwitchMode 6

Just out of curiosity I looked at how the ESPHOMELIB works. Below I have copied a small section of the configuration yaml files used to configure a button in ESPHOMELIB on a sonoff and a shelly1:

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      inverted: True
      mode: INPUT
    name: "sonoff Button"

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO5
      inverted: False
      mode: INPUT
    name: "shelly1 Button"

The only difference is the "inverted" True/False. Everything else is configured the same for both devices including relays, sensors, MQTT etc. and so far everything has worked just perfect, even without ghost switching as you mention. As you can see, the people behind ESPHOMELIB have come to realise that there is a difference of how buttons are handled on different devices and inverting the input is the solution. Notice how it is the Sonoff that gets it's input inverted as I guess it is more logical that pulling a GPIO HIGH indicates a button is pressed.

I will be very happy to help you investigate this further and help you test a solution you may come up with.

Thanks for your help.

ascillato, 2019-01-26:
Let's try the following dirty modification for inverted pushbutton:

* FILE _support_button.ino_ L118

`button = (digitalRead(pin[GPIO_KEY1 +button_index]) == 0);`

* FILE _sonoff_template.h_ L1398

`GPIO_KEY1_NP,`

The following precompiled bin is the latest Tasmota but with the button inverted just for testing your shelly 1.

[Tasmota_shelly_test.zip](https://github.com/arendst/Sonoff-Tasmota/files/2798547/Tasmota_shelly_test.zip)

Please try it with your Shelly 1

Thanks.

tolboel, 2019-01-26:
Hola,

PERFECT !!!!
Did the following:
1) flashed the test firmware
2) connected my iphone to the access point, set the SSID/Password
3) Configured: Generic (18), GPIO4 Relay1 (21), GPIO5 Button1n (90)
4) Configured MQTT
5) SetOption1 1
6) SetOption11 1
7) SetOption32 10
8) rule on button1#state=3 do publish cmnd/topicHOLD/power 2 endon on button1#state=2 do publish cmnd/topicDOUBLEPRESS/power 2 endon
9) rule1 1

It works PERFECTLY !!! Single click toggles relay and sends an MQTT message. Double click and Hold sends only MQTT messages as per the rule1. Just like I wanted it.

I suggest that you create new entries in the enum UserSelectablePins:

GPIO_KEY1, GPIO_KEY1_NP, GPIO_KEY1_INV, GPIO_KEY1_NP_INV for each KEY 

and change the Shelly template GPIO5 to GPIO_KEY1_NP_INV 

This would finally enable me to use my shelly 1 devices just the same as my sonoffs and nodemcu/Wemos.

I hope we will see the new invert feature in button pins in a new release soon. If you need my help to test I will be happy to compile and test a future version using the Arduino IDE. Just send me a link to a beta. Thank you for your great effort.

I have written to the Shelly support page to ask them to change the pinout description of the Shelly1 so that the SW input is described correctly.

ascillato2, 2019-01-26:
Great, please, test it for some hours to see if you have ghost switching or not. Thanks.

ascillato2, 2019-01-26:
Also, please test it selecting shelly module. Thanks.

tolboel, 2019-01-26:
So far so good. Have two devices running configured as shelly modules. Both configured as explained earlier but as Shelly1 and connected to my hass.io and MQTT. One has about 5 meter of wiring connected through my house electrical installation (attic and wall) to the SW input and I have not (yet) experienced any ghost switching. Will keep it running and monitor it closely.

ascillato2, 2019-01-27:
Closing this issue as it has been fixed. Thanks.

closed 2019-01-27 by ascillato2

fhloston, 2019-01-27:
> When this is implemented it makes the use of normal switches connected to the shelly(s) impossible.
> 
> The Shelly(s) are designed to add wifi/mqtt to a normal wall switch or blinds switches. Only the current GPIO_SWT1_NP supports both wallswitch and momentary switches as used with blinds. The requested functionality is mutually exclusive to the previous.
> 
> As this is a small user base use-case it still makes sense to add the extra button functionality to the Generic template.

How does this relate to shelly2 used for covers in combination with momentary switches?

ascillato, 2019-01-27:
Just set the gpios according to your needs ;)

arendst, 2019-01-27:
Just use the shelly2 module and configure your inputs as documented in the wiki. No need to use any of the items noted in this issue.

tolboel, 2019-01-27:
I just downloaded and build the latest 6.4.1.11 and installed it on a Shelly1. I think it is perfectly fine that you have to choose Generic to be able to invert the GPIO button input pin and it looks like it is working perfectly.

Now I can continue installing my Shelly devices and have the same "user interface" on all my light switches (that have been converted to buttons with a spring). Single, double and hold is working like a charm. The Shelly 1 is so much easier to hide in my installation (it is smaller) and it is less intrusive in my electrical system as I can continue using my normal wires and switches. All I need to to is insert a spring in the switch. Also means it is easy to remove if I decide to sell my house one day.

Thank you very much. !!!

jaesonmartin, 2019-12-02:
> Yes I have tried switchmode1 6 and it gives me single click and hold functionality but not double click as this is not supported for switches - only buttons.
>
> It seems to me that if we were able to [...]

Hi brother.. Can i know how do you connect 3 nodemcu in a webpage or in a phone app ?? I want to control multiple esp8266(nodemcu) with a webpage or a phone app..

lesourcil, 2020-06-28:
> Just use the shelly2 module and configure your inputs as documented in the wiki. No need to use any of the items noted in this issue.

Hi,  Can you direct me to the documentation you are refering here, because i don't follow. In the module parameters, i have this
[image]

kokoon, 2022-05-17:
> I just downloaded and build the latest 6.4.1.11 and installed it on a Shelly1. I think it is perfectly fine that you have to choose Generic to be able to invert the GPIO button input pin and it [...]

@tolboel could you help me out, I can't get this to work? Does your shelly-1 act on button press or release? I'm losing my mind, I think I've tried every trick in the book and I just can't make mine work as (I think) you have.