arendst/Tasmota #4915: WifiManager / WiFi softAP mode crashes when Tenda Nova MW3 or MW6 in range

araczkowski, 2019-01-13:
This is not a bug in Tasmota and probably not Arduino core too, but maybe Espressif sdk.

Anyway I would like to ask you about advice how this is possible and ideas to bypass this problem.
Thanks in advance!

Like I reported here https://github.com/arendst/Sonoff-Tasmota/issues/4659

I was able to find out that the ESP crashes after a couple of seconds in WIFI_MANAGER mode, after line:
` WiFi.softAP(my_hostname, NULL, channel);`

```
ets Jan  8 2013,rst cause:4, boot mode:(3,4)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v3ffe9818
~ld
```

**this happend whenever a specific mesh wifi ap is in range -> Tenda Nova MW3 or MW6**

**How it is possible that softAP on ESP is crashing by router? The router is sending something to AP and this is the reason of a crash???**

Anybody has any idea how to bypass this?

PS;
I'm not the only one who is facing this problem, please see the issue here:
https://github.com/esp8266/Arduino/issues/4735

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Development/Compiler/Upload tools used : Platform IO
- [X] Hardware used : S20 or S26 or Wemos D1 mini 
- [X ] If a pre-compiled release or development binary was used, which one? : latest stable on core version 2.3.0
- [X ] You have tried latest release or development binaries? : yes
- [] Provide the output of command ``status 0`` :

Jason2866, 2019-01-13:
Thx, for informing. As you already mentioned it is not a bug from Tasmota and nothing can be done here to prevent this. Please close this issuse. So if someone is searching the info will be found.

araczkowski, 2019-01-13:
Hi @Jason2866 please keep this question (not a bug) open for day or two...

Maybe someone has any idea how this is possible, what exactly can causing this kind of problem?
The router is sending something to device when the device is in AP mode?

I just want to understand how this is even possible.

Thanks

ascillato, 2019-01-13:
Please read https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#wifi-issues-arduino-core-versions-and-expressif-sdk
You can try with Core 2.5.0

ascillato, 2019-01-13:
Also, the AP mode is in channel 1. You can choose another channel compiling by yourself changing in my_user_config.h the wifi ap channel setting

ascillato2, 2019-01-13:
Please, try with core 2.5.0

And then, after that test, please try with another channel for the AP mode and report back. Thanks

araczkowski, 2019-01-13:
Hi @ascillato2 

Thanks for prompt response :)

I can already confirm that it is ok with core 2.5.0, this issue is only on 2.3.0

**Do you have any clue why router can break AP?**

Ps
I will try another chanel on 2.3.0 and report back.
I'm doing this because in my experience 2.3.0 hast best WiFi range on local network ( I have separated network without access to Internet for iot, so all I care is a stable and full range WiFi)

ascillato, 2019-01-14:
> I can already confirm that it is ok with core 2.5.0, this issue is only on 2.3.0

That is a really good news!!! Thanks for sharing :+1: 

> Do you have any clue why router can break AP?

May be it was because they were in the same channel your router AP and your Tasmota AP (1 is the default in Tasmota for Soft AP) so they interfere each other. That issue happened before in #4673 and Theo added the user configuration for Soft AP Channel in my_user_config.h

> I'm doing this because in my experience 2.3.0 hast best WiFi range on local network

Yes, well always will depend on your wifi setup, distances, obstacles, interferences, other wifi networks, amount of devices, etc etc.

In my case, the 3 core versions (2.3.0, 2.4.2 and 2.5.0) have worked the same. Some reports a slightly different RSSI but in the practice was the same. The devices (for testing) with less than 20% signal (far from router) continue having obvious disconnections and the ones with signal greater than 35% have zero or nearly zero disconnections. So, you have to choose the core that works better in your wifi setup.

araczkowski, 2019-01-14:
Hi @ascillato 

bingo :+1: changing a channel can solve the problem :)

Last question concerning this topic, do you think it is possible to figure out in the code (on device) which channel is free for softAP? Just to prevent this kind of problems. Or maybe I should try in loop 1..13 

Thanks again for your help!

I will play with this a little bit more to fully understand this topic but of course the issue can be closed.

PS 
the detail description below

I have quite strong WiFi signal from my mesh WiFi
[image]

for the test I hardcoded the channel to 13:

[image]

now for the first time I'm able to see the WifiManager on 2.3.0  

[image]

closed 2019-01-14 by araczkowski