arendst/Tasmota #4916: Permanently reconnect mqtt Gosund SP1 with hardware v24

hrahlers, 2019-01-13:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
_A clear and concise description of what the bug is._

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ ] _Searched the problem in issues and in the wiki_
- [ ] _Hardware used_ : 
- [ ] _Development/Compiler/Upload tools used_ :
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [ ] _You have tried latest release or development binaries?_ :
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```

**To Reproduce**
_Steps to reproduce the behavior:_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-01-13:
Please provide Status 0 output. Without we cant help and issue will be closed

hrahlers, 2019-01-13:
Thanks for your reply. In that moment I installed on one device the core version 2.3.0 and the sonoff-DE.bin from here (http://thehackbox.org/tasmota/release/020300/). Is there anywhere a German Version of the 2.5.0 .bin file?

ascillato2, 2019-01-13:
Sorry, there isn't because it is aimed for testing. Please, try it anyway to confirm if solves your issue. Then, after that, if works you can go to try to compile by yourself the version you need. All that is explained in the wiki. Compiling by yourself is not easy at first. So, if you need assistant with that, you can find us also in the Tasmota Support Chat. Thanks

ascillato2, 2019-01-13:
Link to latest Tasmota with core 2.5.0: http://thehackbox.org/tasmota/020500/

ascillato2, 2019-01-13:
### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-13 by ascillato2

hrahlers, 2019-01-17:
Ok. I am pretty new in using Tasmota an GitHub. By taking the 2.3.0 core version I could solve my problem but with 2.5.0 there has been no changes: further disconnect messages. Of course it makes sense that a trial version is only available in English. It could have been possible that the German version hides somewhere. Thanks for your help!