arendst/Tasmota #4918: Wrong sonoff recognition for TH16

nicolasvac, 2019-01-13:
**Describe the bug**
The firmware recognise the sonoff th 16 as basic one, but it isn't.

My sonoff is TH16 Rev 1.2

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [X] _Searched the problem in issues and in the wiki_
- [X] _Hardware used_ : Sonoff TH 16 Rev 1.2
- [X] _Development/Compiler/Upload tools used_ : ESPFlashEasy
- [X] _If a pre-compiled release or development binary was used, which one?_ :  Tasmota 6.4.1 classic and then sensors
- [X] _You have tried latest release or development binaries?_ : Yes, 6.4.1
- [X] _Provide the output of command_``status 0`` :
```
16:53:23 RSL: stat/sonoff/STATUS = {"Status":{"Module":4,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:53:23 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-sensors.bin","RestartReason":"Software/System restart","Uptime":"0T00:08:44","StartupUTC":"2019-01-13T15:44:39","Sleep":50,"BootCount":11,"SaveCount":19,"SaveAddress":"F5000"}}
16:53:23 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sensors)","BuildDateTime":"2018-12-24T14:45:55","Boot":7,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
16:53:23 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["VACCARI",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
16:53:23 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":550,"Free":452,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1420C2","FlashMode":3,"Features":["00000809","0FDEE794","0003A3A4","B7FFBFCC","0002BBC0"]}}
16:53:23 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-6392","IPAddress":"192.168.1.23","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.231","Mac":"80:7D:3A:31:58:F8","Webserver":2,"WifiConfig":4}}
16:53:23 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3158F8","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
16:53:23 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sun Jan 13 15:53:23 2019","Local":"Sun Jan 13 16:53:23 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:39","Sunset":"17:17"}}
16:53:23 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-01-13T16:53:23"}}
16:53:23 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-01-13T16:53:23","Uptime":"0T00:08:44","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"VACCARI","BSSId":"A4:91:B1:2A:8A:0F","Channel":1,"RSSI":32}}}
```

**To Reproduce**
Flash Tasmota 6.4.1 classic on to Sonoff TH 16
Then, after install flash Tasmota sensors 6.4.1

**Expected behavior**
The system should recognise the module as it is.

**Screenshots**
[image]

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

nicolasvac, 2019-01-13:
Note that i've changed it manually

Jason2866, 2019-01-13:
That is the way it works :-) You have to select the module manually.

nicolasvac, 2019-01-13:
Oh, sorry. I tought that the module will be auto recognized when i saw the label for the input saying (Sonoff basic)

closed 2019-01-13 by ascillato2