arendst/Tasmota #4926: Sonoff Pow discovery in Home Assistant

hlugt, 2019-01-14:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
_A clear and concise description of what the bug is._
To enable discovery of Sonoff Pow as switch instead of light settings command setOption30 (currently on default 1, which by the way is described as default should be 0?)  to 0 to identify as switch is not echoed in console and leads to restart.
To enable discovery of Sonoff Pow command setOption19 (currently on default 0)  to 1 to enable discovery is not echoed in console and leads to restart.
It happens on both release 6.4.1 (2.4.2) and latest 6.4.1.8 (2.5.0)
On Sonoff Basic no issues.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x ] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : Itead Sonoff Pow R1  
- [ ] _Development/Compiler/Upload tools used_ : download compiled versions ota and local upload
- [x] _If a pre-compiled release or development binary was used, which one?_ : 6.4.1 from release and 6.4.1.8 from dev 020500.
- [x ] _You have tried latest release or development binaries?_ : yes
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
16:02:46 CMD: status 0
16:02:46 MQT: stat/sonoff-p03/STATUS = {"Status":{"Module":6,"FriendlyName":["SonoffPow03"],"Topic":"sonoff-p03","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:02:46 MQT: stat/sonoff-p03/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software Watchdog","Uptime":"0T00:07:18","StartupUTC":"2019-01-14T14:55:28","Sleep":50,"BootCount":55,"SaveCount":219,"SaveAddress":"F8000"}}
16:02:46 MQT: stat/sonoff-p03/STATUS2 = {"StatusFWR":{"Version":"6.4.1.8(fc1198c-sonoff)","BuildDateTime":"2019-01-14T12:13:41","Boot":31,"Core":"2_5_0_BETA2","SDK":"3.0.0-dev(c0f7b44)"}}
16:02:46 MQT: stat/sonoff-p03/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["LOCAL2","LOCAL"],"TelePeriod":60,"SetOption":["54200009","558180C0","00000001"]}}
16:02:46 MQT: stat/sonoff-p03/STATUS4 = {"StatusMEM":{"ProgramSize":547,"Free":456,"Heap":19,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640E0","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
16:02:46 MQT: stat/sonoff-p03/STATUS5 = {"StatusNET":{"Hostname":"sonoff-p03-6038","IPAddress":"192.168.172.143","Gateway":"192.168.172.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.172.1","Mac":"2C:3A:E8:07:57:96","Webserver":2,"WifiConfig":2}}
16:02:46 MQT: stat/sonoff-p03/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.172.80","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_07579696","MqttUser":"mqtt","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
16:02:46 MQT: stat/sonoff-p03/STATUS7 = {"StatusTIM":{"UTC":"Mon Jan 14 15:02:46 2019","Local":"Mon Jan 14 16:02:46 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:44","Sunset":"16:55"}}
16:02:46 MQT: stat/sonoff-p03/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":3000,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
16:02:46 MQT: stat/sonoff-p03/STATUS10 = {"StatusSNS":{"Time":"2019-01-14T16:02:46","ENERGY":{"TotalStartTime":"2018-12-30T22:00:21","Total":9.345,"Yesterday":0.530,"Today":0.002,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":227,"Current":0.000}}}
16:02:46 MQT: stat/sonoff-p03/STATUS11 = {"StatusSTS":{"Time":"2019-01-14T16:02:46","Uptime":"0T00:07:18","Vcc":3.510,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"ON","Wifi":{"AP":2,"SSId":"LOCAL","BSSId":"38:2C:4A:E2:F4:10","Channel":7,"RSSI":56}}}
```

**To Reproduce**
_Steps to reproduce the behavior_ : Give setOption30 0 or setOption19 1 on command line in console.

**Expected behavior**
_A clear and concise description of what you expected to happen._ : Expect command echoed in display and set.

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-01-14:
Try SetOption59 1

ascillato2, 2019-01-14:
Those commands requires a restart. So, they are not populating info in the console. It is designed that way not a bug.

closed 2019-01-14 by ascillato2

hlugt, 2019-01-14:
> Those commands requires a restart. So, they are not populating info in the console. It is designed that way not a bug.

Are you sure? In the Sonoff Basic I do see the info populated. Beside: after the restart the change is not honoured.

hlugt, 2019-01-14:
> Try SetOption59 1

Yes done: no difference.

hlugt, 2019-01-14:
Ok, I see ascillato2 already closed this issue. Ok, rather unsatisfactory in my opinions as the behaviour is not in line of what happens with a Sonoff Basic, and I would expect it to be the same...
Disappointing, but as there is a workaround publishing a mqtt message although a bit tedious...

RevelRob, 2019-01-15:
Hi, I updated to the most recent tasmota from today and am having the same issue as described. When trying to SetOption19 1, it hangs and restarts. SetOption is not honoured after restart. Happens only with Sonoff S31 or POW. Basic and S2X do not hang and option is honoured.

RevelRob, 2019-01-15:
Ok, I think I figured it out.
I updated to 6.4.1.9, did a reset 5 (erase flash but keep wifi), Set serial log to 0 (Configuration > Configure Logging), Changed "Full Topic" in MQTT settings to "%topic%/%prefix%/", then "SetOption19 1" in console, then "PowerRetain 1" in console. So far, everything seems to be working and Hass.io discovers all of my switches.

martikainen87, 2019-01-15:
I can just confirm that I had the same issue, installed 4 new blitzwolf SHP6 today, with the 6.4.1.9 firmware and it seems that "SetOption19 1" doesnt change to the correct order of the topic "%topic%/%prefix%/" to enable auto-discovery correctly.

ascillato, 2019-01-15:
@martikainen87 

Sorry, I can't reproduce your issue

Have you done a full erase before flashing? Seems that there were left some incorrect data in your flash

Jason2866, 2019-01-15:
@ascillato 
Yes did a reset 5. Was in Discord...

ascillato, 2019-01-15:
@RevelRob 

> When trying to SetOption19 1, it hangs and restarts.

That is a bad flashing issue. Sorry.

> I updated to 6.4.1.9, did a reset 5

That is one of the ways of solving a bad flashing.

The reset 5 was the solution for your case. The rest was not necessary.

ascillato, 2019-01-15:
@Jason2866 So, may be a full erase with esptool.py is needed for @martikainen87 issue.

martikainen87, 2019-01-16:
Full erase with esptool was executed before flashing the devices, all of them.  Then I did that RevelRob described, upgraded to 6.4.1.9 (core 2.5.0) and did a reset 5. Even tried it 2 times. 

I am still having issues with them, I thought the fix above solved it, but it didn't. So I'll try a new flash tomorrow. (I'm not getting all the sensors in auto discovery, and I can't control them from HA). I'll update with that I find :)

martikainen87, 2019-01-17:
Just reflashed one of them, with a full erase using 
```bash
esptool.py --port COM6 erase_flash
```

I flashed it with the precompiled sonoff.bin from http://thehackbox.org/tasmota/020500/
Configured module to blitzwolf SHP module
Configured name
configured MQTT
"SetOption19 1"

Auto-discovery only got 2 devices (switch and one sensor) and didn't respond to on/off through HA, nor did the switch update if I switched it on/off through tasmota webgui.

Changed module to Sonoff-basic, issued "SetOption 19 1" and then changed back module to blitzwolf, all sensors showing up and commands working! 
Tried a bit different now, did a restart after first initial config with setoption19 as blitzwolf module, same issue. Then switched to sonoff-basic -> setoption... Success!

So there seems to be something that does not execute properly when u issue a setoption19 with blitzwolf module choosen. All 4 of my blitzwolfs are now working with the above routine.

martikainen87, 2019-01-17:
2 off the plugs are having random restarts. 

```bash
18:29:08 MQT: sonoff_pow/tele/SENSOR = {"Time":"2019-01-17T18:29:08","ENERGY":{"TotalStartTime":"2019-01-17T18:13:00","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":3,"ApparentPower":33,"ReactivePower":32,"Factor":0.10,"Voltage":213,"Current":0.153}}
18:29:11 WIF: Connecting to AP1 Martikainen in mode 11N as sonoff_pow-5320...
18:29:13 RSL: sonoff_pow/tele/SENSOR = {"Time":"2019-01-17T18:29:13","ENERGY":{"TotalStartTime":"2019-01-17T18:13:00","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":1,"ApparentPower":31,"ReactivePower":31,"Factor":0.02,"Voltage":212,"Current":0.146}}
18:29:15 WIF: Connected
18:29:15 MQT: Attempting connection...
18:29:15 MQT: Connected
18:29:15 MQT: sonoff_pow/tele/LWT = Online (retained)
18:29:15 MQT: sonoff_pow/cmnd/POWER = 
18:29:16 MQT: sonoff_pow/tele/SENSOR = {"Time":"2019-01-17T18:29:16","ENERGY":{"TotalStartTime":"2019-01-17T18:13:00","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":8,"ApparentPower":23,"ReactivePower":22,"Factor":0.35,"Voltage":212,"Current":0.110}}
```

Tried a new upgrade, reset 5 and now issued "sleep 0" to see if it improves.

ascillato, 2019-01-21:
@martikainen87

https://github.com/arendst/Sonoff-Tasmota/issues/4988