arendst/Tasmota #4927: Multi-trigger PowerDelta Rule - only one trigger firing

meingraham, 2019-01-14:
@ascillato @ascillato2 - Opening a new issue to continue trying to find a solution to my specific scenario. I am adapting the example you provided in #4905.

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- Development/Compiler/Upload tools used: Arduino IDE
- Hardware used: Sonoff S31
- [ ] If a pre-compiled release or development binary was used, which one?:
- [ ] You have tried latest release or development binaries?:

``status 0`` :
```
/STATUS = {"Status":{"Module":41,"FriendlyName":["Power Monitor"],"Topic":"power_monitor","ButtonTopic":"0","Power":1,"PowerOnState":1,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"9T20:32:24","StartupUTC":"2019-01-04T18:25:29","Sleep":0,"BootCount":19,"SaveCount":137,"SaveAddress":"F9000"}}
/STATUS2 = {"StatusFWR":{"Version":"6.3.0.14(sonoff)","BuildDateTime":"2018-11-30T13:36:30","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":1,"LogHost":"myHost","LogPort":514,"SSId":["mySSID",""],"TelePeriod":300,"SetOption":["00008129","558180C0","000000C0"]}}
/STATUS4 = {"StatusMEM":{"ProgramSize":435,"Free":568,"Heap":16,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","0F002790","00000001","0000009E","000000C0"]}}
/STATUS5 = {"StatusNET":{"Hostname":"power_monitor-0656","IPAddress":"192.168.1.nnn","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"<myMAC>","Webserver":2,"WifiConfig":4}}
/STATUS6 = {"StatusMQT":{"MqttHost":"myHost","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_<myLowerMAC>","MqttUser":"myID","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
/STATUS7 = {"StatusTIM":{"UTC":"Mon Jan 14 14:57:53 2019","Local":"Mon Jan 14 09:57:53 2019","StartDST":"Sun Mar 10 02:00:00 2019","EndDST":"Sun Nov 03 02:00:00 2019","Timezone":99,"Sunrise":"07:34","Sunset":"17:37"}}
/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
/STATUS10 = {"StatusSNS":{"Time":"2019-01-14T09:57:53","ENERGY":{"TotalStartTime":"2018-11-30T12:07:13","Total":235.700,"Yesterday":0.035,"Today":0.014,"Power":2,"ApparentPower":5,"ReactivePower":5,"Factor":0.33,"Voltage":122,"Current":0.043}}}
/STATUS11 = {"StatusSTS":{"Time":"2019-01-14T09:57:53","Uptime":"9T20:32:24","Vcc":3.166,"LoopSet":50,"LoadAvg":31,"POWER":"ON","Wifi":{"AP":1,"SSId":"mySSID","BSSId":"<myBSSId>","Channel":1,"RSSI":66}}}
```

I'm not clear on what `Rule1 5` ("Enable one-shot detection (Perform commands once until trigger is not met)") does exactly.

I'm looking to implement a variation of your scenario:

My rule:
```
Rule1 ON Energy#Power<3 DO PowerDelta 0 ENDON ON Energy#Power>75 DO PowerDelta 2 BREAK ON Energy#Power>2 DO PowerDelta 5 ENDON
```

Note that I am using `BREAK` so that if the Power>75 is triggered, the Power>2 won't also trigger.

I want the rule to trigger only once for when Power<3 until Power>75 (and then only once) or  until Power>2 (and then only once). So, I want it to trigger each time one of the triggers is met, but only once until one of the other conditions occur to trigger the rule.

Is this what `Rule1 5` will does?

The rule seems to trigger (each time) when Power<3 threshold is _**crossed**_ (`PowerDelta 0`).
```
ENERGY#POWER<3 performs "PowerDelta 0"
stat/sonoff-washer_monitor/RESULT = {"PowerDelta":0}
```
However, when  power was greater than 2 (e.g., `Teleperiod` message with power=4), the rule (`PowerDelta 5`) did not trigger. However, when power went back under 3, the rule (`PowerDelta 0`) triggered once again.

I also tried putting the `Energy#Power<3` trigger at the end.

```
Rule1 ON Energy#Power>75 DO PowerDelta 2 BREAK ON Energy#Power>2 DO PowerDelta 5 ENDON ON Energy#Power<3 DO PowerDelta 0 ENDON
```

 The result was the same.

Is the [use of sensor values in Rule triggers](https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#sensors) case sensitive?

> All connected sensors can be a trigger in the form **_as they are represented_** in the teleperiod or Status 8 JSON message

The SENSOR and SNS messages

`/SENSOR = {"Time":"2019-01-14T09:48:20","ENERGY":{"TotalStartTime":"2018-11-30T12:07:13","Total":235.699,"Yesterday":0.035,"Today":0.013,"Period":0,"Power":1,"ApparentPower":31,"ReactivePower":31,"Factor":0.04,"Voltage":122,"Current":0.251}}`

`/STATUS8 = {"StatusSNS":{"Time":"2019-01-14T09:56:40","ENERGY":{"TotalStartTime":"2018-11-30T12:07:13","Total":235.700,"Yesterday":0.035,"Today":0.014,"Power":2,"ApparentPower":4,"ReactivePower":3,"Factor":0.49,"Voltage":122,"Current":0.031}}}`

`ENERGY` is all uppercase and `Power` is proper case. Do the `ON` triggers need to evaluate the parameters exactly as they are represented by TASMOTA in the JSON payload? Thus the trigger should be `ENERGY#Power`?

Also, separately (related?), why is `Period` in the `SENSOR` message coming back as 0 (zero) always? My `TelePeriod` is set to 300.

Thanks.

Mike

ascillato2, 2019-01-14:
> I'm not clear on what Rule1 5

lets put an example

```
Rule1 1
Rule1 5
Rule1 on AM2301#Temperature>20 do publish stat/sonoff/temp %value% endon
```

```
Temp is 10 - No Tasmota Rule Execution (trigger condition is NOT met)
Temp is 25 - Tasmota Rule Execution (trigger condition IS met and the ONCE internal flag is set): Sends mqtt msg with temp=25
Temp is 26 - No Tasmota Rule Execution (trigger condition is met, but was executed before, so do nothing)
Temp is 18 - No Tasmota Rule Execution (trigger condition is NOT met and the ONCE internal flag is cleared)
Temp is 25 - Tasmota Rule Execution (trigger condition IS met): Sends mqtt msg with temp=25
```

> Is the use of sensor values in Rule triggers case sensitive?

No

meingraham, 2019-01-14:
Thanks - I think I finally understand the operating principle of `Rule1 5`.

So, the question now is why the Rule isn't firing the other triggers.

```
Rule1
 ON ENERGY#Power>75 DO PowerDelta 2 BREAK
 ON ENERGY#Power>2 DO PowerDelta 5 ENDON
 ON ENERGY#Power<3 DO PowerDelta 0 ENDON
```
This is what I thought should happen:

- `ENERGY#Power = 1` - TASMOTA Rule Executed (trigger condition IS met and the ONCE internal flag is set): `PowerDelta is set to 0`
- `ENERGY#Power = 2` - TASMOTA Rule not Executed (trigger condition is met, but was executed before): `Do nothing`
- `ENERGY#Power = 4` - TASMOTA Rule Executed (trigger condition IS met and the ONCE internal flag is set): `PowerDelta is set to 5`
- `ENERGY#Power = 50` - TASMOTA Rule not Executed (trigger condition is met, but was executed before): `Do nothing`
- `ENERGY#Power = 0` - TASMOTA Rule Executed (trigger condition IS met and the ONCE internal flag is set): `PowerDelta is set to 0`
- `ENERGY#Power = 80` - TASMOTA Rule Executed (trigger condition IS met and the ONCE internal flag is set): `PowerDelta is set to 2`
  - Even though `ENERGY#Power>2` condition is also met but this **_does not trigger_** because the `ENERGY#Power>75` trigger executes a `BREAK` so `ENERGY#Power>2` does not get to trigger.

What am I missing?

What do you think is going on with the `Period=0` always?

Thanks

Mike

ascillato2, 2019-01-14:
This 2 rules are contradictory:

```
ON ENERGY#Power>2 DO PowerDelta 5 ENDON
ON ENERGY#Power<3 DO PowerDelta 0 ENDON
```

Both will try to be triggered. If you have power=2.5 both will trigger.

> I want the rule to trigger only once for when Power<3 until Power>75 (and then only once) or until Power>2 (and then only once). So, I want it to trigger each time one of the triggers is met, but only once until one of the other conditions occur to trigger the rule.

So,

If power is more than 75 you want powedelta 2
If power is between 2 and 75 you want powerdelta 5
If power is lower than 2 you want powerdelta 0

is this correct?

If so, you indeed need Break command in rules.

So, in pseudo code will be something like:

```
IF POWER>75 DO
  POWERDELTA 2
ELSE IF POWER>2 DO
   POWERDELTA 5
ELSE
  POWERDELTA 0
END IF
```

So, in rules could be something like:

```
Rule1 1
Rule1 5
Rule1 ON ENERGY#Power>75 DO PowerDelta 2 BREAK ON ENERGY#Power>2 DO PowerDelta 5 BREAK  ON ENERGY#Power<2 DO PowerDelta 0 ENDON
```

meingraham, 2019-01-14:
I was not considering that the Rules were using real numbers. I was treating them as Integers especially since Rules don't support >= & <=

In your suggested solution, will either trigger fire if Power=2.0? Is this (floating point values) valid syntax for the trigger portion?

`ON ENERGY#Power>2.5 DO`

This is what I'm trying to accomplish:
If Power >= 75 DO PowerDelta 2
Else If Power >= 3 && Power < 75 DO PowerDelta 5
Else DO PowerDelta 0

So, the example you have does what I need with the minor adjustment of a threshold of 3 instead of 2. I'll give it a try and if it works as desired, I'll close this issue.

Thanks.

meingraham, 2019-01-14:
OK, I had in my mind that Rules were integer comparisons. Learn something new about TASMOTA all the time!

However, now the rule
```
Rule1 ON ENERGY#Power>75 DO Backlog PowerDelta 2; Status 8 BREAK ON ENERGY#Power>3 DO Backlog PowerDelta 5; Status 8 BREAK ON ENERGY#Power<3 DO PowerDelta 0 ENDON
```

 isn't triggering at all.

ascillato2, 2019-01-15:
Ok, so troubleshooting questions:
* Have you done Rule1 1?
* Are you using latest Tasmota version from development branch? 
* There is any message in the console?

meingraham, 2019-01-15:
Yes, I have the Rule on.

The is no output in the Console WebLog other than the periodic STATE and SENSOR messages.

I'm at 6.3.0.14 (self compiled with 2.3.0).

I went back to a version of the rule where only one trigger has a BREAK directive and the rule is triggering. OK, it's still not completely doing what I want, but it fires. I went back to a Rule with more than one BREAK and it quits firing. **_I wonder if the multiple BREAK directives are causing some issues with the Rules engine?_**

Mike

P.S. I'm quite hesitant to go to 6.4.1.x because I'd be exposing myself to potentially having to first troubleshoot the right ESP Core to use based on all the issues people are having.

P.P.S. The SENSOR JSON payload always has Power as an integer value. You are certain that inside the software the Rule trigger value comparison is being performed against floating point values for Power? It is the fact that the JSON has only integers for Power as the reason I used integers in the first version of my rule. Thinking of those whole numbers as floating point when writing the Rules logic is a bit non-intuitive. If it's truly the case that Rules are working against "internal" values that may not correspond to the printed log values, the Rules wiki probably needs to point this out in big bold letters.

This is what I was working toward thinking that ENERGY#Power was strictly an integer value as the JSON payload implies:
```
IF POWER>74 DO    // Power has to be 75 or greater
  POWERDELTA 2
ELSE IF POWER>2 DO    // Power has to be 3 or greater
  POWERDELTA 5
ELSE IF POWER<3    // Power has to be 2 or smaller
  POWERDELTA 0
END IF
```

ascillato2, 2019-01-15:
> I'm at 6.3.0.14 (self compiled with 2.3.0).

It is an old version. Sorry. The BREAK command was added in 6.3.0.15

You should update.

> I'm quite hesitant to go to 6.4.1.x

If you don't want to use core 2.4.2 or 2.5.0, there is no problem, you can use core 2.3.0 with latest Tasmota. There is no issue. The version 6.4.1 with core 2.3.0 is available for download at http://thehackbox.org/tasmota/release/020300/

Or you can just selfcompile.

> You are certain that inside the software the Rule trigger value comparison is being performed against floating point values for Power?

It is TEXT comparison, so it will work. There is no internal variables or something weird. Just text comparison. For Example, VARs MEMs and all that are also TEXT. When you use ADD Tasmota transform it to number, do the math and put as text again. So you are free to use VARs as whatever you want.

Please, update and try again.

ascillato2, 2019-01-15:
I'm closing this issue due to it isn't a Tasmota Bug, but we will continue with this thread until you make it work. Hope it is ok for you.

closed 2019-01-15 by ascillato2

meingraham, 2019-01-15:
`The BREAK command was added in 6.3.0.15` - Voila! I wondered about this... but when I didn't get any errors when I added the Rule, I thought I had a version that supported that directive.

I'll try to get the S31 to take an OTA. I'll close this issue for now.

meingraham, 2019-01-15:
Finally got the S31 to successfully OTA. Now at 6.4.1 (release) with 2.3.0.

Still had to play a little with the Rule to get the desired behavior I outlined in the pseudo code example. For the sake of documenting the outcome:
```
Rule1 ON ENERGY#Power>74.99 DO Backlog PowerDelta 2; Status 8 BREAK ON ENERGY#Power>2.99 DO Backlog PowerDelta 5; Status 8 BREAK ON ENERGY#Power<3.00 DO PowerDelta 0 BREAK ON ENERGY#Power=3.00 DO PowerDelta 0 BREAK
```