arendst/Tasmota #4928: Help required for repeating Switch action with Pulsetime / Rules

tango156157, 2019-01-14:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used : NodeMCU uploader tool
- [X ] Hardware used : Wemos D1 Mini
- [ X] If a pre-compiled release or development binary was used, which one? : Pre Compiled (Tasmota 6.4.1)
- [ ] You have tried latest release or development binaries? : Yes
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

Hi There, I am trying to achieve below from my TASMOTA setup running on Wemos D1 Mini

I want to achieve following from my setup:

1. Configure a relay which has a "ON" duration of 2 seconds. I can do this via Pulsetime command, which works, so relay is Switched on briefly for 2 seconds and then goes to OFF condition in 2 seconds. I have a momentary switch also connected when pushed switches on the relay for 2 seconds.

2. I want to repeat above action continuously till time my Switch is ON so relay switches On for 2 seconds then sleeps for 5 minutes and again triggers for 2 seconds and so on and when Switch is OFF so this 5 minute cycle should stop.

I have tried and failed miserably to achieve this via Rules.....Looking out for forum members to help me out here.... Screenshots of my setup below.

Many Thanks in advance..
[image]
[image]

Jason2866, 2019-01-14:
For help please visit us at Discord chat https://discord.gg/Ks2Kzd4 
Because github is for issues (errors). It is not a forum

ascillato, 2019-01-15:
Another approach:

Cleaning:
`backlog rule1 0; rule2 0; rule3 0; rule2 "; rule3 "`

To set the relay2 to turn off after 2 seconds of being ON:
`pulsetime2 2`

Rules:
```
rule1 on power1#state=1 do backlog power2 1; ruletimer1 4 endon on power1#state=0 do backlog ruletimer1 0; power2 0 endon on rules#timer=1 do backlog power2 1; ruletimer1 4 endon
```