arendst/Tasmota #4929: cmnd/xxx/POWER = at DUAL R2 or else

BigWumpus, 2019-01-14:
**Have you look for this feature in other issues and in the wiki?**

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

BigWumpus, 2019-01-14:
After connection, TASMOTA sends a cmnd-string.
I think this are the datapoints for the mqtt-adapter, the user can work with.
I a DUAL R2 there are two relais to handle, so I must create POWER1 and POWER2 manual.
Could it be better to trasnmitt all the possible datapoints to the mqtt-server on startup?

Jason2866, 2019-01-14:
I didnt understand what you mean / want completly
Have you read https://github.com/arendst/Sonoff-Tasmota/wiki/Commands?
There will no change in behaviour at startup and mqtt handling

ascillato2, 2019-01-14:
If you need help or support on understanding Tasmota, you can read the wiki and use the chat also. Github issues are for bug reporting. Thanks

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-14 by ascillato2