arendst/Tasmota #4930: Sleep 255 or 250 docs say 2 things

ezraholm50, 2019-01-14:
Hello,

Playing around with the new dynamic sleep settings. Now i see this wiki page show the max of 255ms

https://github.com/arendst/Sonoff-Tasmota/wiki/Dynamic%20Sleep

And this one not mentioning 255 but 250. Also i cant set anything higher then 250 via console. 

https://github.com/arendst/Sonoff-Tasmota/wiki/Energy-Saving

I guess it should be updated?

ascillato2, 2019-01-14:
It is 250ms.

Wiki Fixed. Thanks.

About the rest of sensors and devices with sleep:

If you use fixed sleep. You will see that goes to 0 automatically when using some devices like LED STRIP WS2812B. And also you will notice some lag on response if sleep if high.

If you use dynamic sleep you don't need to worry. It will be taken care automatically.

closed 2019-01-14 by ascillato2

ezraholm50, 2019-01-14:
Great, i'm using dynamic on all my 30+ devices with sleep 250. Should make a difference!
Thanks!

ascillato, 2019-01-14:
Sleep 250 is very high. Remember that you can experience lag in response. The recommended is dynamic sleep to 50 (default).

With just sleep to 1 the energy consumption goes very low. Then going up from that value, it isn't too much the difference. Remember that the lower consumption is using core 2.3.0 or 2.5.0 (2.4.2 has a CPU sleep but it doesn't have WIFI sleep - The radio stays ON all the time)

ezraholm50, 2019-01-21:
Great, will try 50, its laggy indeed. And im on 2.3, thanks for the tip.