arendst/Tasmota #4933: MAX_RULE_VARS and (maybe) MAX_RULE_MEMS limits.

benzino77, 2019-01-15:
**Have you look for this feature in other issues and in the wiki?**
yes.
**Is your feature request related to a problem? Please describe.**
Starting from version 6.1.1.9 (this [commit](https://github.com/arendst/Sonoff-Tasmota/commit/6a5c0d29e87e400c1dc8c6aa2df1b0954faa6bd4)) there was a possibility to define own `MAX_RULE_VARS` and `MAX_RULE_TIMERS` in `user_config_override.h`. So one could (reasonably, having in mind that it consumes memory) define his own values for those parameters. Starting from one of the latest dev release (this [commit](https://github.com/arendst/Sonoff-Tasmota/commit/1d23c0b0c5df2e696a86806868f833ed10f0b27b)) there is new rules trigger introduced and a limit to have maximum 8 MEMS or VARS.  

**Describe the solution you'd like**

**_I would like to have the ability to define 16 VARS without compiler errors._** 
I had a conversation with Adrian on Discord, and he explained me that changing the limit for VARS could be rather simple, and it will consume another byte of memory (`uint16_t` instead of `uint8_t` for `vars_event`). He also, very clearly, explained me that changing `MAX_RULE_MEMS` is not so simple (because there have to be reserved another byte in `EEPROM`, which can leads to compatibility problems during update) and only @arendst can make that decision and decide how to do it - so I'm asking Theo to consider that also.

**Describe alternatives you've considered**
I don't have one.

**Additional context**
Significant majority of Tasmota users, just using precompiled bins. Those who compile the binaries on their own, realize that some changes can lead to larger memory consumption (changing number of vars, rule timers, etc.)... at least I want to believe in that ;)

**(Please, remember to close the issue when the problem has been addressed)**
I will.

arendst, 2019-01-15:
Ok got it. It was set to 8 for both Vars and Mems as Adrian said because of the uint8_t. Mems will stay at 5 as they are stored in flash using 10 bytes each as a number is stored as text to ease string compares.

 I'll change Vars to max 16 using uint16_t.

benzino77, 2019-01-15:
Great! Thanks Theo!
Question: Why have you changed MAX_RULE_MEMS to 5 from 8? Was it a mistake and it should be 5 from the beginning, because there are 50bytes reserved in flash for that or there is other reason?

closed 2019-01-15 by benzino77

benzino77, 2019-01-15:
Thanks again for quick answer! 
One more question: what will happen if I change the `MAX_RULE_MEMS` to lets say 8 or 16 (changing `mems_event` to `uint16_t` in last case)? The device which will be flashed with new firmware will lost the previous configuration or something worse will happen?

ascillato, 2019-01-15:
You need to move the rest of the config in settings.h to modify the memory map. MEMs is not just a uint16_t, it is also every MEM. 16 MEMs is A LOT of memory on EEPROM. So, if you do that all your config will be scrambled. Not backward compatible.

benzino77, 2019-01-16:
Forgetting about backward compatibility will it work or there is no chance for that? I'm asking for a case when I have a "green field" - new device (or erased with `esptool` via serial connection), flashed with Tasmota with those "strange" settings (`MAX_RULE_MEMS >5`) will it work or the flash will be scrambled and there is no chance for it to work? It is just "academical" question - I'm just curious.

Thanks again for answering my not related questions (16 VARS is good enough for me)! My feature/change request was fulfilled in couple of hours - **this is amazing!**

PS.
@andrethomas is not a part of this discussion, but his MCP230xx driver pushed Tasmota (using amazing, constantly developed Rules) to completely new level. If he implement his idea regarding using Arduino, as "generic"  I2C slave for Tasmota (I know it is just an idea now) Elon Musk will have to send another Tesla in space, this time without David Bowie but with Tasmota on board ;)

arendst, 2019-01-16:
Making the requested changes will enlarge the settings struc by 110 bytes and move the rules area down approaching the undocumented eeprom save area but still within the available flash page.

So it could work.