arendst/Tasmota #4936: Wired Ethernet connection

wolfgangr, 2019-01-15:
**It would be great to have tasmota'ed gadgets with RJ45-Ethernet-plugs.**

Wireless is not always the best Solution for sensor and switch deployment.
There may be gadgets around with Ethernet and similiar features, but different GUI and protocols.
It would be nice to have a uniform user / admin solution, no matter whether the gadget at hand is connect by air or by wire. 
Swicthing a gadget's mode of connectivity should not require reconfiguration on the server side. 
This would be fine, if we had Tasmota on the wire.

I could not find anything relevant yet in the wiki or in the tasmota sources.
I have placed this question on the SonoffUsers list a few days ago and received some confirmation.

I am well aware that it does not fit the "sonoff-gadget first" and "one size fits them all" approach. 
So may be, in the end, ethernet connectivity might be maintained in some fork.

But even for that, it may be wise to agree at least on some convention of namespace and interface to allow a coherent maintenance of such a fork.

So may be this issue here is just a stub to collect information flowing around, to focus people interested in contributing, and in the end point to some place where the result ist maintained.

*possible alternatives considered:*
hand coding on top of PubSubClinet on arduino 
Pollin's AVRnetIO
https://www.ethersex.de/index.php/Ethersex

Jason2866, 2019-01-15:
Since Tasmota is only for ESP8266 and its derivants there is no support for wired Ethernet.
Tasmota is focused and build for ready made **wireless** devices. Starting with sonoff´s from ITEAD...

wolfgangr, 2019-01-15:
Did some first tests this night, based on the following considerations:

Preferrably we would stay with the ESP8266 platform, of course.

There are two module types around in the arduino CN sourced clone-ecosystem:
WIZnet W5100 and ENC28j60

Both are connected by SPI.
Thus they require access to the relevant 4 GPIO (D5 ... D8 in the NodeMCU naming scheme). 
This alone makes it difficult to add them on sth like the sonoff modules, since I assume you have to tap those right from the TQFP (or whatever) pins.
So, if we say ESP8266, the preferfed hardware would be some thing like the WEMOS D1 or some similar module in the NodeMCU ecosystem.

The ENC28j60 for some time has been considerably cheaper, but meanwhile the W5100 sells for ’ 4 €, so this is not an issue any more.

The W5100 is supposed to be the more robust solution. It implements lot of the TCP/IP stack in hardware. This places much less burden on RAM, Flash, CPU and timing restrictions - in contrast to the ENC28j60, which only does low level packet send/receive and leaves OSI layer 2 and 3 (as far as I can remember...) to the MCU
The arduione on board driver is documented to support W5100 (or its successors W5200, W5500) only, not the ENC28j60.
Earlier tests with AVR-Arduione confiormed the preference for W5100.

So I think if a combined WiFi-Ethernet on the ESP8266 would work at all, there were better chances to do this with the W5100 than with ENC28j60. 
Yes, there are reports on the web of sussfully running ENC28j60 on ESP8266, however, as far as I could figure out, not using the arduino-framework.

So, my first test was to try this simple 20-line hello-world-type example which is shipped along the arduino ethernet driver:
https://github.com/esp8266/Arduino/blob/master/libraries/Ethernet/examples/DhcpAddressPrinter/DhcpAddressPrinter.ino

Results: It partially seems to work a little bit :-(
At some boots, I can see DHCPDISCOVER/DHCPOFFER entries in my (Openwrt) router's log, sometimes I don't.
Sometimes I get the "I Failed to configure Ethernet using DHCP" message provided in the sketch for a DHCP timeout, sometimes I don't. 
Sometimes the watchdog does reboot the module, sometimes it does not.
I haven't even once watched a succesful DHCPACK.

So I can say that at least something ( the DHCPDISCOVER) is sent out.
So I think it is safe to conclude that SPI wiring and basic configuration is OK.
I may well be that the setup is not able to receive from the Ethernet at all.
However, the erratic, unreproducable behavior I watched let's me believe that the ardiono-ethernet-driver is heavily interering with the ESP-WiFi mimics and other ESP-core and arduino-core timing stuff.

Since I can't find any succesful report on the web referring to Ethernet & Arduino-Framwork & ESP8266, my suspicion ist that the Ethernet-driver is just a leftover and not yet successfully ported to ESP8266

I see 3 ways to go from here:
- fix Aruiono-Ethernet-driver on ESP8266
- use third-party driver
- switch to another platform

My next step will be to try the same test on a STM32

wolfgangr, 2019-01-15:
@Jason2866 & core team
OF course, I know.
Is it OK to continue detailed test talk here or should I do this on my own git account?

Can you assist me to port tasmota to some other platform e.g. the STM32?
While such porting is easy with 4-liner `platform.ini` for hello-world-scale playing, tasmotas `platform.ini` has quite some more lines. And there might as well be some hardware specific considerations buried deep in the code that is beyond mere GPIO mapping, I'm afraid.

May I beg for assistance when I stumble across those?
May I even beg for sbdy familiar with tasmota's nuts and bolts just to dump his memory on these questions here?

wolfgangr, 2019-01-15:
> You can ask Adrian and andrethomas in Discord Chat. Both knew Tasmota "inside"

poor old man, me ;-{
got used to WhatsApp, so I'll go to conquer Discord Chat now :-}
Meanwhile, if @ascillato (th correct adrian?) or @andrethomas stumble across this post, maybe they'll attacked by mercy...

But I got your point: 
It looks more promising to invest time in an Ethernet driver for ESP8266-arduino than in a STM32-port of tasmota. 
This also would allow to switch a gadget between Wlan and Ethernet without hardware changes.

andrethomas, 2019-01-15:
It will have to be added to the ESP8266 Arduino core for it to be useful in Sonoff-Tasmota as the physical layer will need to be handled by the core in the same way as the wifi is handled through the SDK. This is the only way for it to be transparently implemented in Tasmota IMHO.

That aside I would also like to see PPP over GSM support but this and the first mentioned is not an easy task to do... it's not as simple as hooking of an Arduino board with an ENCJ80/W5500 via SPI.

Porting Tasmota to the STM32F4 platform will also have its own challenges.

I think something important to remember is that Sonoff-Tasmota is actually intended for Sonoff devices, none of which have physical ethernet support so this would be a low priority for this project.

ascillato, 2019-01-15:
Your idea is very nice. Please open a request in the Arduino core repository. Thanks.

closed 2019-01-15 by ascillato2

wolfgangr, 2019-01-15:
Just encountered this library from teensy developer Paul Stoffregen:
https://www.pjrc.com/arduino-ethernet-library-2-0-0/
claims to be a replacement of genuine Arduino core lib( if not already pulled in .... have to check that....)

> Today I released the Arduino Ethernet Library version 2.0.0, for all Arduino boards (not just Teensy).

.... tested at ....
> Adafruit Huzzah ESP8266 Feather 

ah, I see:
> ESP8266 SPI library requires issue #2677 fixed – transfer(buffer, size)

https://github.com/esp8266/Arduino/issues/2677

can't get the point at first glance .... fooder to read tonight....

wolfgangr, 2019-01-15:
@ascillato2 
> Your idea is very nice. Please open a request in the Arduino core repository. Thanks.

Do you refer to my Ethernet request or to @andrethomas ' PPP over GSM?
Is it OK for you to contiune posting on a closed issue?

btw: I like the way of GSM as a third access path as well.
Actually, I have pojects in mind where I considerd GSP Internet by some openwrt or raspi and ethernetting sensors...
How would the hardware of GSM look like? Most dongles I am aware of ar USB, aren't they?

ascillato, 2019-01-15:
Sorry, I didn't tag you. Yes, the message was intended for you. Thanks for asking for clarification.

The other point, there are bunch of hardware for gsm that has ttl serial connections. Look for sim900 chip for example.

ascillato, 2019-01-15:
The guys from arduino core are already working on the ethernet module. So, may be they can help you better with that.

andrethomas, 2019-01-15:
I did chat with @d-a-v a few weeks ago and he indicated that he is working on physical ethernet and will then look at PPP over GSM.

andrethomas, 2019-01-15:
The ESP8266 Arduino core uses a somewhat modified implementation of lwip :)

andrethomas, 2019-01-15:
@wolfgangr Leave it with d-a-v he will get around to it at some point... he is quite involved in the ESP8266 Arduino core development so the lwip2 can be stripped out and rebuilt that's not a big deal... the problem with most of us is the time management required to focus on day to day stuff vs. what we "like" to focus on.

wolfgangr, 2019-01-16:
I think it fits the interests of anybody if we continue this issue on WESPTA fork
https://github.com/VerboteneZone/WESPOTA/issues/1

ascillato, 2019-01-16:
Just for Reference: It is an already open issue for this at https://github.com/d-a-v/W5500lwIP/issues/4