arendst/Tasmota #4938: MDNS.addService("http", "tcp", WEB_PORT) is executed each second

araczkowski, 2019-01-15:
**Describe the bug**

I'm trying to advertise web server
on sdk core 2.3.0 this is ok - I'm able to see the service using the avahi-discover, 
but on 2.5.0 the device is not visible in network, I don't know why

of course I have 
USE_DISCOVERY and  WEBSERVER_ADVERTISE on in the my_user_config file

I figure out that the 
`MDNS.addService("http", "tcp", WEB_PORT);`
is in the support_wifi.ino

to be sure that this part is executed I added a log info below
```
snprintf_P(log_data, sizeof(log_data), "MDNS.addService");
AddLog(LOG_LEVEL_INFO);
```

now I can see that the line MDNS.addService("http", "tcp", WEB_PORT); is executed each second - see the screenshots below

**it is normal that MDNS.addService is executed each second?
how we can advertise the device on 2.5.0 maybe something was changed in the core version?**

Thanks

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : D1 mini 
- [x] _Development/Compiler/Upload tools used_ : PlatformIO, Arduino SDK Core 2.5.0
- [x] _If a pre-compiled release or development binary was used, which one?_ : latest
- [x] _You have tried latest release or development binaries?_ : yes
- [] _Provide the output of command_``status 0`` :

**To Reproduce**
program device with lates verion compiled on sdk 2.5.0
try to find the http service from device in network via avahi-discover

**Expected behavior**
be able to see the http service via avahi-discover

**Screenshots**
[image]

Jason2866, 2019-01-15:
https://github.com/arendst/Sonoff-Tasmota/issues/3275#issuecomment-406590312
It is not supported. If it works fine, if not Theo wont fix it.

araczkowski, 2019-01-15:
I'm talking about advertise the devices web service in the network -  this option seems to be still supported in Tasmota

3257 is about resolving the MQTT broker address on device via mDNS hostname, 
I see that this is not implemented and not planned 

4793 is about possiblity to set the mDNS on/off
but when it is on, the advertise the devices web service in the network is still possible/supported?

ascillato, 2019-01-15:
As that is a feature of the Arduino core, please try a simple arduino sketch with that, to test if the problem is in the core or in Tasmota. Thanks.

araczkowski, 2019-01-15:
ok - this was my plan too :)
later or tomorrow I will try this one
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266mDNS/examples/mDNS_Web_Server/mDNS_Web_Server.ino
and I will let you know

Thanks and have a nice day

araczkowski, 2019-01-15:
> It is correct for it to be executed frequently otherwise other hosts will never know the device exists.

thanks for clarification :)
I don't have lot of experience with C/C++ code, in Python or Java this kind of addService is done once in lib and the rest is magically ;) done without repeating
so I thought that maybe the MDNS.addService is taking care about repeating this
that is why I asked

araczkowski, 2019-01-15:
according to the advice from @ascillato - Thanks!

I just checked the simple arduino sketch on the same D1 mini:
https://raw.githubusercontent.com/esp8266/Arduino/master/libraries/ESP8266mDNS/examples/mDNS_Web_Server/mDNS_Web_Server.ino

the MDNS.addService is executed only once and works OK
please see the screens below

[image]

[image]

ascillato, 2019-01-15:
Ok. So, it is a Tasmota bug for core 2.5.0. Thanks for reporting.

Another question. Have you tried the latest stage core from arduino repository? The one from today?

andrethomas, 2019-01-15:
Ah yes, was a bit early for me...

arendst, 2019-01-15:
I guess it should have been executed only once in core 2.3.0 too.

I'll change it to being executed only once in next dev version. Pls let me know if it solves your issue as I cannot test it myself.

araczkowski, 2019-01-15:
@arendst 
> I guess it should have been executed only once in core 2.3.0 too.

according to the examples from MDNS + HTTP server sketch the MDNS.addService is called only on setup to advertise the service, after the wifi connection and HTTP server

then in the loop  MDNS.update() is called many times

but I don't know if this was always like this or it is new approach... 
 
Of course I will test this deeply, and I will try to dig this MDNS topic more - I really like this feature - **thank you** :+1:

@ascillato
> Another question. Have you tried the latest stage core from arduino repository? The one from today?

I did the test again, this time on Core/SDK | STAGE/3.0.0-dev(c0f7b44)
The same device, the same network, the same core but different results:
simple arduino sketch -> the device web service is correctly advertised in the network in the network
Tasmota -> no service is advertised

Thanks!

[image]

arendst, 2019-01-15:
Did you enable mDns with command ``SetOption55 1``

araczkowski, 2019-01-15:
@arendst 
I didn't tested this on Tasmota DEV yet 
my last test (and comment) was on latest core 2.5.0 and Tasmota mater (Release 6.4.1)

I will check this deeply tomorrow on Tasmota DEV and I will let you know if is OK
Thanks and Regards

araczkowski, 2019-01-16:
Hi @arendst 

I did a test:

> Program Version | 6.4.1.9(sonoff)
> Core/SDK Version | STAGE/3.0.0-dev(c0f7b44)
> Hostname | my_device_to_test_mdns.local

it doesn't work, so I checked if the:
1. MDNS_ENABLED / SetOption55 == 1
2. MDNS.begin is executed 
3. MDNS.addService is executed 

all above was correct

than I added these lines in support_wifi.ino WifiCheckIp

```
    if ((wifi_status == WL_CONNECTED) && (2 == mdns_begun)) {
      MDNS.update();
      AddLog_P(LOG_LEVEL_DEBUG_MORE, D_LOG_MDNS, "MDNS.update");
    }
```

in DEBUG_MORE we can see what is going on here 

[image]

**I don't know if this is a best way but it works :)**

please see the screens below from avahi-discover, service browser on Android and browser

if you want I can submit a pull request with this code but I suppose that you have better idea how to write this full correctly :)
Of course I will retest this - just please let me know
Thanks a lot! :+1: 

[image]

[image]

[image]

ascillato2, 2019-01-16:
Great, please do the PR. Thanks.

PR #4951 (2019-01-17): MDNS.update();
Fixes #4938

tested on core 2.3.0, 2.4.2 and 2.5.0
:+1:
closed 2019-01-16 by ascillato2