arendst/Tasmota #4939: HOLD retain

netpok, 2019-01-15:
**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
Currently one can set retain of button/switch on all messages. This means if HOLD message is published due to a longpress it will overwrite the state of the switch. It would be great if HOLD messages could be published without retain.

**Describe the solution you'd like**
It could be changed globally in code but that would not be backward compatible, so there should be a settings for it.

**Describe alternatives you've considered**

- An option which defines the hold state retain flag (not really backward compatible, but one can publish hold messages retained, I'm not seeing the use of this, because nothing wold clear the retained message.)
- An option which overrides the hold state retain flag, with the default setting of true (this is backward compatible, but one cannot set the system to publish only retained hold messages)

**Additional context**
I would implement this feature but first I need your opinion about the following:

- should it be an alternate setting or an override setting
- should it be global (1 bit of eeprom)  or per device type (switch and button, 2 bits of eeprom)

Here is the truth table of the two possible solution:

- Alternative setting:

| Retain | HoldRetain | Single Press | Long Press   |
|--------|------------|--------------|--------------|
| False  | False      | Not Retained | Not Retained |
| False  | True       | Not Retained | Retained     |
| True   | False      | Retained     | Not Retained |
| True   | True       | Retained     | Retained     |

- Override setting:

| Retain | HoldRetain | Single Press | Long Press   |
|--------|------------|--------------|--------------|
| False  | False      | Not Retained | Not Retained |
| False  | True       | Not Retained | Not Retained |
| True   | False      | Retained     | Not Retained |
| True   | True       | Retained     | Retained     |

ascillato, 2019-01-15:
Hi

You don't need to add any extra feature. Please check the wiki. There is an example of this at rules. You just need to change in the example the command publish to publish2

ascillato2, 2019-01-15:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-15 by ascillato2

netpok, 2019-01-15:
This is not the publish/publish2 problem. This is about the ButtonRetain function when ButtonTopic is set.

ascillato, 2019-01-15:
Rules were added by Theo to modify the standard behaviours of Tasmota to do whatever you want. You can make the behaviour you want with the example of the wiki at rules. Please, check that. Thanks

meingraham, 2019-01-15:
@ascillato - `There is an example of this at rules. You just need to change in the example the command publish to publish2`

@netpok - `This is about the ButtonRetain function when ButtonTopic is set.`

If this requires a Rule to implement, and also requires ButtonTopic or SwitchTopic, then it cannot be accomplished because you cannot have Rules when either ButtonTopic or SwitchTopic are set; yes?

netpok, 2019-01-15:
Sorry, I misunderstood your reply, but I still think this can be useful, because to reach the mentioned functionality, you would need the following rules:

> on power1#boot=1 do var5 OFF endon
on power1#boot=0 do var5 ON endon
on power1#state=1 do var5 OFF endon
on power1#state=0 do var5 ON endon
on button1#state=2 do publish2 cmnd/desk/POWER1 %var5% endon
on button1#state=3 do publish cmnd/desk/POWER1 HOLD endon

Assuming only a 4 character long topic name one button takes up 258 characters from rule space and one variable, using this method on a 4ch leaves you with less than 1 free rule slot and 1 free variable.

With relying on a somewhat undefined use of commands one can use:

>on power1#boot do backlog var5 1; sub5 %value% endon
on power1#state do backlog var5 1; sub5 %value% endon
on button1#state=2 do publish2 cmnd/desk/POWER1 %var5% endon
on button1#state=3 do publish cmnd/desk/POWER1 HOLD endon

But this is producing floats in the messages which are probably not really accepted in home automation toolkits.

ascillato, 2019-01-15:
From your original request you only need

```
Rule1 1 
Rule1 on button1#state=3 do publish cmnd/desk/button1 HOLD endon
```

netpok, 2019-01-15:
Looks like I haven't included in the first message, that the buttons do use mqtt for single press action, with retain enabled.

The desired output of the system:
Single press: retained mqtt message to turn on or off the device
Long press: non-retained message with the contents of HOLD

ascillato, 2019-01-15:
Ok. As that is a super specific use case, the best it is indeed to use rules.

And this is the same as the example in the wiki. Just change publish to publish2

netpok, 2019-01-15:
I think this could be a not so specific use case, because this means you keep the default mechanism which is preferred in most cases but you can also create alternative function without the need of rules or a republishing mechanism on server side. This only costs one (or two exta bit of eeprom) and could be a great addition to tasmota and I think it could be used by many. I would implement this and send a PR, but of course the last word is yours.

ascillato, 2019-01-15:
The issue with using only buttontopic or switchtopic is the fallback feature that it has. That is why, it is better for use cases where you are not wanting to control the local relay, to use rules.

ascillato, 2019-01-15:
So, I think it is better and more clear just to let the user make their rules for their use-cases. Yours seems to be a complex automation, but again is your use-case, so it is valid that you make your own rules for that.

ascillato, 2019-01-15:
Hope you understand the extra implications your request has for newbie users and for general use.

netpok, 2019-01-15:
I also get your point about newbie users, but this feature assumes that one already using buttonretain and wants to decrease the effect of that setting (so it shouldn't affect the current and new users unless specifically enabling the feature. Also the method you mentioned is only persistent with using local flash writes.

I will implement this feature and test it, if it works I will send a PR if it works correctly and as intended.

ascillato, 2019-01-15:
Rules as well as the rest of features are persistent. All settings survive local flash and also ota and otaurl.

ascillato, 2019-01-15:
Sorry but I don't get it why the rules proposed are not enough for you. They produce the behaviour you want and also are already implemented.
Remember that rules were implemented by Theo for alowing user for making these types of behaviours for Tasmota and solve a wide range of use-cases.