arendst/Tasmota #4940: AM2301 not reading values

nicolasvac, 2019-01-15:
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Development/Compiler/Upload tools used : ESPEasylash
- [X] Hardware used : Sonoff TH16
- [X] If a pre-compiled release or development binary was used, which one? : Tasmota sensors 6.4.1
- [X] You have tried latest release or development binaries? :
- [X] Provide the output of command ``status 0`` :
```
9:38 RSL: Group 0, Index 1, Command STATUS, Data 0
16:39:38 MQT: stat/sonoff_02/STATUS = {"Status":{"Module":4,"FriendlyName":["SONOFF-02"],"Topic":"sonoff_02","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:39:38 MQT: stat/sonoff_02/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-sensors.bin","RestartReason":"Power on","Uptime":"0T00:00:50","StartupUTC":"2019-01-15T15:38:48","Sleep":50,"BootCount":22,"SaveCount":47,"SaveAddress":"F9000"}}
16:39:39 MQT: stat/sonoff_02/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sensors)","BuildDateTime":"2018-12-24T14:45:55","Boot":7,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
16:39:39 MQT: stat/sonoff_02/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":3,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["VACCARI",""],"TelePeriod":60,"SetOption":["00008009","558180C0","00000001"]}}
16:39:39 MQT: stat/sonoff_02/STATUS4 = {"StatusMEM":{"ProgramSize":550,"Free":452,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1420C2","FlashMode":3,"Features":["00000809","0FDEE794","0003A3A4","B7FFBFCC","0002BBC0"]}}
16:39:39 MQT: stat/sonoff_02/STATUS5 = {"StatusNET":{"Hostname":"sonoff_02-6392","IPAddress":"192.168.1.102","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.200","Mac":"80:7D:3A:31:58:F8","Webserver":2,"WifiConfig":4}}
16:39:39 MQT: stat/sonoff_02/STATUS6 = {"StatusMQT":{"MqttHost":"raspberrypi-01.vaccari.lan","MqttPort":8883,"MqttClientMask":"sonoff_02","MqttClient":"sonoff_02","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
16:39:39 MQT: stat/sonoff_02/STATUS7 = {"StatusTIM":{"UTC":"Tue Jan 15 15:39:39 2019","Local":"Tue Jan 15 16:39:39 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:38","Sunset":"17:20"}}
16:39:39 MQT: stat/sonoff_02/STATUS10 = {"StatusSNS":{"Time":"2019-01-15T16:39:39","AM2301-04":{"Temperature":null,"Humidity":null},"AM2301-14":{"Temperature":null,"Humidity":null},"TempUnit":"C"}}
16:39:39 MQT: stat/sonoff_02/STATUS11 = {"StatusSTS":{"Time":"2019-01-15T16:39:39","Uptime":"0T00:00:51","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"VACCARI","BSSId":"A4:91:B1:2A:8A:0F","Channel":6,"RSSI":30}}}
```

Hi, i've just got a shiny DHT 16 flashed with tasmota sensors, bought an AM2301 sensor, configured in GPIO14, but i get null results.

With weblog 3, i get this:

DHT: Timeout waiting for start signal high pulse

nicolasvac, 2019-01-15:
[image]

Module configuration, tried both without and with GPIO4

ascillato, 2019-01-15:
Hi,

Have you searched in issues? This has been addressed before several times. Most of the times was just bad wiring. Can you check older issues and also your wiring? Thanks

nicolasvac, 2019-01-15:
Clear view of the cables: 
Black goes in to VCC
White goes to OUT
Green goes to GND
Red goes to unnamed

[image]

Jason2866, 2019-01-15:
Black to VCC
Green to  GND
White to GPIO14
Just GPIO14 as AM2301 configured. GPIO 4 set to none
Is plug really completly in socket? Most of the time this is the issue!

nicolasvac, 2019-01-15:
So i've just tested with a circuit tester and found out that the sensor board is just crappy.
I'm gonna send it back
Can i ask you guys a recommendation about the best sensor to use for TH 16 ?

Thanks for help!

Jason2866, 2019-01-15:
The original from ITEAD are good

closed 2019-01-15 by ascillato2