arendst/Tasmota #4942: New rule trigger on SerialReceived/SSerialReceived

GianCann, 2019-01-15:
**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
Manage action when some data received on serial bridge

**Describe the solution you'd like**
Trigger like:
on SerialReceived#state do command %value% endon
and
on SerialReceived#data=xyz do command %value% endon

**Describe alternatives you've considered**
...

**Additional context**
Example: Using in congiuction with Nextion Intelligent Touch Display

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-01-15:
Nice idea.

What would be the difference between the triggers you propose:

SerialReceived#state

And

SerialReceived#data

GianCann, 2019-01-15:
Hi @ascillato2

SerialReceived#state is usefull to send specific MQTT message to other device, passing the %value% on payload

SerialReceived#data=xyz is usefull to intercept specific value and translate in local action (power a relay/change settings/show specific data on display via SerialSend...)

ascillato, 2019-01-15:
ok, so `SerialReceived#data` and `SerialReceived#data=xyz` is enough for your idea, right?

GianCann, 2019-01-15:
Yes, sure.
#state are not necessary at this point

tjareson, 2019-01-29:
Is there anything that can be done to support this feature request?
My use case is an RFID controlled lock. RFID reader is connected to serial port of sonoff. In case that the server is not responding to the MQTT msg of the serial bridge, certain RFIDs can be stored in tasmota locally to allow for open the lock in case of network issues, server outage etc

fvanroie, 2019-02-07:
I too am looking for a way to execute a rule on SerialSend. The solution proposed by @GianCann would fit our purpose too.

Jason2866, 2019-04-09:
@arendst  Can you take a look at this feature request?  Because @ascillato  is super busy and will have no time the next weeks
Thx

arendst, 2019-04-09:
Will consider

ascillato2, 2019-04-10:
Closing this request as has been added.

Thanks.

closed 2019-04-10 by ascillato2