arendst/Tasmota #4945: ability to use data elements of STATUS messages as data items to publish to Domoticz ie. RSSI

K1WIZ, 2019-01-15:
**Have you look for this feature in other issues and in the wiki?**

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-01-16:
Hi,

Why you need this? RSSI is already sent to domoticz on every mqtt message.

K1WIZ, 2019-01-16:
But it's not.   I'm sending the analog value of pin A0 by rule (IDX 421).   This is a water tank level sensor.   The unit is located near the fringe of the wifi network so I setup a second dummy custom sensor (IDX422) to display just the wifi signal over time.   (domoticz has awesome history graphs useful in troubleshooting and analysis).    Wifi is not sent to idx 421, so therefore I need a way to get the rssi data everytime telemetry fires.

My current rule:

> 02:37:49 MQT: tele/sonoff/INFO1 = {"Module":"Generic","Version":"6.4.1.8(sonoff)","FallbackTopic":"cmnd/DVES_01CD81_fb/","GroupTopic":"sonoffs"}
02:37:49 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"tank","IPAddress":"10.50.50.85"}
02:37:49 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
02:37:57 MQT: tele/sonoff/STATE = {"Time":"2019-01-16T02:37:57","Uptime":"0T00:00:14","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"MIND CONTROL BG","BSSId":"0E:EC:DA:38:D5:8A","Channel":1,"RSSI":98}}
02:37:57 MQT: tele/sonoff/SENSOR = {"Time":"2019-01-16T02:37:57","ANALOG":{"A0":1024}}
02:37:57 RUL: TELE-ANALOG#A0 performs "publish domoticz/in {"idx":421,"nvalue":0,"svalue":"1024"}"
02:37:57 MQT: domoticz/in = {"idx":421,"nvalue":0,"svalue":"1024"}

thanks.

ascillato, 2019-01-16:
The rssi is sent with every idx like power on and the rest of the standards of domoticz Tasmota menu. Please, check domoticz driver code in Tasmota. It is sent along with batery level and other data.

ascillato, 2019-01-16:
So, if you use a rule, it is no need to send those data that are being sent in the others idx.

K1WIZ, 2019-01-16:
im sorry but I see no evidence that the rssi is being sent along with the primary sensor value (in this case analog pin A0).   I have other tasmota driven esp8266 located around the site and no rssi data is visible from them on domoticz either.   they are all talking to Domoticz via my broker and here's what I see on the broker for topic domoticz/in:

> {"idx":193,"nvalue":0,"svalue":"10.2","Battery":38,"RSSI":3}
{"idx":404,"nvalue":0,"svalue":"14.1;32.4;2","Battery":46,"RSSI":5}
{"idx":377,"nvalue":0,"svalue":"20.9"}
{"idx":403,"nvalue":0,"svalue":"24.3;19.0;2","Battery":46,"RSSI":7}
{"idx":405,"nvalue":0,"svalue":"","Battery":38,"RSSI":7}
{"idx":380,"nvalue":1,"svalue":"","Battery":85,"RSSI":7}
{"idx":420,"nvalue":0,"svalue":"-2.4;66.7;1","Battery":45,"RSSI":6}
{"idx":372,"nvalue":0,"svalue":"22.6","Battery":40,"RSSI":4}
{"idx":382,"nvalue":0,"svalue":"0;5811.1","Battery":85,"RSSI":7}
{"idx":384,"nvalue":1,"svalue":"","Battery":88,"RSSI":5}
{"idx":421,"nvalue":0,"svalue":"1024"}

In the last line (this one being the tank sensor) the rssi is not sent because I had to use a rule to convince the esp8266 to send the raw value of the analog pin (A0).

ascillato, 2019-01-16:
Yes, exactly, in a rule is not there, but in the rest of idx it is there as your log states:

{"idx":193,"nvalue":0,"svalue":"10.2","Battery":38,"RSSI":3}

So, there you have RSSI

That is why, there is no need to duplicate the information adding more information in a rule. It is already sent to Domoticz with the standards idx.

ascillato2, 2019-01-16:
Could you use RSSI in domoticz from the standard messages?

ascillato2, 2019-01-17:
Closing this issue as there is no feedback. When you have time, please tell us if you could configure your Domoticz to see the rssi from the standard idx messages. Thanks

closed 2019-01-17 by ascillato2