arendst/Tasmota #4947: Add a new SetOption parameter to append 0xFF 0xFF 0xFF bytes  to SerialSend2/SSerialSend2

GianCann, 2019-01-16:
**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
Itead Nextion Display it's a very powerful and low cost HMI device, simple to programming with his free GUI editor.
https://www.itead.cc/display/nextion.html

The display interact with other devices trought simple serial communications.
Creating a very engaging and interactive user interface is really a breeze.

For example, to update the text showed in a TextBox (named "txtTemp") positioned in a specific page (named "page2") we have to send this simple comand:
page2.txtTemp.txt="Hello world"

Change a background color of this TextBox to Red:
page2.txtTemp.bco=63488

In conjuction with ESP8266/Tasmota can act as remote control console that fit at the place of wall switchs.
Can we use it to show data from other sources (sensors/switchs and so on...), send command to other device, set-up parameters) and offers a multipage interface (up to 255 pages) with unlimited possibility to expand Tasmota/ESP8266's ecosystem.

I tryed to use Nextion with Tasmota (receiving data it's from his very-very simple) but for send a command it's required that the string (plain text) are terminated with 3 '0xFF' bytes.
(page2.txtTemp.txt="Hello world" + 0xFF + 0xFF + 0xFF )

I don't find any simple solution to send thes 3 bytes over MQTT messages (without using other external software like NodeRed).

**Describe the solution you'd like**
I ask to Tasmota's Developers Team, to implement a new SetOption parameter to append thes 3 0xFF bytes to other commande bytes when serial data are sended wit

**Describe alternatives you've considered**
See this topic:
https://github.com/arendst/Sonoff-Tasmota/issues/3560#issuecomment-453861912

**Additional context**
The display also offers 8 I / O pins.
Think a simple ESP-01 connected to Nextion... ;)

**(Please, remember to close the issue when the problem has been addressed)**

GianCann, 2019-01-16:
Also, when this SetOption is Enable, trim 0xFF + 0xFF + 0xFF from SerialReceived data.
Thank you

ascillato2, 2019-01-16:
Hi, 

This request is in an already open issue https://github.com/arendst/Sonoff-Tasmota/issues/3560

You already tag Theo there. Please, wait for his answer. May be he can came up with a more general solution for your request.

Thanks

GianCann, 2019-01-16:
ok @ascillato2,

I think to open a 'Feature request' to better explain the case.

Sorry for mistake ;)

ascillato, 2019-01-16:
There is no mistake. No need to apologize. 

All the information related to that use case and the feature request has been very well explained in the referenced open issue.

Anyway if you think that more information is needed to be added, please do. 

Let's wait for Theo's answer.

Thanks for sharing your ideas and tests. Very appreciated

arendst, 2019-01-17:
As this is specific to NextIon it won't be implemented as a SetOption.

You seem to have a workaround using Node-Red so apparently you are able to send three 0xFF via serial.

What you actually need is a feature called escape sequence allowing to send escape chacaters like Backspace (\b). This is implemented but currently does not allow octal or hexadecimal input.

I'll extend it with allowing to send hex values as \xFF or octal as \777 equal like C standard.

GianCann, 2019-01-17:
Yes @arendst, you say good.

And, thinking to other device as the SIM800/900 GSM module, multiple bytes.
Nextion? Use SerialDelimiter(x) FFFFFF
GSM800? Use SerialDelimiter(x) 0D0A

Thank you very much!

arendst, 2019-01-17:
Well 0D0A have always been available as \r\n. Look for Escape Sequences on google.

GianCann, 2019-01-17:
Thank you!
Now i know that with SerialDelimiter it's possibile to use not only one escape char.... ;)

I go to read the code to understand better

arendst, 2019-01-17:
Latest additions allow to send hexadecimal bytes using escape sequence \x like in ``SSerialSend3 Hello World \xFF\xFF\xFF``

closed 2019-01-17 by arendst

GianCann, 2019-01-17:
Thank you very much!!!
I try to build this night ;)

gitolicious, 2019-01-21:
For me this doesn't look right. I am sending
`SerialSend3 Hello World \xFF\xFF\xFF`
Snooping the serial output with [terminalbpp](https://sites.google.com/site/terminalbpp/) I see
```
48 65 6C 6C 6F 20 57 6F 72 6C 64 20 FF FF FF 46 
5C 78 46 46 5C 78 46 46
```
which translates to `Hello World ÿÿÿF\xFF\xFF` in ASCII. Does it work properly with serial bridge for you, @GianCann?

GianCann, 2019-01-27:
Hi @gitolicious,
I try later, because in these days I working on other section of my project and I don't have the time for test this enhancement.

andreapede, 2019-05-06:
Same issue from @gitolicious also in 6.5.9 core 2.4.2

Jason2866, 2019-05-07:
Does work (sended):
 ```
17:34:50 CMD: SSerialsend3 HelloWorld\xFF\xFF\xFF\n
17:34:50 MQT: stat/sonoff/RESULT = {"SSerialSend":"Done"}
```
Received (captured with device in modus SSerialsend5)
`17:34:51 MQT: tele/sonoff-53CF5F/RESULT = {"SSerialReceived":"48656c6c6f576f726c64ffffff0a"}`